MichaIng/DietPi #5382: Open Beta v8.3 | Please help testing and hardening the upcoming release

MichaIng, 2022-03-26:
RC version | v8.3.0
---------- | -------
v8.3.0 PR | #5377
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2022-04-02**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.3
____
### v8.3.0
_(2022-03-26)_

New images:
- Container | Initial support for container systems has been added to DietPi, using hardware ID 75. For now it is assumed that the container guest does not require an own network stack, but uses the host network interface or one managed by the container engine. This works well with systemd-nspawn which we use for DietPi image generation and for testing DietPi features in CI/CD pipelines. Further support and images for specific container engines, like Docker or LXC with Proxmox and QEMU will follow.

New software:
- PHP Composer | The PHP package manager has been made a dedicated install option.

Changes:
- DietPi-Banner | It is now possible to hide the credits section from the DietPi banner. Run "dietpi-banner" to configure its content. Many thanks to @math-gout for contributing this feature: https://github.com/MichaIng/DietPi/pull/5355
- DietPi-Drive_Manager | Adding an NFS drive now allows to select a specific export from an NFS server, detected via "showmount -e <IP/hostname>". Entering a custom path is still possible, including the single slash "/" which was always used before. If the NFS server has no root export defined via "fsid=0", this solves the issue that the full root directory structure is mounted and allows to mount multiple shares from the same server. Many thanks to `@bamyasi` for doing this suggestion: https://dietpi.com/phpbb/viewtopic.php?t=5488
- DietPi-Drive_Manager | NFS and Samba network drives can now be mounted to any directory on the server, not necessarily below /mnt.
- DietPi-DDNS | It is now possible to select the IP family to associate with your DDNS domain. For new setups, IPv4 is used by default for best compatibility. If you configured your DDNS domain already, no IP family is forced to not change existing behaviour, unless selected explicitly. We recommend to use IPv4 so that your server can be reached with any client from any network, regardless whether client and network support IPv6 or not. Many thanks to @TomBayne for suggesting this feature: https://github.com/MichaIng/DietPi/issues/5354
- DietPi-Software | motionEye: We worked together with other contributors to revive motionEye and port it over to Python 3, which also allowed us to re-enable it on Debian Bullseye systems. It requires further careful testing before a stable release can be done, but common functionality works. We enabled it with a "beta" mark in DietPi-Software. Visit the new home of motionEye, and if you want, contribute or help testing: https://github.com/motioneye-project/motioneye
- DietPi-Software | Node-RED: The "nodered" service user is now added to the "spi" system group automatically, relevant on Raspberry Pi to grant it access to SPI-attached sensors and similar. Many thanks to `@devifast` for reporting a related issue: https://dietpi.com/phpbb/viewtopic.php?t=10134
- DietPi-Software | TasmoAdmin: Reduced the downloaded data size from ~150 MiB to ~4 MiB by downloading the runtime files of the latest release only instead of the whole GitHub repository archive.

Fixes:
- Network | Resolved an issue where the systemd network targets could have been reached before the network adapter was even detected. Many thanks to `@Totila` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=10167
- DietPi-WiFi-Monitor | The first default gateway check is not done after the first tick rate interval instead of immediately after service start. This adds another grace period at boot to reduce the chance that the WiFi monitor reconfigures the interface which is just about to be configured for the first time. This should usually not be required, but may help with cases where the initial DHCP lease or WPA authentication time out of fail.
- DietPi-TimeSync | Resolved an issue where the script threw a syntax error where it shouldn't, which however didn't affect functionality. Many thanks to @adminy for reporting this issue: https://github.com/MichaIng/DietPi/issues/5347
- DietPi-Login | Resolved an issue where a prompt about an unsupported SSH client terminal was incorrectly shown. Many thanks to @jahway603 and @Dagon for reporting this issue: https://github.com/MichaIng/DietPi/issues/5341, https://github.com/MichaIng/DietPi/issues/5366
- DietPi-Software | Kodi: Resolved an issue on Raspberry Pi Bullseye systems where libraries were missing to run Kodi via KMS/DRM. Many thanks to `@derebo` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=10161
- DietPi-Software | Mopidy: Resolved an issue where there installation failed. Many thanks to `@amibumpin` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=10188
- DietPi-Software | rTorrent: Resolved an issue where the web interface didn't work with Nginx because the PHP handler wasn't set. Many thanks to @vinhtq115 for providing the solution: https://github.com/MichaIng/DietPi/pull/5375

elitearmedforce, 2022-04-01:
Recently installed fresh install of RPi4 64bit. Faced a weird issue with Kodi during startup. Installed Kodi from dietpti-software and then updated to beta version of dietpi. From the beginning, Kodi didn't work and even with beta update it didn't work. I think in 64bit install, the GPU/RAM memory split asked is 64/72 instead of 256(I am not sure about this as I installed it days ago), so I changed it to 256 and still didn't work even after restart. I am not sure what I did to fix it, by checking the terminal history, just **installed all the build packages** from [Kodi for bullseye](https://forums.raspberrypi.com/viewtopic.php?t=323303) and **restarted it and worked**. My point/suggestion is to install these packages during Kodi install.

Why did I remove 32bit of dietpi?
Well, I kinda fcked up with mariadb-common package for Kodi. I updated packages with "apt update" and found mariadb-common was creating issues with the update. I remembered that I installed it with NextCloud when I tried out the software, which isn't used anymore, so removed it with "apt purge mariadb-server*/mariadb-common*", later with huge regret found out that was used by Kodi :DDDDD. BTW, I did remove some files related to MariaDB and libxxxxx:armfh before the purge command so it was kinda messed up already.

**No issues with the latest beta**(the Kodi issue for existed before i updated to latest beta)

Specs:
SBC: RPi4
RAM: 4GB

Issue pic(poor quality):
[image]

Terminal History
[image]

MichaIng, 2022-04-02:
When running `apt purge ...`, always watch the list of additionally purged dependants, before confirming, where `kodi-bin` and `kodi` were listed 😉. It uses the MariaDB client library, which is the reason, not the MariaDB server, of course.

> the GPU/RAM memory split asked is 64/72 instead of 256

That is intended, using more than the default 64/72 MiB is a waste with recent RPi kernel/firmware and KMS.

> installed all the build packages from [Kodi for bullseye](https://forums.raspberrypi.com/viewtopic.php?t=323303) and restarted it and worked

You mean the packages from the second post? Well, those are required for compiling Kodi, not for running Kodi after installing the compiled binary package.

> Faced a weird issue with Kodi during startup.

Okay, so the actual issue is, based on the error message, the missing `swrast_dri.so`. So we need to find our which package ships this file and ask RPi devs/Dom to add it to the RPi Kodi package dependencies. There it is: https://packages.debian.org/bullseye/amd64/libgl1-mesa-dri/filelist

So this is sufficient:
```sh
apt install libgl1-mesa-dri
```

This was solved already for DietPi v8.3: https://github.com/MichaIng/DietPi/commit/293f35b
A Kodi reinstall after applying to the beta would have hence solved it as well 🙂.

elitearmedforce, 2022-04-02:
> When running `apt purge ...`, always watch the list of additionally purged dependants,

laxness in reading the warning while removing the package as I thought it was used only by Nextcloud.

> That is intended, using more than the default 64/72 MiB is a waste with recent RPi kernel/firmware and KMS.

I will put it back to normal.

> So we need to find our which package ships this file and ask RPi devs/Dom to add it to the RPi Kodi package dependencies.

Good that the same problem is already notified by another user. I didn't read the changelog😅.
It didn't work for me as I installed Kodi before upgrading to the beta branch, i think. now it is fine.

Thank you again for the hard work🎉

MichaIng, 2022-04-02:
### DietPi v8.3 has been released: #5397

Many thanks to all testers ❤️!

closed 2022-04-02 by MichaIng

EdWaQ, 2022-04-09:
Is there any way to upgrade Kodi from 18 to 19 ?

Joulinar, 2022-04-09:
I opened a new issue for your question as it has nothing to do with the Beta. As well 8.3 has been released already.