MichaIng/DietPi #5384: USB3 external HD very erratic / slow

jcw, 2022-03-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.2.2
- Distro version | bullseye 0
- Kernel version | `Linux duppie 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux`
- SBC model | RPi 3 Model B+ (aarch64)
- Power supply used | official supply, 5.1V 2.5A
- SD card used | 4 GB nameless

#### Additional Information (if applicable)
- Software title | no s/w (other than switching to OpenSSH for SFTP, see below), this is a hardware issue
- Can this issue be replicated on a fresh installation of DietPi? yes, IMO

#### Steps to reproduce
1. install fresh dietpi
2. attach external 2.5" USB Seagate disk (One Touch HDD, 5 TB)
3. migrate root to external drive
4. enable internal SFTP in OpenSSH
5. send a bunch of files to this machine over SFTP

#### Expected behaviour
- write performance of 10 .. 20 MB/sec

#### Actual behaviour
```
$ vmstat 3
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 0  0      0  27636 150540 746148    0    0     0   944 10242  321  3  3 94  1  0
 1  0      0  24940 150540 749592    0    0     0     0 10529  253  3  3 94  0  0
 0  0      0  22644 150548 752876    0    0     0  1981 10587  433  3  3 92  2  0
 0  1      0  29116 150548 746036    0    0     0  2331 10564  473  3  3 93  1  0
 0  0      0  25672 150556 749496    0    0     0  1088 10711  444  3  3 91  3  0
 0  0      0  23548 150564 752652    0    0     0  1067 10489  347  3  3 93  2  0
 0  0      0  30836 150564 744544    0    0     0     0 10240  233  2  3 94  0  0
 0  0      0  28116 150572 748132    0    0     0  4339 11220  717  3  4 89  5  0
 0  0      0  24872 150572 751312    0    0     0     0 10221  228  2  3 95  0  0
 1  0      0  31060 150224 745160    0    0     0    61 10463  256  3  3 94  1  0
 0  0      0  28760 150224 748324    0    0     0     0 10186  226  2  3 95  0  0
 0  0      0  24460 150232 751512    0    0     0   817 10611  341  2  3 93  1  0
```

#### Extra details
- assumes that EXT4's "lazy init" has been completed, i.e. the system is 100% idle before starting this transfer
- there are warning messages related to the USB disk driver:

```
[    3.831023] usb 1-1.3: The driver for the USB controller dwc_otg_hcd does not support scatter-gather which is
[    3.831046] usb 1-1.3: required by the UAS driver. Please try an other USB controller if you wish to use UAS.
``` 

I've looked through a few issues, but can't seem to find a clear explanation or fix:
- https://github.com/raspberrypi/linux/issues/875
- https://github.com/raspberrypi/linux/issues/2813
- https://github.com/raspberrypi/linux/issues/3855

Motivation: I want to replace a very old RasPi1B off-site backup system with a newer model to improve disk write speed.
This setup is virtually identical to that old one, i.e. root moved to 5T ext 2.5" USB drive w/ single EXT4 partition.

MichaIng, 2022-03-28:
Many thanks for your report.

First of all, the UAS warning is expected and not an issue:
- RPi models up to 3+ by default use the `dwc_otg` USB controller, which does not support scatter-gather and UAS. The kernel module is still automatically loaded and hence the warning printed. It should be possible to mute it via `echo 'blacklist uas' > /etc/modprobe/disable-uas.conf`, it remains unused anyway, so safe to disable completely. This is stated in your links as well.
- Furthermore UAS is a USB 3 feature, while RPi up to 3+ have USB 2 only.
- The individual USB quirks or module blacklisting otherwise is required on RPi 4, where UAS is enabled by default (via `dwc2` USB controller), if drives do not support it well, like for some Seagate drives, loading to various critical issues, the reason why I opened the issue at RPi GitHub.

I'm not too familiar with `vmstat`, these are stats over 3 seconds, showing zero disk I/O? What I'm not sure about is whether filesystem cache writes are included, probably the uploaded data are kept in cache and written in a batch to disk later. At least for filling full filesystem blocks this is definitely done, but may hold larger data as well. And I'm not sure either whether the OpenSSH SFTP server has an internal cache layer. What does the SFTP client show as transfer rate, e.g. when you upload one bigger file?

> enable internal SFTP in OpenSSH

Have you tried to keep using the standalone SFTP server? Shouldn't make much difference, but maybe worth a try. Also I just coincidentally requested upstream to make the internal one the default, to lower SFTP init time and reduce memory usage, and all guides around contain the step to switch to internal-sftp. Feedback so war however was that the standalone one is kept for security reasons, to separate internal memory, just in case there are bugs in the SFTP server code. Quite interesting that the vast majority of guides state to switch to internal-sftp while OpenSSH developers explicitly prefer to keep the standalone version the default, with quite good arguments: https://bugzilla.mindrot.org/show_bug.cgi?id=3397

jcw, 2022-03-28:
Aha, good idea - a single-file SFTP transfer sheds some light on this:

```
$ sftp duppie.fritz.box
Connected to duppie.fritz.box.
sftp> put Ske*
Uploading Sketchnotes 2014.pdf to /home/dietpi/Sketchnotes 2014.pdf
Sketchnotes 2014.pdf                               100%  523MB  10.8MB/s   00:48
sftp>
```

And the corresoponding vmstat lines are:
```
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 2  0      0  44232 179708 709480    0    0     0     0 18492 4057  3  8 89  0  0
 1  0      0  41280 179708 712352    0    0     0     0 37127 12798  7 22 72  0  0
 1  0      0  41464 179716 711656    0    0     0  5471 36518 11695  6 23 63  7  0
 1  1      0  45780 179716 705004    0    0     0  3424 36655 11671  6 23 65  5  0
 2  1      0  43060 179728 707048    0    0     1  9343 26438 7263  9 21 53 18  0
 3  0      0  57336 179728 693628    0    0     0 26720 36282 11553  7 28 47 19  0
 1  0      0  36588 179748 713268    0    0     0 16511 36089 11558  6 27 53 14  0
 1  0      0  38840 179760 711460    0    0     0     9 37785 12839  7 21 70  1  0
 2  1      0  39692 179776 710536    0    0     0 14753 36611 12064  8 24 57 11  0
 3  0      0  39012 179816 712240    0    0     0 24701 36253 11665  7 28 46 19  0
 0  0      0  39516 179816 711836    0    0     0     0 37588 12876  6 22 71  0  0
 3  0      0  39708 179828 711180    0    0     0    11 37491 12418  7 22 70  1  0
 2  0      0  39156 179840 711900    0    0     0  8207 36170 11268  7 24 61  8  0
 1  1      0  43288 179840 707896    0    0     0 18551 35706 11326  7 27 54 12  0
 3  1      0  38196 179856 713132    0    0     0 24753 36450 11519  7 27 47 19  0
 1  0      0  36208 179856 714544    0    0     0  5859 37251 12366  6 25 66  4  0
 0  1      0  39624 179864 710872    0    0     0  1773 30532 8940  5 16 66 13  0
 0  0      0  39624 179872 710872    0    0     0 18597 12608 2219  0  5 80 15  0
 0  0      0  39624 179872 710872    0    0     0     0 8043   14  0  0 100  0  0
 0  0      0  39624 179872 710872    0    0     0     0 8039   13  0  0 100  0  0
```

That's a lot better. On the low end of my expectations, but definitely workable. I'll dig some more.

jcw, 2022-03-28:
The plot thickens. Restarting my backup (based on the brilliant `duplicacy` tool, which I've been using for years), I now see sustained upload rates of 15..17 MB/sec - until it drops back to erratic/slow rates after a few mins of perfect fast uploads:

```
Uploaded chunk 819 size 1292762, 16.93MB/s 13:46:15 0.4%
Uploaded chunk 820 size 5245046, 16.88MB/s 13:48:38 0.4%
Uploaded chunk 821 size 4585883, 16.89MB/s 13:47:45 0.4%
Uploaded chunk 822 size 1633740, 16.90MB/s 13:47:26 0.4%
Uploaded chunk 823 size 1128431, 16.91MB/s 13:47:13 0.4%
```

Here is the vmstat output, around the time things slow down again:

```
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 4  2      0  26244  96120 806544    0    0    13 15329 17903 2784 19 19 46 16  0
 1  1      0  29420  96200 804616    0    0    25 15068 18579 2878 17 20 41 22  0
 2  0      0  24136  96292 810424    0    0    25 12260 17669 2599 17 17 44 21  0
 1  1      0  28952  96356 807088    0    0    21 15804 17688 3212 18 18 43 21  0
 4  1      0  26488  96432 807904    0    0    21 14468 17643 2787 17 19 42 22  0
 1  3      0  25644  96508 809968    0    0    21 17172 18111 2989 19 17 46 17  0
 1  0      0  24232  96544 811580    0    0    12  5504 15162 1988  8 11 70 10  0
 2  0      0  27356  96556 809272    0    0     1  2265 10807  503  3  4 91  3  0
 1  0      0  25528  96556 812408    0    0     0     0 10423  224  3  3 95  0  0
 0  1      0  26236  96572 810156    0    0     1 19552 13719 2303  3  8 72 17  0
 1  0      0  23884  96576 813304    0    0     1  5776 11471  998  2  5 88  5  0
 0  0      0  26080  96584 810908    0    0     0    29 10309  233  2  3 94  0  0
 0  0      0  28584  96596 808640    0    0     1    11 10597  269  3  3 92  2  0
 0  0      0  25684  96596 811804    0    0     0     0 10294  209  3  2 95  0  0
 0  0      0  27340  96604 809628    0    0     0    11 10553  270  3  3 93  1  0
 0  0      0  25244  96604 812668    0    0     0     0 10376  232  2  3 95  0  0
 0  0      0  26136  96616 810196    0    0     1     9 10406  243  3  3 94  1  0
 0  0      0  22812  96632 813724    0    0     3  5628 11687  898  3  5 88  5  0
```

But a manual SFTP at this point is again going through at 11 MB/s.
Even more surprising: restarting the backup is fast again ... but only for 210 seconds. Whaaaa?
So it looks like the problem is on the sending or network side, it's NOT due to DietPi or the RasPi box.

jcw, 2022-03-28:
Hm, perhaps the issue is SFTP, not USB? The backups use 4 threads on the sender, each opening an SFTP session.
I've switched to single-threaded, but the problem persists, again after ≈ 200 seconds of uploads, very puzzling.
Tried external SFTP. Same behaviour: erratic massive slowdown after ≈ 200s. I give up - no idea what's going on!

MichaIng, 2022-03-28:
Nasty, the same works with sustainable transfer rates on RPi 1? Also to rule out USB you could test to upload to SD card, or to not burn it's condition, right into `/dev/null`. I'll see what I get on RPi 2, currently with [Green End SFTP server](https://packages.debian.org/bullseye/gesftpserver) and comparing with OpenSSH SFTP server.

jcw, 2022-03-28:
My current/old RasPi1 backup server only reaches 3 MB/s, but that's with internet links (50 MB/s) in between.
There are more variables I need to check, maybe the 5 TB Seagate is an issue, or the fact that I run the root FS on it.
The other difference is 64-bit on the RPi3 vs 32-bit on RPi1. Perhaps I should try a 32-bit setup on RPi3 as well?
For now (don't laugh), I have added an every-5-min-reboot in cron, since the uploader auto-resumes ...

MichaIng, 2022-03-28:
> For now (don't laugh), I have added an every-5-min-reboot in cron

Hard to hold 😛. I just thought why not an OpenSSH server restart, but indeed such does not kill open connections. Probably:
```sh
pid=$(pgrep -f internal-sftp)
(( $pid )) && kill "$pid"
```
So only open "internal-sftp" processes are killed, if running, which are open SFTP connections. I just verified on our server that those spawn and their PID can be obtained like that.

jcw, 2022-03-28:
Ah - that's _much_ more convenient! The recovery is now just a few seconds as the uploader reconnects.
And, perhaps tellingly: the 15..17 MB/s uploads do indeed resume this way. Could it be an SFTP issue after all?

FWIW, I ended up with this crontab line, since there will be more than one PID going:

```
$ crontab -l
# m h  dom mon dow   command
*/6 * * * * pid=$(pgrep -f internal-sftp); [ -n "$pid" ] && sudo kill $pid
```

It's definitely a nasty workaround. But hey, it keeps my backups going :)
(most of them are incremental and short, but this is the initial backup, of about 2 TB)

Thanks for helping out!

MichaIng, 2022-03-28:
> FWIW, I ended up with this crontab line, since there will be more than one PID going:

Jep, that matches my edited post, looks good.

Indeed it sounds like an SFTP or network issue then 🤔. Probably gesftpserver is worth a try, it can be invoked via "Subsystem" into OpenSSH just like the dedicated OpenSSH SFTP server is by default.

jcw, 2022-03-28:
Another hint? When stuck, i.e. when the internal SFTP server slows down, I can make it resume by logging into SSH again!
IOW, no process kills involved. I've observed this behaviour three times so far. Doesn't solve my use case, but it's very odd.

MichaIng, 2022-03-29:
> All I did to unblock was hit keys a few times in a second ssh session, without hitting return. Crazy stuff.

Hmm, hitting keys produces entropy to fill `/dev/random`. Is the hardware random generator running?
```sh
systemctl status rngd
```

MichaIng, 2022-03-29:
Okay, entropy is not the issue then 🤔.

jcw, 2022-03-29:
I also noticed that the crontab kill trick is not always working. But hey, let's not waste too much time on this. My backup is progressing despite all the hiccups, and tomorrow it should all become much less relevant once it switches to incremental.
I can live with an internal-sftp kill every 6 mins + a daily reboot (assuming no-one else runs into this issue).

It's not an elegant approach, I admit. Something, somewhere, is definitely fishy.

jcw, 2022-03-29:
(I'll just keep adding notes in case it rings any bells - if/when you have suggestions for things to try, just let me know)

```
$ while :; do time sync; sleep 10; done

real	0m17.744s
user	0m0.001s
sys	0m0.027s

[etc...]
```

While working properly, these times fall in the range 6..20s. When blocked, they drop to 1..2s.
That seems to indicate that the data has not been written out, i.e. it's not USB doing the blocking.

The incoming files from the backup app are 2..10 MB each, the backup app reports ≈ 10 MB/s upload rate.
They use a hash as name and are saved in 256 separate dirs, currently ≈ 1800 files each. Nothing extreme, IMO.

MichaIng, 2022-03-29:
Hmm, that is indeed much data kept in filesystem cache. Isn't it actually an indicator that the USB drive is blocking in a way, when it works while data can be kept in filesystem cache and hangs when data is not or cannot be kept in filesystem cache?

So a theory is that, as long as there is sufficient free RAM, the SFTP upload is stored to filesystem cache. As fast as there is no free RAM anymore (free meaning here "used" + cache/buffer), data cannot be cached anymore, and since (or until) it is not written out to disk, the upload hangs, respectively slows down. You could verify this by keeping `htop` open and watch the blue and yellow parts of the memory bar. I'm actually mixing the "buffer" and "cache" terms, also since I do not 100% know which one is relevant here, but both are generally related to caching file I/O to block devices. My guess now is that SFTP upload is fast until the memory bar is filled, and once this has reached, upload slows down.

If that is true, then the question is why the data is not written out to disk, respectively why so slow. One issue issue is that until RPi 3, Ethernet and USB share the same bus, so it makes sense that steady network I/O to USB cannot stay at full speed once actual USB I/O starts. It is discussed here (generally true for RPi 3 as well, but AFAIK the mentioned numbers are not all fully correct): https://raspberrypi.stackexchange.com/questions/45130/why-do-the-usb-ports-and-ethernet-port-share-the-same-controller

I'm not sure about the exact bus vs USB vs Ethernet bandwidth, but in theory SFTP upload could stay at full speed, occupying the bus bandwidth, until RAM is full, then SFTP upload cannot proceed, but instead bandwidth becomes free for USB disk write out, but once that has finished, SFTP upload should proceed again. So it would then be in waves.

jcw, 2022-03-29:
I'll look into this further tomorrow. I understand that the single USB bus limits the disk + network bandwidth, but that's 480 Mbit/sec - I would expect that 15 MB/s going through twice should be easily sustainable. The `vmstat` output above also shows memory use - it doesn't look to me like the stalled writes are related to changes in memory use, but I could be wrong.

Anyway, for me 10 MB/s _sustained_ throughput would be enough, given my 50 Mbit/sec internet uplink speed. That's not the case right now - without the tricks added so far, I do not see things unblocking by themselves, as would happen if it was just a matter of bandwidth and back-pressure / flow-control.

Another puzzle is that `sync` sometimes seems to unblock (or maybe that's just me hitting keys again? pfff ...).

jcw, 2022-03-30:
I haven't progressed with any new details or issues on this. Am currently running with the kill-internal-sftp-every-6-min kludge, and things are holding out well enough to manage my backups from a few different computers around the house.
I'll re-check after substantial system updates.

Am closing this issue again, since by now it doesn't really seem to be specific to DietPi in any way.

closed 2022-03-30 by jcw

MichaIng, 2022-04-02:
Doing some SFTP upload tests as well:
- Server: Raspberry Pi 2 with [gesftpserver](https://packages.debian.org/bookworm/gesftpserver) invoked via Dropbear SSH server
- Client: Virtual machine with [openssh-client](https://packages.debian.org/bullseye/openssh-client); You used the OpenSSH client as well, right?
- Upload to `/dev/null`:
    ```sh
    fallocate -l 10G test
    sftp
    put test /dev/null
    ```
    - Speed stays between 7 - 10 MiB.
    - Buffer and cache are not fulled, as expected.
    - CPU usage on a single core stays at/close to 100%, so it seems in my particular case the CPU is a bottleneck as well? Other SBCs here however do not suffer from the USB/Ethernet bus sharing, so testing those has no point.
- Upload to ext4 filesystem on USB drive:
    ```sh
    put test /mnt/sda/test
    ```
    - Speed stays between 7 - 10 MiB while buffer and cache are filled, as expected. It's actually the yellow bar, which is filled, hence the cache, not the buffer.
    - Good to see is that the swap file remains unused even when all physical RAM is filled by the cache. This makes sense since there is no point to swap the filesystem cache, aimed to reduce disk I/O, back to disk 😉.
    - Here upload speed stays as it was even with full cache, no hanging or slowdown. This shows that indeed the limiting factor on RPi 2 is either the Ethernet port itself, or the CPU, since the additional USB writes do not affect upload speed, compared to writing to `/dev/null`. Our benchmarks show that RPi 2 network speed indeed practically caps at 11 MiB/s. This also matches [other resources](https://notenoughtech.com/raspberry-pi/raspberry-pi-internet-speed/). The RPi 3+ now has a Gbit Ethernet interface, but limited by USB2.0 bus, leading to nearly same upload speeds when writing to a USB drive directly, but at least it shouldn't perform worse.
- Upload to USB drive but with OpenSSH SFTP server:
    - Bandwidth starts a little lower, was below 7 MiB/s at first, peaking 8.1 MiB/s, yeah, it stays a little lower in average.
    - Also CPU usage of the `sftp-server` process is significantly higher. Dropbear, as before, stays at/close to 100% CPU usage, but the `sftp-server` sub process stays around 40% now, while it some something with 15% with the Green End SFTP server. However, it runs on a different core, so shouldn't affect upload speed.
    - Also here, upload speed does not worsen or hang when cache fills the whole physical RAM, i.e. USB drive I/O starts.

I could test with OpenSSH server instead of Dropbear (not just the OpenSSH SFTP server). But will take care the DietPi v8.3 release first.

jcw, 2022-04-06:
Yes, I use OpenSSH on both client (non-RasPi, mostly various Macs, some Linux) and server.

My mass-uploads have ended, all the backup clients are now up to date (sending with SFTP to the RasPi 3B+), so I don't have very lengthy sessions anymore. What I did see today, on the last bigger sends, is that the streaming averaged to over 12 MB/s. Maybe I was simply expecting too much - the RPi3 does have LAN and HD both going through the same USB2 interface after all. And it all being SSH, there is also the encryption loading down the CPU, as you mentioned.

I'm fine with leaving this issue closed. For higher throughputs, something beyond RPi3 will probably work better.

MichaIng, 2022-04-06:
I'm actually not sure how much of the CPU usage really is related to encryption, but possible. Comparisons with NFS and Samba could be done.

> streaming averaged to over 12 MB/s

This was more than 6 minutes? 12 MiB/s consistently would be fine, but that it was going down to zero after buffer/cache was filled is still strange.

jcw, 2022-04-06:
Yes, that last session was averaged over roughly 20 min. Still with the kill-sftp-every-6-min hack installed.