MichaIng/DietPi #5385: NanoPi NEO2 latest Kernel update breaks NanoHatOLED

maartenlangeveld, 2022-03-28:
NanoPi NEO2 (DietPi bullseye)

```
uname -a
Linux [hostname] 5.15.25-sunxi64 #22.02.1 SMP Sun Feb 27 09:24:04 UTC 2022 aarch64 GNU/Linux
```

After latest kernel update my NanoHatOLED application does not start anymore.

```
/home/NanoHatOLED# /bin/sh ./NanoHatOLED
./NanoHatOLED: 1: Syntax error: "(" unexpected
```

Is it possible to revert to previous kernel? How?

MichaIng, 2022-03-28:
Many thanks for your report. More issues with Armbian Linux 5.15 😢. But the NanoHatOLED application may generally require an update for Linux 5.15 and above, hence probably not a bug in the Armbian kernel. Just to assure we're talking about the very same application: It's what you installed from [here](https://github.com/friendlyarm/NanoHatOLED), right?

Can you provide logs of the failing application?

You can downgrade the kernel, but note that this way you reintroduce the [Dirty Pipe vulnerability](https://dietpi.com/blog/?p=1379):
```sh
apt install --reinstall --allow-downgrades 'linux-image-current-sunxi64=21.08.1' 'linux-dtb-current-sunxi64=21.08.1'
apt-mark hold linux-image-current-sunxi64 linux-dtb-current-sunxi64
```

maartenlangeveld, 2022-03-28:
Hi @MichaIng,

Yes, I mean https://github.com/friendlyarm/NanoHatOLED

I will re-think about downgrading.

Thanks, Maarten

MichaIng, 2022-03-28:
Cross linking our issue at FriendlyELEC: #13

It was just an idea that it may require an update, but shouldn't hurt to have FriendlyELEC informed as well as their README explicitly states that it works with Armbian. And they will know better how to debug as well.

Btw, did you already update the application to the latest version by rerunning the [install instructions](https://github.com/friendlyarm/NanoHatOLED#installation)?

maartenlangeveld, 2022-03-28:
Yes I did the update to no avail!

MichaIng, 2022-03-28:
Good. https://github.com/friendlyarm/NanoHatOLED/issues/7 seems to have been solved with a BakeBit submodule update. Latest NanoHatOLED however uses latest BakeBit already, both are developed by FriendlyELEC anyway.

Running `oled-start` does not print any output/errors?

maartenlangeveld, 2022-03-28:
Hi MichaIng,

Application log:

```
open of gpio 0 returned -1: No such file or directory
error opening gpio sysfs entries
```

maartenlangeveld, 2022-03-29:
Is a work around possible?

MichaIng, 2022-03-29:
Okay, GPIO device tunables seem to be missing completely, as well as board specific /proc/cpuinfo. I'm not sure what NanoHatOLED requires exactly, I guess the GPIO tunables are the critical part. I checked currently available device tree overlays for enabling GPIO in general, but there is only `sun50i-h5-gpio-regulator-1.3v.dtbo`, which is present with the Linux 5.10 kernel already. Best is to test with a fresh Armbian Bullseye image: https://www.armbian.com/nanopi-neo-2/
If it fails the same way, report at Armbian: https://forum.armbian.com/forum/27-allwinner-a64-h5-h6-and-h616/
The NanoPi NEO2 is still an officially supported device, i.e. there is an Armbian maintainer for it.

maartenlangeveld, 2022-03-29:
Hi MichaIng,

How do install dietpi when installing fresh Armbian Bullseye image: https://www.armbian.com/nanopi-neo-2/ ?

It will take a lot of working re-installing all applications and customizations. So at this moment not really an option for me. Maybe in future when I have more time.

MichaIng, 2022-03-29:
The idea is indeed to just flash Armbian and verify that NanoHatOLED fails the same way, so that it can be reported to Armbian as a bug. Once you install DietPi, Armbian will say they do not support 3rd party users, reasonably, so that would break the purpose 😉.

maartenlangeveld, 2022-03-29:
Good point, I'll check it this week and let you know.

Thanks a lot for your great help and support!

maartenlangeveld, 2022-03-29:
Have reverted to previous kernel
```
2022-03-29 16:19:19 root@M-DNS:~# uname -a
Linux M-DNS 5.10.60-sunxi64 #21.08.1 SMP Wed Aug 25 18:29:57 UTC 2021 aarch64 GNU/Linux
```
```
2022-03-29 16:19:10 root@M-DNS:~# cat /proc/cpuinfo
processor       : 0
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

maartenlangeveld, 2022-03-29:
I think so.

Result from fresh Armbian installation (https://www.armbian.com/nanopi-neo-2/) are the same:

```
Last login: Tue Mar 29 17:27:35 2022 from 192.-.-.-
root@nanopineo2:~# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

MichaIng, 2022-03-29:
Okay, that is one issue, the other is the missing GPIO sysfs entries. Can you show:
```sh
ls -l /sys/class/gpio
```

maartenlangeveld, 2022-03-29:
output of `ls -l /sys/class/gpio`
```
root@nanopineo2:~# ls -l /sys/class/gpio
ls: cannot access '/sys/class/gpio': No such file or directory
```

But is that not due to NanoHatOLED not running ?

MichaIng, 2022-03-29:
Nope, NanoHatOLED uses the GPIO devices, it does not create them. It fails because the GPIO devices are missing.

maartenlangeveld, 2022-03-29:
Have reported to Armbian forum:

https://forum.armbian.com/topic/20173-armbian-51526-sunxi64-nanopi-neo2-gpio-devices-are-missing-proccpuinfo-lacks-processor-hardware-info-nanohatoled-fails/

MichaIng, 2022-03-29:
Thanks. I'm testing on NanoPi NEO3 now. Different SoC, but need to replicate #5378 anyway.

MichaIng, 2022-03-29:
On NEO3, `/proc/cpuinfo` misses those two keys already with Linux 5.10. But it is a Rockchip SoC, no Allwinner, so probably this is expected. USB 3 flash drive works fine, `/sys/class/gpio` is present with all expected devices.

Lol after kernel upgrade, it doesn't boot anymore, or network does not come up anymore. Nasty with a device without HDMI, and I'm not keen to remove the tight case yet for attaching a serial console.

... next attempt, now `armbian.com` is down, hence I cannot upgrade... Will try again tomorrow.

maartenlangeveld, 2022-03-30:
Yes, problematic when only serial console is left over to connect. Good luck!

MichaIng, 2022-03-31:
On NEO3, the GPIO interface remains. There was a much more critical issue since it became unbootable with the new kernel. We need to switch to a NanoPi R2 device tree to keep it up: https://github.com/MichaIng/DietPi/commit/86726aa

But now it boots and GPIO interface/devices are there. So it seems to be an issue specific to the NEO2 or Allwinner 64-bit kernel.

MichaIng, 2022-03-31:
About your report at the Armbian forum:

> Moved to P2P help.

I do not understand the rules in which actual support is granted by Armbian. The NEO2 is an officially supported board, the kernel upgrade breaks the GPIO interface completely, and they just move it to P2P (users help users), as if you just had a personal issue with your individual setup 🤔. Igor itself is listed as maintainer for this board, now I'm not sure whether he even reads posts in P2P forum. Let's see.

maartenlangeveld, 2022-04-02:
Hi @MichaIng, 
I'm afraid I'll have to stick with Armbian Linux 5.10 kernel for my NanoPi NEO2 as I get the impression the Armbian team does not have the resources and/or does not feel the need to address this GPIO issue.

Since I have marked the latest kernel update as hold back woth APT, the update notification for these packages remains also. Is there a way to suppress the notification for these two packages only?

MichaIng, 2022-04-02:
This should make APT not considering those packages at all:
```sh
cat << '_EOF_' > /etc/apt/preferences.d/ignore-kernel-upgrades
Package: linux-image-current-sunxi64 linux-dtb-current-sunxi64
Pin: *
Pin-Priority: -1
_EOF_
```

MichaIng, 2022-04-03:
See here, it seems to be a deprecation in upstream Linux itself: https://forum.armbian.com/topic/20115-gpio-depreciation-sysfs-on-o-pi-no-sysclassgpio-anymore/

This is now in "obsolete": https://www.kernel.org/doc/Documentation/ABI/obsolete/sysfs-gpio

Looks like a lot of tools/software developers haven't recognised this, neither did I. While the Armbian maintainer is looking into a way to restore it, as it is officially deprecated, it won't last forever. So then the ball is again at FriendlyELEC to move to recent GPIO library, i.e. [libgpiod](https://packages.debian.org/bullseye/libgpiod2). The package text also states that it makes features available, not present in the obsolete sysfs API.

MichaIng, 2022-04-03:
Found a kernel config to restore it, posted in both topics. Hopefully the maintainers can apply this quickly and make a new kernel release 🙂.

maartenlangeveld, 2022-04-08:
Hi MichaIng,

Thanks for your help on this issue on the Armbian forum! If I understood well, the issue is being addressed.

Do don't know how it works but when is the 'repaired' kernel available as package through APT update/upgrade?

Thanks,
Maarten

MichaIng, 2022-04-08:
I don't know when Armbian will release the new kernel, but it shouldn't take so long. Keep a view at the login banner for available APT upgrades, respectively run `apt update` to check for those.

maartenlangeveld, 2022-05-30:
Yesterday evening Armbian 22.05 was released which solves the problem.
@MichaIng many thanks for reporting to the Armbian dev team and your help in identifying and solving this issue !!

MichaIng, 2022-05-30:
That is great news, many thanks for reporting 🙂. I'll mark this as closed then.

closed 2022-05-30 by MichaIng

hoppel118, 2022-07-04:
Hi,

I am new to this and bought a NanoPi Neo (H3). The NanoHat OLED works fine with UbuntuCore 16.04 and kernel 4.14. Now I installed a new microSD card with latest Armbian Bullseye. But the NanoHat doesn't work.

Armbian is up-to-date:

```
uname -a
Linux nanopineo 5.15.48-sunxi #22.05.3 SMP Wed Jun 22 07:35:10 UTC 2022 armv7l GNU/Linux
```

```
cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 48.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: Allwinner sun8i Family
Revision	: 0000
Serial		: 02c000816978def6
```

I followed the installation instructions: [here](https://github.com/friendlyarm/NanoHatOLED#installation) for Armbian Stretch and after that for Arabian Buster. In both cases the NanoPi rebooted automatically and I didn't see any issues while the script was executed.

I don't, if the following information is relevant:

```
gpio readall
piBoardRev: Unable to determine board revision from /proc/cpuinfo
 -> Is not NanoPi based board. 
 ->  You may want to check:
 ->  http://www.lemaker.org/
open /sys/class/sunxi_info/sys_info failed.# 
```

```
cat /tmp/nanohat-oled.log
k1 events: 0
k2 events: 0
k3 events: 0
k1 events: 0
k2 events: 0
k3 events: 0
```

```
ls -l /sys/class/gpio
total 0
--w--w---- 1 root dialout 4096 Jul  5 00:51 export
lrwxrwxrwx 1 root root       0 Jul  5 00:48 gpio0 -> ../../devices/platform/soc/1c20800.pinctrl/gpiochip0/gpio/gpio0
lrwxrwxrwx 1 root root       0 Jul  5 00:48 gpio2 -> ../../devices/platform/soc/1c20800.pinctrl/gpiochip0/gpio/gpio2
lrwxrwxrwx 1 root root       0 Jul  5 00:48 gpio3 -> ../../devices/platform/soc/1c20800.pinctrl/gpiochip0/gpio/gpio3
lrwxrwxrwx 1 root root       0 Jul  5 00:48 gpiochip0 -> ../../devices/platform/soc/1c20800.pinctrl/gpio/gpiochip0
lrwxrwxrwx 1 root root       0 Jul  5 00:48 gpiochip352 -> ../../devices/platform/soc/1f02c00.pinctrl/gpio/gpiochip352
--w--w---- 1 root dialout 4096 Jul  5 00:48 unexport
```

What am I missing? 

Thanks and regards Hoppel

hoppel118, 2022-07-05:
The more often I press the knobs, the more events I see here:

```
cat /tmp/nanohat-oled.log
k1 events: 0
k2 events: 0
k3 events: 0
k3 events: 1
found pid 569
found pid 686
found python pid: 569
found python pid: 686
k2 events: 1
k1 events: 1
k2 events: 1
k1 events: 1
k2 events: 1
k3 events: 1
```

So the buttons work, but the display stays black. 

This is what I see, when I execute the script:

```
sudo -H ./install-compat.sh                                                                                                                                            <master>

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for xxxxxxx: 

Welcome to NanoHatOLED Installer.

Requirements:
1) Must be connected to the internet
2) This script must be run as root user

Steps:
Installs package dependencies:
  - python3       interactive high-level object-oriented language, python3 version
  - BakeBit       an open source platform for connecting BakeBit Sensors to the Pi

Checking Internet Connectivity...
=================================
Connected

Checking User ID...
===================
root

Checking for Updates...
=======================
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]                                                                                              
Get:3 http://deb.debian.org/debian bullseye-backports InRelease [44.2 kB]                                                                                            
Get:4 http://security.debian.org bullseye-security InRelease [44.1 kB]                                                                                               
Hit:6 https://itai-nelken.github.io/weekly-box86-debs/debian  InRelease                                                                                              
Hit:7 http://deb.volian.org/volian scar InRelease                                  
Hit:5 http://mirrors.dotsrc.org/armbian-apt bullseye InRelease
Get:8 http://security.debian.org bullseye-security/main armhf Packages [205 kB]
Fetched 333 kB in 5s (64.8 kB/s) 
Reading package lists... Done

Installing Dependencies
=======================
[... 438 lines cut ...]
  _____  ______  _____ _______       _____ _______ 
 |  __ \|  ____|/ ____|__   __|/\   |  __ \__   __|
 | |__) | |__  | (___    | |  /  \  | |__) | | |   
 |  _  /|  __|  \___ \   | | / /\ \ |  _  /  | |   
 | | \ \| |____ ____) |  | |/ ____ \| | \ \  | |   
 |_|  \_\______|_____/   |_/_/    \_\_|  \_\ |_|   
 
Please restart to implement changes!
To Restart type sudo reboot
To finish changes, we will reboot the Pi.
Pi must reboot for changes and updates to take effect.
If you need to abort the reboot, press Ctrl+C.  Otherwise, reboot!
Rebooting in 5 seconds!
Rebooting in 4 seconds!
Rebooting in 3 seconds!
Rebooting in 2 seconds!
Rebooting in 1 seconds!
Rebooting now!  
Connection to x.x.x.x closed by remote host.
Connection to x.x.x. closed.
```

What can I do to analyse the issue? Any idea?

Thanks Hoppel

hoppel118, 2022-07-07:
Found a solution for Armbian 22.05.04 Bullseye (5.15.48) and described it here:

https://forum.armbian.com/topic/22276-nanopi-neo-h3-nanohat-oled-doesnt-work/#comment-143564

Regards Hoppel

MichaIng, 2022-07-08:
Awesome, many thanks for sharing!