MichaIng/DietPi #5386: Webmin install failed

omavoss, 2022-03-28:
#### Details:
- Date           | Mon Mar 28 23:53:56 CEST 2022
- DietPi version | v8.3.0 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RasPi4 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```
- Date           | Tue Mar 29 00:00:37 CEST 2022
- DietPi version | v8.3.0 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RasPi4 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install webmin`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libauthen-pam-perl libio-pty-perl libnet-ssleay-perl perl-openssl-defaults
The following NEW packages will be installed:
  libauthen-pam-perl libio-pty-perl libnet-ssleay-perl perl-openssl-defaults
  webmin
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 28.6 MB of archives.
After this operation, 304 MB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 perl-openssl-defaults arm64 5 [7360 B]
Get:2 https://deb.debian.org/debian bullseye/main arm64 libnet-ssleay-perl arm64 1.88-3+b1 [316 kB]
Get:3 https://deb.debian.org/debian bullseye/main arm64 libauthen-pam-perl arm64 0.16-3+b8 [27.0 kB]
Get:4 https://deb.debian.org/debian bullseye/main arm64 libio-pty-perl arm64 1:1.15-2 [36.0 kB]
Get:5 https://download.webmin.com/download/repository sarge/contrib arm64 webmin all 1.990 [28.3 MB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 28.6 MB in 4s (7108 kB/s)
Selecting previously unselected package perl-openssl-defaults:arm64.
(Reading database ... 28160 files and directories currently installed.)
Preparing to unpack .../perl-openssl-defaults_5_arm64.deb ...
Unpacking perl-openssl-defaults:arm64 (5) ...
Selecting previously unselected package libnet-ssleay-perl.
Preparing to unpack .../libnet-ssleay-perl_1.88-3+b1_arm64.deb ...
Unpacking libnet-ssleay-perl (1.88-3+b1) ...
Selecting previously unselected package libauthen-pam-perl.
Preparing to unpack .../libauthen-pam-perl_0.16-3+b8_arm64.deb ...
Unpacking libauthen-pam-perl (0.16-3+b8) ...
Selecting previously unselected package libio-pty-perl.
Preparing to unpack .../libio-pty-perl_1%3a1.15-2_arm64.deb ...
Unpacking libio-pty-perl (1:1.15-2) ...
Selecting previously unselected package webmin.
Preparing to unpack .../archives/webmin_1.990_all.deb ...
Unpacking webmin (1.990) ...
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Please help, thank you.

MichaIng, 2022-03-29:
Many thanks for your report.

Is this the complete output? It seems to end in the middle without any error message 🤔. Otherwise, please retry the package install and paste the pull output here:
```sh
apt install webmin
```

Btw, the first error about network time sync that got mixed inside, indicates that you removed `systemd-timesyncd` but didn't change the time sync mode to "custom" via `dietpi-config` > Advanced Options.

omavoss, 2022-03-29:
Hello,

I restoring my DietPi from Backup v. 8.2.2, then I update it to 8.3.0,  then I chanching timesync to custom and here is the output from apt install webmin:

root@RasPi4:~# apt install webmin
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
E: Paket webmin kann nicht gefunden werden.
root@RasPi4:~#

 

Now I executed install webmin via dietpi-software with this error:

 - Command: systemctl stop webmin                                                                                    │
                                           │  - Exit code: 5                                                                                                      │
                                           │  - DietPi version: v8.3.0 (MichaIng/beta) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                                     │
                                           │  - Image creator: DietPi Core Team                                                                                   │
                                           │  - Pre-image: Raspberry Pi OS Lite 64-bit                                                                            │
                                           │  - Error log:                                                                                                        │
                                           │ Failed to stop webmin.service: Unit webmin.service not loaded.                                                       │
                                           │                                                                                                                      │
                                           │                         Retry          : Re-run the last command that failed                                         │
                                           │                         DietPi-Config  : Edit network, APT/NTP mirror settings etc                                   │
                                           │                         Open subshell  : Open a subshell to investigate or solve the issue                           │
                                           │                         Send report    : Uploads bugreport containing system info to DietPi                          │
                                           │                                        ●─ Devs only ──────────────────────────────────────●                          │
                                           │                         Change command : Adjust and rerun the command                                                │
                                           │                                                                                                                      │
                                           │                                                                                                                      │
                                           │                                  <Ok>                                      <Exit>

The output from Putty is here:

#### Details:
- Date           | Tue Mar 29 16:45:06 CEST 2022
- DietPi version | v8.3.0 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux RasPi4 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl stop webmin`
- Exit code      | 5
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to stop webmin.service: Unit webmin.service not loaded.

Now I peformed apt-install webmin with this output from putty:

root@RasPi4:~# apt install webmin
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
webmin ist schon die neueste Version (1.990).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
root@RasPi4:~#

I hope this helps. Many thanks.

Am 29.03.2022 um 13:14 schrieb MichaIng:
>
> Many thanks for your report.
>
> Is this the complete output? It seems to end in the middle without any error message 🤔. Otherwise, please retry the package install and paste the pull output here:
> apt install webmin
>
> Btw, the first error about network time sync that got mixed inside, indicates that you removed systemd-timesyncd but didn't change the time sync mode to "custom" via dietpi-config > Advanced Options.
>
> —
> Reply to this email directly, [view it on GitHub](https://github.com/MichaIng/DietPi/issues/5386#issuecomment-1081740253), or [unsubscribe](https://github.com/notifications/unsubscribe-auth/AEH5UPPEKYXEQL2DDTR3BMTVCLQ2PANCNFSM5R4P5ZTA).
> You are receiving this because you authored the thread.

omavoss, 2022-03-29:
Another machine, similar problem:

#### Details:
- Date           | Tue Mar 29 21:46:07 CEST 2022
- DietPi version | v8.3.0 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | `Linux RasPi1 5.10.103+ #1530 Tue Mar 8 13:00:30 GMT 2022 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl stop webmin`
- Exit code      | 5
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to stop webmin.service: Unit webmin.service not loaded.
```

Joulinar, 2022-03-29:
The error seems to be to stop the service as the but `webmin.service` is not yet loaded. If you are hit be the error, could you select the option `Change command` on the error screen. This should open another window with current command. Simply remove it and type `true`. Afterwards retry.

MichaIng, 2022-03-29:
Verified, a strange one:
```sh
2022-03-29 22:01:54 root@VM-Bullseye:/tmp/DietPi-Software# invoke-rc.d webmin start
Failed to start webmin.service: Unit webmin.service not found.
invoke-rc.d: initscript webmin, action "start" failed.
Unit webmin.service could not be found.
2022-03-29 22:01:59 root@VM-Bullseye:/tmp/DietPi-Software# l /etc/init.d/webmin
-rwxr-xr-x 1 root root 1.9K Mar  3 05:10 /etc/init.d/webmin
```
The `webmin` package install itself fails to start the Webmin service, but all error messages are suppressed, so it cannot be seen. The service is there, it looks like a valid init script, is executable, but when trying to start it, it says "not found" 🤔.

MichaIng, 2022-03-29:
Ah, a simple `systemctl daemon-reload` fixes it. I wouldn't have guessed it is necessary. Let's check what Webmin changed in their postinst script recently.

MichaIng, 2022-03-29:
Basically here is the upstream report: https://github.com/webmin/webmin/issues/1612

MichaIng, 2022-03-29:
Workaround landed for DietPi v8.3: https://github.com/MichaIng/DietPi/commit/dbd4b7c
We could do this in general, but on the other hand if the `webmin` package wouldn't contain the bug, the service did always run anyway.

omavoss, 2022-03-29:
Workaround unsuccessful:

Hmmm...diese Seite ist leider nicht erreichbar192.168.115.35 hat eine Verbindung verweigert.
Versuchen Sie Folgendes:

Die Verbindung wird überprüft…
[Proxy und Firewall überprüfen](chrome-error://chromewebdata/#buttons)
ERR_CONNECTION_REFUSED

omavoss, 2022-03-29:
I hate git, please reopen this comment . . .

MichaIng, 2022-03-29:
I actually missed that you were commenting on the other issue. You don't need to run these `systemctl` commands on DietPi, since we install an own systemd service and start it at the end of `dietpi-software` run anyway. So for you, just skipping the service stop attempt via "Change command" should work. And yes, assure to try connecting via HTTPS, which is used by default by Webmin, reasonably. If it still fails, please paste the output of:
```sh
journalctl -u webmin
```

omavoss, 2022-03-30:
root@RasPi4:~# journalctl -u webmin
-- Journal begins at Wed 2022-03-30 19:17:01 CEST, ends at Wed 2022-03-30 19:53:31 CEST. --
-- No entries --
root@RasPi4:~#

Joulinar, 2022-03-30:
did you tried https already?

`https://<your.IP>:10000`

omavoss, 2022-03-30:
Danke für die freundliche Unterstützung.

Ja, aber alle verwendeten Browser meckern fehlende Zertifikate an.

Ich habe wie Du sicherlich noch weißt, 6 RasPi-Geräte, 4 davon (1x Zero-W, 1 x RasPi1, 1 x RasPi2, 2 x RasPi3) sind mit dem 32bit-Image von Dietpi versehen, der RasPi4 läuft mit dem 64bit-Image. Und genau der RasPi4 macht die Probleme, die anderen 5 nicht.

Bitte schreib hier, welche Konsolenbefehle ich eingeben muss, damit man danach aus den Ausgaben von Putty die richtigen Schlussfolgerungen ziehen kann. Ich kann der Diskussion nicht mehr vollständig folgen.

Joulinar, 2022-03-30:
> Ja, aber alle verwendeten Browser meckern fehlende Zertifikate an.

This is as expected as the Webmin certificate is not a real one. Nothing your need to worry about. Usually there should be an option to accept the `risk` and continue.

omavoss, 2022-03-30:
Danke für alles. Ich habe Webmin auf dem RasPi4 deinstalliert und danach nochmals installiert. Außerdem habe ich die angebotenen Kernel-Updates installiert und nun funktioniert Webmin auch auf dieser Maschine.

MichaIng, 2022-04-02:
Marking this as closed now. Workaround is in place, will be removed again with next Webmin release, where the missing `systemctl daemon-reload` has been added.

closed 2022-04-02 by MichaIng

MichaIng, 2022-04-19:
A new release now has this solved package wise: https://github.com/webmin/webmin/releases/tag/1.991