MichaIng/DietPi #5387: Mariadb upgrade fails.

cocoflan, 2022-03-29:
When upgrading error  mariadb upgrade

[image]

[image]

MichaIng, 2022-03-29:
Many thanks for your report.

Can you please check whether the missing file is really missing?
```sh
ls -l /etc/mysql/mariadb.cnf
dpkg -S /etc/mysql/mariadb.cnf
```
It is shipped with the [mariadb-common](https://packages.debian.org/bullseye/all/mariadb-common/filelist) package itself, hence should be present when the post-installation script is running.

cocoflan, 2022-03-29:
Yes file is missing checked it with the commands.

Joulinar, 2022-03-29:
Does it mean your issue has been solved?

cocoflan, 2022-03-29:
No now nextcloud is giving problems, but i restore again before the updates, so i have to retest the updates with the command, because i forgot the error message on nextcloud.

cocoflan, 2022-03-29:
Now i get this error when trying to open nextcloud web interface:

Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.

Joulinar, 2022-03-29:
Is the database server running?

cocoflan, 2022-03-29:
so i did it again with your command, and at first i get:

Configuration file '/etc/mysql/mariadb.cnf', does not exist on system.
Installing new config file as you requested.
update-alternatives: warning: alternative /etc/mysql/my.cnf.fallback (part of li
nk group my.cnf) doesn't exist; removing from list of alternatives
update-alternatives: warning: forcing reinstallation of alternative /etc/mysql/m
ariadb.cnf because link group my.cnf is broken

Then i did the upgrade and then i got the problem with nextcloud.

cocoflan, 2022-03-29:
ls -l /lost+found
total 0

dmesg -l emerg,alert,crit,err
[    3.163864] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.383901] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.383978] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.411047] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    4.714579] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

journalctl -u mariadb
-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Tue 2022-03-29 19:03:07 CES
T. --
Mar 29 19:01:33 RPI4 systemd[1]: Starting MariaDB 10.3.34 database server...
Mar 29 19:01:33 RPI4 sh[1154]: [108B blob data]
Mar 29 19:01:33 RPI4 sh[1154]: Fatal error in defaults handling. Program aborted
Mar 29 19:01:33 RPI4 mysqld[1162]: [98B blob data]
Mar 29 19:01:33 RPI4 mysqld[1162]: Fatal error in defaults handling. Program abo
rted
Mar 29 19:01:33 RPI4 systemd[1]: mariadb.service: Main p
rocess exited, code=exited, status=1/FAILURE
Mar 29 19:01:33 RPI4 systemd[1]: mariadb.service: Failed
 with result 'exit-code'.
Mar 29 19:01:33 RPI4 systemd[1]: Failed to start MariaDB
 10.3.34 database server.

MichaIng, 2022-03-29:
> Fatal error in defaults handling

I think other files are broken or missing, like here, same error due to a corrupted config file: https://www.plesk.com/kb/support/unable-to-start-mariadb-service-in-a-plesk-server-cant-read-dir-of-etc-my-cnf-errcode-2/

The MariaDB log file hopefully shows which one it is.

cocoflan, 2022-03-29:
systemctl restart mariadb.service
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

cat /var/log/mysql/error.log
cat: /var/log/mysql/error.log: No such file or directory

Joulinar, 2022-03-29:
Could you try to start database 

`sudo -u mysql /usr/sbin/mysqld`

If it did not exist, try following

`sudo -u mysql /usr/sbin/mariadbd`

cocoflan, 2022-03-29:
sudo -u mysql /usr/sbin/mysqld
/usr/sbin/mysqld: Can't read dir of '/etc/mysql/conf.d/' (Errcode: 2 "No such file or directory")
Fatal error in defaults handling. Program aborted

sudo -u mysql /usr/sbin/mariadbd
sudo: /usr/sbin/mariadbd: command not found

cocoflan, 2022-03-29:
ok done i get this now.

sudo -u mysql /usr/sbin/mysqld
2022-03-29 21:39:59 0 [Note] /usr/sbin/mysqld (mysqld 10.3.34-MariaDB-0+deb10u1) starting as process 5860 ...

sudo -u mysql /usr/sbin/mariadbd
sudo: /usr/sbin/mariadbd: command not found

MichaIng, 2022-03-29:
Looks good, please terminate this mysqld process and restart the service:
```sh
killall mysqld
systemctl restart mariadb
sleep 5
systemctl status mariadb
```

cocoflan, 2022-03-29:
killall mysqld
mysqld: no process found

systemctl restart mariadb
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

sleep 5

systemctl status mariadb
● mariadb.service - MariaDB 10.3.34 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: 
enabled)
   Active: failed (Result: exit-code) since Tue 2022-03-29 21:43:24 C
EST; 42s ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
  Process: 9864 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/ru
n/mysqld (code=exited, status=0/SUCCESS)
  Process: 9865 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START
_POSITION (code=exited, status=0/SUCCESS)
  Process: 9872 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR
= ||   VAR=`cd /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemc
tl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/S
UCCESS)
  Process: 9932 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WS
REP_START_POSITION (code=exited, status=2)
 Main PID: 9932 (code=exited, status=2)
   Status: "MariaDB server is down"

Mar 29 21:43:21 RPI4 systemd[1]: Starting MariaDB 10.3.34 database server...
Mar 29 21:43:22 RPI4 mysqld[9932]: 2022-03-29 21:43:22 0 [Note] /usr/sbin/mysqld
 (mysqld 10.3.34-MariaDB-0+deb10u1) starting as process 9932 ...
Mar 29 21:43:24 RPI4 systemd[1]: mariadb.service: Main p
rocess exited, code=exited, status=2/INVALIDARGUMENT
Mar 29 21:43:24 RPI4 systemd[1]: mariadb.service: Failed
 with result 'exit-code'.
Mar 29 21:43:24 RPI4 systemd[1]: Failed to start MariaDB
 10.3.34 database server.

MichaIng, 2022-03-29:
More broken fails. Whenever it fails, check and add directory the content of:
```sh
cat /var/log/mysql/error.log
```

cocoflan, 2022-03-29:
cat /var/log/mysql/error.log
2022-03-29 21:37:50 0 [Note] InnoDB: Using Linux native AIO
2022-03-29 21:37:50 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2022-03-29 21:37:50 0 [Note] InnoDB: Uses event mutexes
2022-03-29 21:37:50 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-03-29 21:37:50 0 [Note] InnoDB: Number of pools: 1
2022-03-29 21:37:50 0 [Note] InnoDB: Using generic crc32 instructions
2022-03-29 21:37:50 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2022-03-29 21:37:50 0 [Note] InnoDB: Completed initialization of buffer pool
2022-03-29 21:37:50 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2022-03-29 21:37:50 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2022-03-29 21:37:50 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2022-03-29 21:37:50 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-03-29 21:37:50 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-03-29 21:37:50 0 [Note] InnoDB: Waiting for purge to start
2022-03-29 21:37:50 0 [Note] InnoDB: 10.3.34 started; log sequence number 1746415803; transaction id 8334273
2022-03-29 21:37:50 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/SSD250ARM/dietpi_userdata/mysql/ib_buffer_pool
2022-03-29 21:37:50 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-03-29 21:37:50 0 [ERROR] /usr/sbin/mysqld: unknown option '--“”'
2022-03-29 21:37:50 0 [ERROR] Aborting

2022-03-29 21:37:58 0 [Note] InnoDB: Using Linux native AIO
2022-03-29 21:37:58 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2022-03-29 21:37:58 0 [Note] InnoDB: Uses event mutexes
2022-03-29 21:37:58 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-03-29 21:37:58 0 [Note] InnoDB: Number of pools: 1
2022-03-29 21:37:58 0 [Note] InnoDB: Using generic crc32 instructions
2022-03-29 21:37:58 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2022-03-29 21:37:58 0 [Note] InnoDB: Completed initialization of buffer pool
2022-03-29 21:37:58 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2022-03-29 21:37:58 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2022-03-29 21:37:58 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2022-03-29 21:37:58 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-03-29 21:37:58 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-03-29 21:37:58 0 [Note] InnoDB: 10.3.34 started; log sequence number 1746415812; transaction id 8334273
2022-03-29 21:37:58 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/SSD250ARM/dietpi_userdata/mysql/ib_buffer_pool
2022-03-29 21:37:58 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-03-29 21:37:58 0 [ERROR] /usr/sbin/mysqld: unknown option '--“”'
2022-03-29 21:37:58 0 [ERROR] Aborting

2022-03-29 21:39:00 0 [Note] InnoDB: Using Linux native AIO
2022-03-29 21:39:00 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2022-03-29 21:39:00 0 [Note] InnoDB: Uses event mutexes
2022-03-29 21:39:00 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-03-29 21:39:00 0 [Note] InnoDB: Number of pools: 1
2022-03-29 21:39:00 0 [Note] InnoDB: Using generic crc32 instructions
2022-03-29 21:39:00 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2022-03-29 21:39:00 0 [Note] InnoDB: Completed initialization of buffer pool
2022-03-29 21:39:00 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2022-03-29 21:39:00 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2022-03-29 21:39:00 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2022-03-29 21:39:00 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-03-29 21:39:00 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-03-29 21:39:00 0 [Note] InnoDB: 10.3.34 started; log sequence number 1746415821; transaction id 8334273
2022-03-29 21:39:00 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/SSD250ARM/dietpi_userdata/mysql/ib_buffer_pool
2022-03-29 21:39:00 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-03-29 21:39:00 0 [ERROR] /usr/sbin/mysqld: unknown option '--“”'
2022-03-29 21:39:00 0 [ERROR] Aborting

2022-03-29 21:39:59 0 [Note] InnoDB: Using Linux native AIO
2022-03-29 21:39:59 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2022-03-29 21:39:59 0 [Note] InnoDB: Uses event mutexes
2022-03-29 21:39:59 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-03-29 21:39:59 0 [Note] InnoDB: Number of pools: 1
2022-03-29 21:39:59 0 [Note] InnoDB: Using generic crc32 instructions
2022-03-29 21:39:59 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2022-03-29 21:39:59 0 [Note] InnoDB: Completed initialization of buffer pool
2022-03-29 21:39:59 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2022-03-29 21:39:59 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
[... cut ...]
2022-03-29 21:43:22 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-03-29 21:43:22 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-03-29 21:43:22 0 [Note] InnoDB: 10.3.34 started; log sequence number 1746415839; transaction id 8334273
2022-03-29 21:43:22 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/SSD250ARM/dietpi_userdata/mysql/ib_buffer_pool
2022-03-29 21:43:22 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-03-29 21:43:22 0 [ERROR] /usr/sbin/mysqld: unknown option '--“”'
2022-03-29 21:43:22 0 [ERROR] Aborting

MichaIng, 2022-03-29:
> [ERROR] /usr/sbin/mysqld: unknown option '--“”'

Can you check:
```sh
/usr/bin/galera_recovery
systemctl cat mariadb
```

cocoflan, 2022-03-29:
# /lib/systemd/system/mariadb.service
# It's not recommended to modify this file in-place, because it will be
# overwritten during package upgrades.  If you want to customize, the
# best way is to create a file "/etc/systemd/system/mariadb.service",
# containing
#	.include /usr/lib/systemd/system/mariadb.service
#	...make your changes here...
# or create a file "/etc/systemd/system/mariadb.service.d/foo.conf",
# which doesn't need to include ".include" call and which will be parsed
# after the file mariadb.service itself is parsed.
#
# For more info about custom unit files, see systemd.unit(5) or
# https://mariadb.com/kb/en/mariadb/systemd/
#
# Copyright notice:
#
# This file is free software; you can redistribute it and/or modify it
# under the terms of the GNU Lesser General Public License as published by
# the Free Software Foundation; either version 2.1 of the License, or
# (at your option) any later version.

[Unit]
Description=MariaDB 10.3.34 database server
Documentation=man:mysqld(8)
Documentation=https://mariadb.com/kb/en/library/systemd/
After=network.target

[Install]
WantedBy=multi-user.target
Alias=mysql.service
Alias=mysqld.service

[Service]

##############################################################################
## Core requirements
##

Type=notify

# Setting this to true can break replication and the Type=notify settings
# See also bind-address mysqld option.
PrivateNetwork=false

##############################################################################
## Package maintainers
##

User=mysql
Group=mysql

# To allow memlock to be used as non-root user if set in configuration
CapabilityBoundingSet=CAP_IPC_LOCK

# Prevent writes to /usr, /boot, and /etc
ProtectSystem=full

# Doesn't yet work properly with SELinux enabled
# NoNewPrivileges=true

PrivateDevices=true

# Prevent accessing /home, /root and /run/user
ProtectHome=true

# Execute pre and post scripts as root, otherwise it does it as User=
PermissionsStartOnly=true

ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld

[... 243 lines cut ...]
# Number of files limit. previously [mysqld_safe] open-files-limit
LimitNOFILE=32768

# Maximium core size. previously [mysqld_safe] core-file-size
# LimitCore=

# Nice priority. previously [mysqld_safe] nice
# Nice=-5

# Timezone. previously [mysqld_safe] timezone
# Environment="TZ=UTC"

# Library substitutions. previously [mysqld_safe] malloc-lib with explicit paths
# (in LD_LIBRARY_PATH) and library name (in LD_PRELOAD).
# Environment="LD_LIBRARY_PATH=/path1 /path2" "LD_PRELOAD=

# Flush caches. previously [mysqld_safe] flush-caches=1
# ExecStartPre=sync
# ExecStartPre=sysctl -q -w vm.drop_caches=3

# numa-interleave=1 equalivant
# Change ExecStart=numactl --interleave=all /usr/sbin/mysqld......

# crash-script equalivent
# FailureAction=

MichaIng, 2022-03-29:
So `/usr/bin/galera_recovery` returns an empty string as expected. Then I have no idea where this `--“”` command line argument shall come from. The other two variables are set nowhere.

Let's reinstall every MariaDB package:
```sh
apt install --reinstall mariadb-server mariadb-server-10.3 galera-3 mariadb-client-10.3 mariadb-server-core-10.3 mariadb-client-core-10.3 libconfig-inifiles-perl
```

cocoflan, 2022-03-29:
apt install --reinstall mariadb-server mariadb-server-10.3 galera-3 mariadb-client-10.3 mariadb-server-core-10.3 mariadb-client-core-10.3 libconfig-inifiles-perl
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 7 reinstalled, 0 to remove and 0 not upgraded.
Need to get 17.0 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main arm64 mariadb-server-core-10.3 arm64 1:10.3.34-0+deb10u1 [5579 kB]
Get:2 https://deb.debian.org/debian buster/main arm64 mariadb-server-10.3 arm64 1:10.3.34-0+deb10u1 [4032 kB]
Get:3 https://deb.debian.org/debian buster/main arm64 galera-3 arm64 25.3.25-2 [792 kB]
Get:4 https://deb.debian.org/debian buster/main arm64 libconfig-inifiles-perl all 3.000001-1 [51.9 kB]
Get:5 https://deb.debian.org/debian buster/main arm64 mariadb-client-10.3 arm64 1:10.3.34-0+deb10u1 [1136 kB]
Get:6 https://deb.debian.org/debian buster/main arm64 mariadb-client-core-10.3 arm64 1:10.3.34-0+deb10u1 [5426 kB]
Get:7 https://deb.debian.org/debian buster/main arm64 mariadb-server all 1:10.3.34-0+deb10u1 [32.0 kB]
Fetched 17.0 MB in 2s (7989 kB/s)         
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 59224 files and directories currently installed.)
Preparing to unpack .../0-mariadb-server-core-10.3_1%3a10.3.34-0+deb10u1_arm64.d
eb ...
Unpacking mariadb-server-core-10.3 (1:10.3.34-0+deb10u1) over (1:10.3.34-0+deb10
u1) ...
Preparing to unpack .../1-mariadb-server-10.3_1%3a10.3.34-0+deb10u1_arm64.deb ..
.
/var/lib/mysql: found previous version 10.3
Unpacking mariadb-server-10.3 (1:10.3.34-0+deb10u1) over (1:10.3.34-0+deb10u1) .
..
Preparing to unpack .../2-galera-3_25.3.25-2_arm64.deb ...
Unpacking galera-3 (25.3.25-2) over (25.3.25-2) ...
Preparing to unpack .../3-libconfig-inifiles-perl_3.000001-1_all.deb ...
Unpacking libconfig-inifiles-perl (3.000001-1) over (3.000001-1) ...
Preparing to unpack .../4-mariadb-client-10.3_1%3a10.3.34-0+deb10u1_arm64.deb ..
.
Unpacking mariadb-client-10.3 (1:10.3.34-0+deb10u1) over (1:10.3.34-0+deb10u1) .
..
Preparing to unpack .../5-mariadb-client-core-10.3_1%3a10.3.34-0+deb10u1_arm64.d
eb ...
Unpacking mariadb-client-core-10.3 (1:10.3.34-0+deb10u1) over (1:10.3.34-0+deb10
u1) ...
Preparing to unpack .../6-mariadb-server_1%3a10.3.34-0+deb10u1_all.deb ...
Unpacking mariadb-server (1:10.3.34-0+deb10u1) over (1:10.3.34-0+deb10u1) ...
Setting up libconfig-inifiles-perl (3.000001-1) ...
Setting up mariadb-server-core-10.3 (1:10.3.34-0+deb10u1) ...
Setting up galera-3 (25.3.25-2) ...
Setting up mariadb-client-core-10.3 (1:10.3.34-0+deb10u1) ...
Setting up mariadb-client-10.3 (1:10.3.34-0+deb10u1) ...
Setting up mariadb-server-10.3 (1:10.3.34-0+deb10u1) ...
Job for mariadb.service failed because the control process exited with error cod
e.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
Setting up mariadb-server (1:10.3.34-0+deb10u1) ...
Processing triggers for systemd (241-7~deb10u8) ...

MichaIng, 2022-03-29:
```
cat /var/log/mysql/error.log
```
Same error about the invalid option as before? If so, probably we misinterpret it and it isn't an actual command-line argument but internally added from a config file. So let's check all of them:
```sh
find /etc/mysql -type f ! -name 'debian-start' -exec ls -l {} \; -exec cat {} \;
```

Which device/SBC is it actually?

cocoflan, 2022-03-29:
cat /var/log/mysql/error.log
2022-03-29 22:53:39 0 [ERROR] /usr/sbin/mysqld: unknown option '--“”'
2022-03-29 22:53:39 0 [ERROR] Aborting

2022-03-29 22:53:43 0 [Note] InnoDB: Using Linux native AIO
2022-03-29 22:53:43 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2022-03-29 22:53:43 0 [Note] InnoDB: Uses event mutexes
2022-03-29 22:53:43 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2022-03-29 22:53:43 0 [Note] InnoDB: Number of pools: 1
2022-03-29 22:53:43 0 [Note] InnoDB: Using generic crc32 instructions
2022-03-29 22:53:43 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2022-03-29 22:53:43 0 [Note] InnoDB: Completed initialization of buffer pool
2022-03-29 22:53:43 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2022-03-29 22:53:44 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2022-03-29 22:53:44 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2022-03-29 22:53:44 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2022-03-29 22:53:44 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2022-03-29 22:53:44 0 [Note] InnoDB: 10.3.34 started; log sequence number 1746415857; transaction id 8334273
2022-03-29 22:53:44 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/SSD250ARM/dietpi_userdata/mysql/ib_buffer_pool
2022-03-29 22:53:44 0 [Note] Plugin 'FEEDBACK' is disabled.
2022-03-29 22:53:44 0 [ERROR] /usr/sbin/mysqld: unknown option '--“”'
2022-03-29 22:53:44 0 [ERROR] Aborting

find /etc/mysql -type f ! -name 'debian-start' -exec ls -l {} \; -exec cat {} \;
-rw------- 1 root root 277 Dec  5 23:28 /etc/mysql/debian.cnf
# Automatically generated for Debian scripts. DO NOT TOUCH!
[client]
host     = localhost
user     = root
password = 
socket   = /var/run/mysqld/mysqld.sock
[mysql_upgrade]
host     = localhost
user     = root
password = 
socket   = /var/run/mysqld/mysqld.sock
basedir  = /usr
-rw-r--r-- 1 root root 336 May 17  2021 /etc/mysql/mariadb.conf.d/50-mysql-clients.cnf
#
# These groups are read by MariaDB command-line tools
# Use it for options that affect only one utility
#

[mysql]
# Default is Latin1, if you need UTF-8 set this (also in server section)
default-character-set = utf8mb4

[mysql_upgrade]

[mysqladmin]

[mysqlbinlog]

[mysqlcheck]

[mysqldump]

[mysqlimport]

[mysqlshow]

[mysqlslap]
-rw-r--r-- 1 root root 3940 May 17  2021 /etc/mysql/mariadb.conf.d/50-server.cnf
#
# These groups are read by MariaDB server.
# Use it for options that only the server (but not clients) should see
#
# See the examples of server my.cnf files in /usr/share/mysql

# this is read by the standalone daemon and embedded servers
[... 185 lines cut ...]
#
# The MariaDB/MySQL tools read configuration files in the following order:
# 1. "/etc/mysql/mariadb.cnf" (this file) to set global defaults,
# 2. "/etc/mysql/conf.d/*.cnf" to set global options.
# 3. "/etc/mysql/mariadb.conf.d/*.cnf" to set MariaDB-only options.
# 4. "~/.my.cnf" to set user-specific options.
#
# If the same option is defined multiple times, the last one will apply.
#
# One can use all long options that the program supports.
# Run program with --help to get a list of available options and with
# --print-defaults to see which it would actually understand and use.

#
# This group is read both both by the client and the server
# use it for options that affect everything
#
[client-server]

# Import all .cnf files from configuration directory
!includedir /etc/mysql/conf.d/
!includedir /etc/mysql/mariadb.conf.d/
“”

It is a Atom based intel machine.

cocoflan, 2022-03-29:
I can always go back with my backup and skip the 6 upgrades for the moment, i just wanted to inform you of the problem.

Joulinar, 2022-03-29:
Problem looks like dedicated to your. 

Could you have a look into file `/etc/mysql/mariadb.cnf` and probable remove the last line containing the double quote `“”`. Or simply hash it. Once done restart MariaDB

cocoflan, 2022-03-29:
Yes it is fixed, how that "" came there no clue.

cocoflan, 2022-03-29:
Thanks for all your help !

Joulinar, 2022-03-29:
well `/etc/mysql/mariadb.cnf` was the file where your issue was starting with, as it was missing and your package update failed on. Maybe it has been created incorrectly by recreating missing configs? 🤔

cocoflan, 2022-03-29:
Yes maybe further investigation is needed. I did nothing special on that system that is running for years, and correction it is a RPI4 where the problem was sorry.

MichaIng, 2022-03-30:
Very strange with the `“”`, this file is just extracted form the checksum/hashed checked package, nothing else 🤔. I'm still thinking there was some filesystem corruption going on, but nothing visible in `/lost+found` and `dmesg`.

`/etc/mysql/mariadb.cnf` btw is fully correct without `“”`, matching the package content all MariaDB servers I have access to.

cocoflan, 2022-03-30:
I am using F2FS on one of the disks on that RPI4, but it is only used to use dietpi-sync not as the user_data location that location is an ext4 ssd disk.

MichaIng, 2022-03-31:
MariaDB is stored in the system drive, hence this shouldn't matter. Also there are no issues known with F2FS, even when using it as system drive 😉.

I'll mark this as closed now, as the issue has been solved. Feel free to reopen when required.

closed 2022-03-31 by MichaIng