MichaIng/DietPi #5388: NTP Time checking every 15mins, although dietpi-config set to check on boot only

pied-stuff, 2022-03-29:
Pihole log is showing updatets to NTP server every 15mins, although dietpi-config set to check on boot only

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
bullseye 1
Linux pi 5.10.103+ #1530 Tue Mar 8 13:00:30 GMT 2022 armv6l GNU/Linux
`
2022-03-29 11:28:41 | A | time.cloudflare.com | 192.168.1.9 | OK, answered by localhost#5335 | IP (3.4ms)
2022-03-29 11:10:43 | A | time.cloudflare.com | 192.168.1.9 | OK, answered by localhost#5335 | IP (3.1ms) 
2022-03-29 10:55:35 | A | time.cloudflare.com | 192.168.1.9 | OK, answered by localhost#5335 | IP (1.4ms)
`

Joulinar, 2022-03-29:
What device IP address `192.168.1.9 ` is?

pied-stuff, 2022-03-29:
RPi Zero W (armv6l)

Joulinar, 2022-03-29:
Did you set cloutflare as NTP server inside dietpi settings?

MichaIng, 2022-03-29:
To check the NTP mirror which the DietPi cron job and systemd-timesyncd use:
```sh
cat /etc/systemd/timesyncd.conf
```
This is not done by our cron job (there is only an hourly and a daily one) but by some other process. Since you use a local DNS server or proxy, is it probably `cloudflared` which may internally do time syncs to assure being in sync with Cloudflare servers (where it is usually used as tunnel to enable web services through Cloudflare, not as DNS proxy).

closed 2022-04-14 by MichaIng