MichaIng/DietPi #5391: Enable WiFi and Ethernet adapters on first boot

matteocarnelos, 2022-03-30:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

The need of this feature comes from an issue I've already posted [here](https://dietpi.com/phpbb/viewtopic.php?t=9126), so the need to have both interfaces enabled during first boot to set up a WiFi Hotspot using DietPi Automation. 
Unfortunately, due to some recent updates, the workaround that I posted there does not work anymore, as the update requires a reboot and the "disable wifi" setting takes effect. Therefore, the installation fails since `wlan0` is not detected.

To reproduce this issue you just have to perform a fresh install of the latest DietPi, setting the following in the `dietpi.txt` file:
```
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_INSTALL_SOFTWARE_ID=60
```

#### Describe the solution you'd like:

I would like that, in `dietpi.txt`, the following settings:
```
AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=1
```
Would actually enable both adapters. Right now, the settings above are equal to enabling only the WiFi, as it is stated in the comment next to those settings.

I think that letting the user enable both adapters instead of forcing him to have only one enabled gives more flexibility to DietPi and solves the issue that I'm having. However, I don't know if this feature may break some other components or functionalities. From my searches made so far it should work just fine, although some settings might need to be modified. For instance, the `AUTO_SETUP_NET_USESTATIC` and related settings will probably need to be duplicated (e.g. `AUTO_SETUP_NET_WIFI_USESTATIC` and `AUTO_SETUP_ETH_USESTATIC`).

#### Describe alternatives you've considered:

- The installation of WiFi Hotspot enables the onboard wifi, reboots the system, and than continues with the installation.
- New workaround: connect to internet using WiFi on first boot (so the adapter does not get disabled) and then enable the Ethernet adapter in the `Automation_Custom_Script.sh`. However, enabling the Ethernet adapter is not as easy as enabling the WiFi one, I couldn't find any function that does the job.
- New workaround: do not use DietPi Automation to install WiFi Hotspot and make a custom script that, once DietPi is all set up, enables the adapters, reboots the system and install the WiFi Hotspot using `dietpi-software install 60`.

#### Additional context

I think it is necessary to modify just the [dietpi-firstboot.bash](https://github.com/MichaIng/DietPi/blob/master/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash) file. I can try to submit a PR if you tell me that it's something doable and useful 😉

---
Thanks!

matteocarnelos, 2022-03-31:
I've just tested those settings on a Raspberry Pi 3 B+ using the last DietPi image (ARMv8) and it doesn't work.
As far as I understood here's what happens:
- `AUTO_SETUP_NET_WIFI_ENABLED=0` disables the wifi adapter of the raspberry. However, this setting doesn't take effect until a reboot
- DietPi-Update updates the dietpi version from 8.2.1 to 8.2.2 and it reboots the system. Here the setting above takes effect and the wifi adapter of the raspberry is fully disabled
- The installation of WiFi Hotspot enables the wifi modules using `/boot/dietpi/func/dietpi-set_hardware wifimodules enable` but it doesn't enable the onboard wifi of the raspberry (using `/boot/dietpi/func/dietpi-set_hardware wifimodules onboard_enable`). Therefore, there's still no enabled wifi adapter
- The installation of WiFi Hotspot gets the wifi interface (`local wifi_iface=$(G_GET_NET -t wlan iface)`) but it is empty since no adapter is enabled
- The command `ip a add 192.168.42.1 dev "$wifi_iface"` fails because `wifi_iface` is empty and it throws the following error page

---

#### Details:
- Date           | Thu Mar 31 20:15:34 CEST 2022
- DietPi version | v8.2.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.15.30-v8+ #1536 SMP PREEMPT Mon Mar 28 13:53:14 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ip a add 192.168.42.1 dev `
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Cannot find device ""
```

---

Here's also the last part of the log, from here you can clearly see that `wlan0` wasn't detected:
```
[ SUB1 ] DietPi-Set_hardware > wifimodules (enable)
[ INFO ] DietPi-Set_hardware | Please wait, enabling WiFi modules...
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: iw wireless-tools crda wpasupplicant
[  OK  ] wifimodules enable | Completed
[FAILED] DietPi-Software | An interface of type "wlan" does not exist.
[  OK  ] DietPi-Software | /etc/dhcp/dhcpd.conf: backup to /etc/dhcp/dhcpd.conf.bak_310322_0
[ INFO ] DietPi-Software | For a full list of backup items, please see /var/tmp/dietpi/logs/G_BACKUP_FP.db
[  OK  ] DietPi-Software | sed -Ei /(allow-hotplug|auto)[[:blank:]]+wlan/q0 /etc/network/interfaces
[  OK  ] DietPi-Software | Comment in /etc/network/interfaces converted to setting: allow-hotplug
[FAILED] DietPi-Software | ip a add 192.168.42.1 dev
```

MichaIng, 2022-04-02:
> DietPi-Update updates the dietpi version from 8.2.1 to 8.2.2 and it reboots the system.

Ah, that is the key here, many thanks for pointing that out. That is indeed new since DietPi v8.2, if a kernel upgrade was applied during DietPi update, and it is generally reasonable since without a reboot in this case other installs would fail, e.g. Docker and everything which makes use of nftables (directly or indirectly via iptables).

I'll do the following:
- On first boot, when WiFi modules are disabled, on RPi, onboard WiFi is kept enabled.
- Only at the end of first run installs, if WiFi modules are still disabled, i.e. no install option or manual config re-enabled them, onboard WiFi is marked to be disabled from next reboot on as well.

MichaIng, 2022-04-02:
Solved with: https://github.com/MichaIng/DietPi/commit/f917321

closed 2022-04-02 by MichaIng

matteocarnelos, 2022-04-02:
Thank you very much for the fix! I'll test it in these days😉

MichaIng, 2022-04-04:
Yes exactly. I'll update our images.

matteocarnelos, 2022-04-10:
Hey @MichaIng, are the images updated? The `DietPi_RPi-ARMv8-Bullseye` image is still based on the 8.1.2 version.

MichaIng, 2022-04-10:
I just cleared that one from the Cloudflare cache, so now you should get the v8.3 based one.

matteocarnelos, 2022-04-10:
Thank you very much!