MichaIng/DietPi #5395: Dietpi-TimeSync is missing

eddiermar, 2022-04-01:
### Creating a bug report/issue

Hi there. Today i updated my system, then i received a message that "apt autoremove" should be executed so I did it. After this I went to dietpi-cleaner and everything worked ok. But when i tryed to use dietpi-update got this message:

Failed to start systemd-timesyncd.service

#### Required Information
- DietPi version | 
`G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | bullseye 1
- Kernel version | Linux lucky1 5.10.103+ #1530 Tue Mar 8 13:00:30 GMT 2022 armv6l GNU/Linux
- SBC model | RPi B+ (armv6l)
- Power supply used | 5V 2A
- SD card used | Adata 16GB

#### Steps to reproduce

1. Run "apt update && apt upgrade"
2. Run "apt autoremove"
3. Go to ditepi-launcher --> dietpi-cleaner -->run
4. Go to dietpi-update --> run
5. Got this message:
[image]

6. Then this message appears:
#### Details:
- Date           | Fri Apr  1 10:11:28 -05 2022
- DietPi version | v8.2.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi B+ (armv6l) (ID=1)
- Kernel version | `Linux lucky1 5.10.103+ #1530 Tue Mar 8 13:00:30 GMT 2022 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync

#### Expected behaviour

- Just update without errors

#### Actual behaviour

- Dietpi-update stops

#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.
```

Joulinar, 2022-04-01:
Did you use a different time sync service or did you recall that you removed it? If using an own time syn service, you would need to disable ours using `dietpi-config`

MichaIng, 2022-04-01:
It's not removed, but masked 🤔:
```sh
systemctl unmask systemd-timesyncd
```

In case you installed something (there is no software option in `dietpi-software` which would do) which provides time sync functionality and in turn masked `systemd-timesyncd`, it would make sense to switch to "Custom" time sync mode via `dietpi-config` > Advanced Options, instead of unmasking it.

eddiermar, 2022-04-02:
Hi again... I have checked and found that `apt autoremove` had deleted `systemd-timesyncd` so I then reinstalled it via `apt install systemd-timesyncd` Rebooted my system and everything is ok again.

Thanks!

MichaIng, 2022-04-02:
Did you upgrade the system from Buster to Bullseye or was it a Bullseye image in the first place? `systemd-timesyncd` from Bullseye on is pre-installed and marked as "manually installed", hence shouldn't be autoremoved. But in case of upgraded systems, it may happen.

eddiermar, 2022-04-02:
> Did you upgrade the system from Buster to Bullseye or was it a Bullseye image in the first place? `systemd-timesyncd` from Bullseye on is pre-installed and marked as "manually installed", hence shouldn't be autoremoved. But in case of upgraded systems, it may happen.

Interesting... I have upgraded my system from buster to bullseye last year and then I run `apt update && apt upgrade` once a week... This was the first time that `apt autoremove` deleted  `systemd-timesyncd`... Just for your information!

MichaIng, 2022-04-02:
Okay, then it makes sense. In our migration guide, in the meantime, we have the suggested command inside to install (which implies the "manual" mark) `systemd-timesyncd`: https://dietpi.com/blog/?p=811#2.2.4-software-migrations
But, that is not 1 year old yet and also the command wasn't inside since the beginning, AFAIK.

So yeah, generally it is one thing that needs to be taken care on a distro upgrade: The package didn't exist in Buster, but was part of `systemd` itself. With Bullseye it became a dedicated package, and generally can hence be autoremoved. And additionally, at first any time sync daemon was a dependency of `systemd`, preventing the autoremoval unless another time sync daemon was installed, but the changed just two weeks ago: https://metadata.ftp-master.debian.org/changelogs//main/s/systemd/systemd_247.3-7_changelog
```
  * Demote systemd-timesyncd from Depends to Recommends.
    This avoids a dependency cycle between systemd and systemd-timesyncd and
    thus makes dist upgrades more predictable and robust.
    It also allows minimal, systemd based containers where no NTP client is
    strictly necessary.
    To ensure that systemd-timesyncd is installed in a default installation
    created by d-i, bump its priority to standard.
```

MichaIng, 2022-04-02:
Marking this as closed then. Issue understood, but not a problem with new images and not when following our upgrade guide.

closed 2022-04-02 by MichaIng