MichaIng/DietPi #5396: Nukkit not install correctly

richardpdunngb, 2022-04-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title Nukkit
- Was the software title installed freshly . new install on fresh dietpi setup
- Can this issue be replicated on a fresh installation of DietPi?

Reference code: a7b6c5de-5a8c-4660-9b27-094616e89a69
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...fresh install of dietpi
2. ...installed mc
3. installed nukkit

#### Expected behaviour

- ...nukkit should install and run. 

#### Actual behaviour

- only two files are coppied, nukkit.jar & nukkit.yml
- nothing else installed or in /use/local/bin/nukkit folder

#### Extra details

- ...

MichaIng, 2022-04-02:
It is expected that only those two files are in this directory at first. The service should however start. You can check it via:
```sh
systemctl status nukkit
```

richardpdunngb, 2022-04-02:
thanks for reply
please see below response from command

root@DietPi:~# systemctl status nukkit
● nukkit.service - nukkit (DietPi)
     Loaded: bad-setting (Reason: Unit nukkit.service has a bad unit file setting.)
     Active: inactive (dead)

Apr 02 19:55:58 DietPi systemd[1]: /etc/systemd/system/nukkit.service:7: Unbalanced quoting, ignoring: "/usr/bin/java -jar /usr/local/bin/nukkit/nukkit.jar'"
Apr 02 19:55:58 DietPi systemd[1]: nukkit.service: Unit configuration has fatal error, unit will not be started.
Apr 02 20:13:10 DietPi systemd[1]: /etc/systemd/system/nukkit.service:7: Unbalanced quoting, ignoring: "/usr/bin/java -jar /usr/local/bin/nukkit/nukkit.jar'"
Apr 02 20:13:10 DietPi systemd[1]: nukkit.service: Unit configuration has fatal error, unit will not be started.
Apr 02 21:08:47 DietPi systemd[1]: /etc/systemd/system/nukkit.service:7: Unbalanced quoting, ignoring: "/usr/bin/java -jar /usr/local/bin/nukkit/nukkit.jar'"
Apr 02 21:08:47 DietPi systemd[1]: nukkit.service: Unit configuration has fatal error, unit will not be started.
Apr 02 21:09:10 DietPi systemd[1]: /etc/systemd/system/nukkit.service:7: Unbalanced quoting, ignoring: "/usr/bin/java -jar /usr/local/bin/nukkit/nukkit.jar'"
Apr 02 21:09:10 DietPi systemd[1]: nukkit.service: Unit configuration has fatal error, unit will not be started.
Apr 02 21:34:51 DietPi systemd[1]: /etc/systemd/system/nukkit.service:7: Unbalanced quoting, ignoring: "/usr/bin/java -jar /usr/local/bin/nukkit/nukkit.jar'"
Apr 02 21:34:51 DietPi systemd[1]: nukkit.service: Unit configuration has fatal error, unit will not be started.

MichaIng, 2022-04-02:
There was an unintended single quote character 🤔. Can you try the following:
```sh
sed -i "s/nukkit.jar'/nukkit.jar/" /etc/systemd/system/nukkit.service
systemctl daemon-reload
systemctl restart nukkit
sleep 3
systemctl status nukkit
```

MichaIng, 2022-04-02:
Solved here: https://github.com/MichaIng/DietPi/commit/5eca8d5#diff-ff905774288e5cd43c5adcde46191c8990ae3fc1472fe5315344493aa946aade

I hope it is really the only issue, but fits very well to the error message.

richardpdunngb, 2022-04-02:
Many many thanks. Very swift and great solution fix.
Seams to be working now.

root@DietPi:~# systemctl status nukkit
● nukkit.service - nukkit (DietPi)
     Loaded: loaded (/etc/systemd/system/nukkit.service; disabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-04-02 21:57:08 BST; 30s ago
   Main PID: 1446 (java)
      Tasks: 30 (limit: 1070)
        CPU: 1min 5.887s
     CGroup: /system.slice/nukkit.service
             └─1446 /usr/bin/java -jar /usr/local/bin/nukkit/nukkit.jar

Apr 02 21:57:22 DietPi Nukkit[1446]: [46B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [63B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [47B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [47B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [64B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [51B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [50B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [56B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [55B blob data]
Apr 02 21:57:23 DietPi Nukkit[1446]: [67B blob data]
root@DietPi:~#

MichaIng, 2022-04-02:
Great, many thanks for testing/feedback. Great last minute fix before DietPi v8.3 release 😄.

closed 2022-04-02 by MichaIng