MichaIng/DietPi #5398: Weird behaviour of dietpi-dashboad service

JappeHallunken, 2022-04-03:
I tried to update the dietpi-dashboard via `sudo dietpi-software reinstall 200`, which failed.
The solution to this was, to stop the service manually beforehand.
This service is controlled by systemd by default, so I changed the mode to "DietPi controlled", but this doesn't changed the update to fail without stopping it manually beforehand.
I also noticed, that the service didn't autostart after reboot, when set to "DietPi controlled". This only works when controlled by systemd.
I didn't mess with the "Include/Exclude" option, it was always set to "included".

So dietpi can't control this service for some reason?

#### Required Information
- DietPi version | 8.3.1
- Distro version | bullseye
- Kernel version | Linux Dietpi 5.15.30-v8+ #1536 SMP PREEMPT Mon Mar 28 13:53:14 BST 2022 aarch64
- SBC model | RPi 4

MichaIng, 2022-04-03:
Many thanks for your report.

It is intended the DietPi does not control (start/stop on installs/maintenance/backups etc.) the dashboard service. This would break cases where tasks are done through the dashboard itself, installing software through it or using its web terminal feature, at it would stop itself then.

Note that `DietPi controlled` for such services only works when "including" them, but I don't recommend it for above reasons. Revert the change to have it starting at boot again:
```sh
systemctl enable dietpi-dashboard
```

The running service however shouldn't break the reinstall/update. Could be give more details or paste the console output about where/how it fails?

JappeHallunken, 2022-04-03:
Yes this makes a lot sense for me now, why it is handled by systemd.

I tried to update to the nightly build with "sudo dietpi-software reinstall 200".
No errors where thrown, I just noticed in the footer of the dashboard webpage, that the version number didn't changed there. I cleared my browser cache to make sure, it is not a leftover from the previous version, but it does not change.
From this I assumed, that the update failed somehow.

However, I did some more testing and figured out, that a restart of the service is necessary to show the new version on the webpage (which might be trivial for experienced users, but not for a DAU like me :)

I suggest to update the update command in the doc to:
`dietpi-software reinstall 200 && systemctl restart dietpi-dashboard`
to make it bulletproof.

Sorry for this inconvenience and thank your for your help and patience.

MichaIng, 2022-04-03:
Ah right. That info is also missing on our docs. Usually on reinstalls services are restarted, so we forgot to add it 😅 . I'll do that now.

MichaIng, 2022-04-03:
Info added (deployment will be done the next days): https://github.com/MichaIng/DietPi-Docs/commit/358eebc

I'll mark this as closed then.

closed 2022-04-03 by MichaIng

JappeHallunken, 2022-04-03:
I also just noticed, that some software mentioned in the docs is missing here: [https://dietpi.com/dietpi-software.html](https://dietpi.com/dietpi-software.html)

MichaIng, 2022-04-03:
Yes, it isn't been updated with new software titles for a while: https://github.com/MichaIng/DietPi-Website/issues/13