MichaIng/DietPi #5399: diet pi-update failed

tauchvogel, 2022-04-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used |5V 1A RAVpower
- SD card used | SanDisk

#### Additional Information (if applicable)
- Software title | ?
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour
The system should be updated.
- ...

#### Actual behaviour
The dashboard of Dietpi told me that an diet-update could be made to update the system. I started the system-update by typing "dietpi-update" in my console. The update process started and after a while it failed with the first screenshot I posted in this issue.

I tried to send a bug report via the system, but this failed too. This shows the second screenshot.

What can I do to update the system?

[image]

[image]

Joulinar, 2022-04-03:
This looks like some issues to resolve DNS correctly. Can you try to ping anything on the web?

Btw: no need to do screen prints. You could simply copy the error messages from SSH terminal. :wink:

tauchvogel, 2022-04-04:
I can not ping anything on the web. The test in dietpi-config failed too.

Joulinar, 2022-04-04:
Are you able to ping an IP address directly like `ping 1.1.1.1`?

tauchvogel, 2022-04-04:
Sorry for the delay. I installed all new.  Now it works.

Joulinar, 2022-04-04:
Going to close this issue. Feel free to reopen if needed.

closed 2022-04-04 by Joulinar