MichaIng/DietPi #5400: Iteration v8.4 (release plan)

MichaIng, 2022-04-03:
## Day 0: Init new development cycle
To be done until: `2022-04-03`
- [ ] Create a new GitHub milestone https://github.com/MichaIng/DietPi/milestones
- [ ] Select issues for the new iteration in the new milestone
- [ ] Create a new section in the [DietPi Wiki](https://github.com/MichaIng/DietPi/wiki/Iteration-plans)
- [ ] A new development cycle starts with a subversion increment, or major version increment when we drop support for an old Debian version or old DietPi versions. The RC version during the development cycle is "-1", to clearly identify unstable version strings.
- [ ] Init [`CHANGELOG.md`](https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt)
- [ ] Add support to [DietPi-Survey_report](https://github.com/MichaIng/DietPi/blob/dev/.meta/dietpi-survey_report)

## 3rd week: Beta phase
To be done until: `2022-04-23`
- [ ] Merge live patches from `master` and remove in `dev`.
- [ ] Create [pull requests](https://github.com/MichaIng/DietPi-Docs/pulls) for all [required documentation changes](https://github.com/MichaIng/DietPi-Docs/issues?q=is%3Aopen+is%3Aissue+label%3A"Next+DietPi+release"):

To be done at: `2022-04-23`
- [ ] Launch the initial beta as `.0` RC version
- [ ] Advertise on GitHub
- [ ] Advertise on [DietPi Forum](https://dietpi.com/phpbb/viewforum.php?f=9)
- [ ] Advertise on DietPi-Banner [MOTD](https://dietpi.com/motd)

To be done at: `2022-04-24`
- [ ] Advertise on website slider: https://dietpi.com/
- [ ] Advertise on Twitter ([DietPi account](https://twitter.com/DietPi_))
- [ ] Advertise on Facebook ([DietPi group](https://www.facebook.com/groups/167403007229675))

To be done from: `2022-04-24` - `2021-04-28`
- [ ] Address bug reports with further beta iterations, e.g. `.1` and `.2` RC versions

## 2 days before the release
To be done until: `2022-04-28`
- [ ] Have a release PR open on the main repo which needs to be linked from the [changelog](https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt)/release notes.
- [ ] Assure that [update server version](https://github.com/MichaIng/DietPi/dev/beta/.update/version) and [DietPi-Globals default version](https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-globals) match.
- [ ] Have the docs release PR ready and reviewed, with release notes and required changes merged into dev already.
- [ ] Have a PR on the website ready to advertise the new release via website slide and have the docs PRs all ready and reviewed.

## Release
To be done at: `2022-04-30`
- [ ] Make the release
- [ ] Publish the release notes in [DietPi Docs](https://dietpi.com/docs/releases/)
- [ ] Link the release notes from [website slider](https://dietpi.com/)
- [ ] Mention the release notes in [DietPi Forum](https://dietpi.com/phpbb/viewforum.php?f=10)
- [ ] Update the [MOTD](https://dietpi.com/motd)
- [ ] Update software list in [DietPi Wiki](https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list)

## Release announcements
To be done at: `2022-05-01`
- [ ] Publish on Twitter ([DietPi account](https://twitter.com/DietPi_))
- [ ] Publish on several Facebook groups (e.g. [DietPi group](https://www.facebook.com/groups/167403007229675))
- [ ] Announce on [Hacker News](https://news.ycombinator.com/)
- [ ] Forge and spread a press release
- [ ] Determine if a hotfix release is needed
- [ ] Create a new release plan

egidijus, 2022-04-06:
Hello, where can I find which kernel is being used?

Joulinar, 2022-04-06:
We don't have own kernel we use. The kernel depends on your SBC and the used base image. Usually it is Armbian or RPI OS.

closed 2022-05-02 by MichaIng