MichaIng/DietPi #5404: MotionEye Installation failed

HolgerTB, 2022-04-04:
System: Raspberry Pi Zero W / Dietpi Armv6 32 Bit - fresh install

During the motioneye installation this error occurred
 
Building wheels for collected packages: motioneye, pillow                                                              
                    │   Building wheel for motioneye (pyproject.toml): started                                                               
                    │   Building wheel for motioneye (pyproject.toml): finished with status 'done'                                           
                    │   Created wheel for motioneye: filename=motioneye-0.43.0-py3-none-any.whl size=723385                                  
                    │ sha256=b1f0436810981097af66a972bf0e1b678b09899719fba70f4c45f203f90d13f3                                                
                    │   Stored in directory:                                                                                                 
                    │ /tmp/pip-ephem-wheel-cache-hvepzqeu/wheels/5d/19/3b/8df45c45f762bba3d414e214316b57965795dbbf58c209bce5                 
                    │   Building wheel for pillow (setup.py): started                                                                        
                    │   Building wheel for pillow (setup.py): finished with status 'error'                                                   
                    │   error: subprocess-exited-with-error                                                                                  
                    │                                                                                                                        
                    │   × python setup.py bdist_wheel did not run successfully.                                                              
                    │   │ exit code: 1 

During handling of the above exception, another exception occurred:                                              
                    │                                                                                                                        
                    │       Traceback (most recent call last):                                                                               
                    │         File "<string>", line 2, in <module>                                                                           
                    │         File "<pip-setuptools-caller>", line 34, in <module>                                                           
                    │         File "/tmp/pip-install-hplcilrq/pillow_a8dd44d448ae42c484be9ef66631eb4e/setup.py", line 1009, in <module>      
                    │           raise RequiredDependencyException(msg)                                                                       
                    │       __main__.RequiredDependencyException:                                                                            
                    │                                                                                                                        
                    │       The headers or library files could not be found for jpeg,                                                        
                    │       a required dependency when compiling Pillow from source.                                                         
                    │                                                                                                                        
                    │       Please see the install instructions at:                                                                          
                    │          https://pillow.readthedocs.io/en/latest/installation.html                                                     
                    │                                                                                                                        
                    │                                                                                                                        
                    │       [end of output]                                                                                                  
                    │                                                                                                                        
                    │   note: This error originates from a subprocess, and is likely not a problem with pip.                                 
                    │ error: legacy-install-failure                                                                                          
                    │                                                                                                                        
                    │ × Encountered error while trying to install package.                                                                   
                    │ ╰─> pillow                          

Several retries were not successful

HolgerTB, 2022-04-04:
after reboot motioneye service did not start. A library was missing

apt install libatlas3-base

fixed the problem

MichaIng, 2022-04-04:
Many thanks for your report.

Dammit, I knew this already but forgot to add it to our code: https://github.com/motioneye-project/motioneye/issues/2399

`libgfortran5` is pulled in by something else already, but `libatlas3-base` indeed is required when using piwheels, hence on ARMv6 and ARMv7 systems.

MichaIng, 2022-04-06:
Solved upstream by removing `numpy` use and dependency: https://github.com/motioneye-project/motioneye/pull/2403

closed 2022-04-06 by MichaIng