MichaIng/DietPi #5405: Set both eth0 and wlan0 to use static IPs, but unable to start wlan0 if eth is not connected

mvimercati, 2022-04-04:
### Creating a bug report/issue

#### Required Information

- DietPi version | G_DIETPI_VERSION_CORE=8
    G_DIETPI_VERSION_SUB=3
    G_DIETPI_VERSION_RC=1
    G_GITBRANCH='master'
    G_GITOWNER='MichaIng'
- Distro version | bullseye
- Kernel version | Linux DietPi 5.15.32-v7+ # 1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
- SBC model | RPI 3B
- Power supply used | 5V 2A POWER SUPPLY
- SD card used | 64GB microSD Sandisk

#### Steps to reproduce
1. Set static IP address for ethernet (eth0)
2. Set static IP address for wifi (wlan0)
3. Set wlan0 to connect an AP
4. Reboot Disconnect ethernet cable

#### Expected behaviour
At boot the wlan0 adapter connects to the AP and the defined static IP address is used

#### Actual behaviour
_ifup_ fails on wlan0, and the static IP address is not assigned. No connections working.

#### Extra details
Workaround: Set eth0 to use DHCP; at this point the wlan0 is able to configure
Seems working when eth cable is connected

mvimercati, 2022-04-04:
Probably duplicate of #2103. Sorry, found only right now.

MichaIng, 2022-04-04:
The issue, I guess, is the doubled default route/gateway. Even if the Ethernet cable is not connected, the adapter is detected at boot, usually earlier than the WiFi adapter, and with static IP + gateway is configured and hence a default route for eth0 created. Since no two default routes can exist, the wlan0 configuration afterwards fails.

While there are way to work around this with routing tables, the only real solution is to have always only one interface configured with gateway and the other without, i.e. remove the gateway line within the eth0 block in /etc/network/interfaces.

If you want WiFi to work as fallback for Ethernet, when its not connected, [ifplugd](https://packages.debian.org/bullseye/ifplugd) is a daemon which allows to configure and deconfigure Ethernet when the cable gets connected resp. disconnected. Here an example about how to configure it: https://dietpi.com/phpbb/viewtopic.php?p=25620#p25620

mvimercati, 2022-04-05:
Yes, it makes sense. Maybe I could try to use ifplugd to toggle the gateways between the two interfaces when I plug the eth cable. Unfortunatelly wifi is baseline connection for the rpi.. Thanks for you suggestions!

MichaIng, 2022-05-29:
Did it work, respectively can this issue be closed?

MichaIng, 2022-07-23:
Marking as closed. Feel free to reopen if required.

closed 2022-07-23 by MichaIng