MichaIng/DietPi #5406: Use encrypted Wifi password in dietpi.txt config file

robinemde, 2022-04-05:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Put dietpi.txt files aside for later re-installs of Raspis are revealing the Wifi password

#### Describe the solution you'd like:

- Raspberry Pi OS allows to use the "wpa_passphrase" command to encrypt the password and use that instead of clear text password.

#### Describe alternatives you've considered:

- Use wpa_supplicant file instead of dietpi.txt, but this means you have two config files to backup and no central file for all config.

#### Additional context

- ...

Joulinar, 2022-04-05:
Hi,

you mean the `dietpi-wifi.txt`? Because a WiFi password is not stored in `dietpi.txt` 😉 

As well `dietpi-wifi.txt` is used on initial first boot only and will be removed afterwards. We store the WiFi password inside `wpa_supplicant` once WiFi has been setup.

robinemde, 2022-04-05:
Hi Joulinar,

I am sorry, yes, I mean the dietpi-wifi.txt, of course. And yes, that means you have two (dietpi) config files anyway...
I already guessed that the process of handling the wifi password (read the file, insert into proper config files, delete the dietpi-wifi.txt file) was exactly the way you described. Thanks.

However, I once tried to use my encrypted wifi password in the dietpi-wifi.txt file, but it did not work.
If the DietPi config mechanism just takes the password string and puts it into wpa_supplicant (without adding quotes), then it should check if quotes are used. With quotes = clear text. Without quotes = encrypted pw.

I would really like to see this working in DietPi.

MichaIng, 2022-04-05:
Looks like `wpa_supplicant.conf` allows a hashed key indeed: https://manpages.debian.org/wpa_passphrase
"Encryption" is probably not the right wording here, but sounds like a 256 bit hash, generated from both, SSID and plain text key, so that it only works in combination.

DietPi does not much more than generating a `wpa_supplicant.conf` from `dietpi-wifi.txt`, so if using the hashed password does not work OOTB, then probably different quotation or some other syntax needs to be used. I cannot find any hint in the [man page](https://manpages.debian.org/wpa_supplicant.conf), but will try to figure it out.

robinemde, 2022-05-13:
Hi, did you find out anything or is there already anything going on behind the scenes to adapt the feature for DietPi...?

MichaIng, 2022-05-21:
Done with: https://github.com/MichaIng/DietPi/commit/5d48572

closed 2022-05-27 by MichaIng