MichaIng/DietPi #5407: DietPi-Automation | WiFi Hotspot installation does not set the wifi country code

matteocarnelos, 2022-04-05:
### Creating a bug report/issue

#### Required Information
- DietPi version
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `bullseye 1`
- Kernel version | `Linux DietPi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B+ (armv7l)`
- Power supply used | 5V 3A
- SD card used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | WiFi Hotspot
- Was the software title installed freshly or updated/migrated? Freshly installed using DietPi-Automation
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Perform a fresh install of the latest DietPi, setting the following in the `dietpi.txt` file:
```
AUTO_SETUP_AUTOMATED=1
AUTO_SETUP_INSTALL_SOFTWARE_ID=60
```

#### Expected behaviour

- After the install, the wifi country code should be set accordingly to `AUTO_SETUP_NET_WIFI_COUNTRY_CODE` in `dietpi.txt`

#### Actual behaviour

- The wifi country code is not set

#### Extra details

Issue probably due to the fact that, since the wifi adapter is disabled during first boot, the country code is not applied, as said by @MichaIng in [this forum post](https://dietpi.com/phpbb/viewtopic.php?p=35490#p35490):
> The country code is applied automatically on first boot and applied to /etc/default/crda to be loaded on every next boot via udev rules, as fast as the WiFi device is attached/found by the kernel, but indeed only when WiFi is enabled dietpi.txt wise. Actually it makes sense to do that regardless whether WiFi is enabled or not, so that the correct country code is applied as fast as any WiFi adapter is enabled.

As said, a possible fix is to set the wifi country code regardless wether WiFi is enabled or not.

Thanks for the help!

MichaIng, 2022-04-05:
Many thanks for your report.

How do you know that the country code is not set correctly? Did you check `iw reg get`?

The `dietpi-config`/`dietpi.txt` option sets it via CRDA, but since the forum post I found out that with modern kernel it is set automatically without CRDA, i.e. CRDA is actually obsolete nowadays and has hence been removed for Debian Bookworm (next stable) already: https://packages.debian.org/crda

I can basically verify this on my RPis and x86 systems with Bullseye, where changing the country code via CRDA to something that does not match my country does not survive a reboot, it is set automatically correctly.

When connecting via WiFi to an access point, this all makes sense since the AP can pass the country code. But when using the adapter as hotspot, I don't know how the kernel should know which country code is the correct one. Basically I need to read into this topic and run some tests.

matteocarnelos, 2022-04-06:
> How do you know that the country code is not set correctly? Did you check iw reg get?

After the install, if I go to `dietpi-config` > `Network Options: Adapters` > `WiFi` the country code is not set:

[image]

And here's the output of `iw reg get`:
```
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)

```

matteocarnelos, 2022-04-06:
Unfortunately, after purging CRDA and rebooting, the country code was still not set. So yes, I think we should set it during WiFi Hotspot installation.

As a temporary workaround I put the following in `Automation_Custom_Script.sh`:
```
# Set WiFi Country Code
/boot/dietpi/func/dietpi-set_hardware wificountrycode "$(sed -n '/^[[:blank:]]*AUTO_SETUP_NET_WIFI_COUNTRY_CODE=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)"
```

MichaIng, 2022-04-22:
Solved with: https://github.com/MichaIng/DietPi/commit/c864974

closed 2022-04-22 by MichaIng