MichaIng/DietPi #5409: curl failed to verify the legitimacy of the server

Adicke, 2022-04-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.2,2`
- Distro version | ` buster 1`
- Kernel version | `5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B+ (armv7l)`

#### Steps to reproduce
1. run: `dietpi-update`
2. OK to update
3. "Do you wish to continue..." OK

#### Expected behaviour
- DietPi should run through the update

#### Actual behaviour
- Update stops with the error message: 

> curl failed to verify the legitimacy of the server and therefore could not
> │ establish a secure connection to it. To learn more about this situation and

- tried "Retry". Doesn't help
- console says:

> [FAILED] DietPi-Update | Downloading update archive
>  - Command: curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz

#### Extra details
- `grep 'ssh.dietpi.com' /root/.ssh/known_hosts` shows the key. I tried to add it again. Doesn't help

Joulinar, 2022-04-07:
That is more an issue of verifying the certificate for GitHub. Therefore it doesn't make much sense to install keys for DietPi server. Maybe a temp GitHub issue and you could try later again.

Adicke, 2022-04-07:
I've done that after it has been [recommended](https://github.com/MichaIng/DietPi/issues/5137#issuecomment-1006092054) the last time the issue came up.

Joulinar, 2022-04-07:
Totally different issue 😉

Adicke, 2022-04-07:
Time is correct. 
Second curl didn't work out: 

> curl: (60) SSL certificate problem: self signed certificate in certificate chain
> More details here: https://curl.haxx.se/docs/sslcerts.html
>

Adicke, 2022-04-08:
Thank you. 
That was it.

closed 2022-04-08 by Adicke