MichaIng/DietPi #5410: Only Info: Concurrent execution of DietPi-Software detected

Axellarator, 2022-04-07:
Got an Internet Outage during DP Update 8.3. Had to remove the empty /tmp/DietPi-Update directory.   After that DP Update was back in business.

MichaIng, 2022-04-07:
Best is usually to terminate the orphaned SSH session, e.g. via `htop`, navigating to the SSH subprocess (the one that doesn't have `htop` itself as child process but has the `dietpi-update` as child), hit F9 and return. This should terminate the running `dietpi-update` as well, which also removes `/tmp/DietPi-Update` via EXIT trap.

An alternative to avoid such issues when SSH connections are lost, [GNU/screen](https://www.gnu.org/software/screen/manual/screen.html) (`apt install screen`) is a nice tool which allows to decouple the shell session from the SSH session. A screen session can be detached and re-attached, so when the SSH connection gets lost while a task or script is running, it continues in the background, and when you re-connect via SSH, you can re-attach to that screen session and continue the process.

Axellarator, 2022-04-13:
Thanks,
screen is useful! 
@Joulinar: /tmp/DietPi-Update was empty and somehow survived the reboot. But let us stop it here. The Raspberry Pi B is my oldest and has only a sentimental value. DietPi 8.3.1 works fine after the update. The update took a while, but I won't use the old horse for production or anything useful.

closed 2022-04-13 by Axellarator

MichaIng, 2022-04-13:
> /tmp/DietPi-Update was empty and somehow survived the reboot

That can only be if `/tmp` is no tmpfs: `df -T /tmp`
Note that DietPi is running an update check on boot, so running `dietpi-update` too quickly after boot may cause this expectedly. You can check whether it's really running via `htop`.