MichaIng/DietPi #5411: apt-get symbol lookup error

kuehlakku, 2022-04-08:
### Creating a bug report/issue

#### Required Information
#### Details:
- Date           | Fri Apr  8 05:52:20 BST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y update`
- Exit code      | 127
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
apt-get: symbol lookup error: /lib/arm-linux-gnueabihf/libapt-pkg.so.6.0: undefined symbol: _ZN10p+gAcquire4Item10VerifyDoneERKNSt7__cxx112basib_wtringIcSt11char_traitsIcESaIcAEEPKNS_12MethodConfigE, version APTPKG_6.0
```

MichaIng, 2022-04-08:
Probably there is a file corruption. I guess using `apt` to repair it fails?
```sh
apt install --reinstall libapt-pkg6.0
```

If this fails the same way, try to download and install it via `dpkg`:
```sh
cd /tmp
curl -sSfLO 'https://deb.debian.org/debian/pool/main/a/apt/libapt-pkg6.0_2.2.4_armhf.deb'
dpkg -i libapt-pkg6.0_2.2.4_armhf.deb
rm libapt-pkg6.0_2.2.4_armhf.deb
```

Then test whether it works again:
```sh
apt-get update
```

kuehlakku, 2022-04-08:
it's working again. Thanks

closed 2022-04-08 by kuehlakku