MichaIng/DietPi #5413: Is there any way to upgrade Kodi from 18 to 19 ?

Joulinar, 2022-04-09:
Is there any way to upgrade Kodi from 18 to 19 ?

_Originally posted by @EdWaQ in https://github.com/MichaIng/DietPi/issues/5382#issuecomment-1093932844_

Joulinar, 2022-04-09:
Pls can you share some more information on your system

#### Required 
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

EdWaQ, 2022-04-09:
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

buster 1

Linux DietPi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

I hope i´m right with them.

MichaIng, 2022-04-09:
For Raspberry Pi on Buster, only Kodi 18.7 is available. For newer Kodi you need to upgrade to Bullseye: https://dietpi.com/blog/?p=811

Or you can compile it yourself, of course. These build instructions are for Bullseye, but it should work on Buster as well: https://forums.raspberrypi.com/viewtopic.php?p=1935075#p1935075

MichaIng, 2022-05-29:
Marking this as closed. Feel free to reopen when there are still questions.

closed 2022-05-29 by MichaIng