MichaIng/DietPi #5415: Unable to install single file php gallery

dodo1967, 2022-04-09:
#### Details:
- Date           | Sat Apr  9 16:03:12 BST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux server 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `curl -ILfvm 10 https://sye.dk/sfpg/Single_File_PHP_Gallery_4.8.1.zip`
- Exit code      | 22
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 2a02:2350:5:10a:4a:9025:f784:46f3:443...
* Connected to sye.dk (2a02:2350:5:10a:4a:9025:f784:46f3) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [4278 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [520 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [52 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [52 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=*.sye.dk
*  start date: Apr  8 19:45:30 2022 GMT
*  expire date: Jul  7 19:45:29 2022 GMT
*  subjectAltName: host "sye.dk" matched cert's "sye.dk"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
} [5 bytes data]
* Using Stream ID: 1 (easy handle 0x241ed50)
} [5 bytes data]
> HEAD /sfpg/Single_File_PHP_Gallery_4.8.1.zip HTTP/2
> Host: sye.dk
> user-agent: curl/7.74.0
> accept: */*
> 
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [265 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [265 bytes data]
* old SSL session ID is stale, removing
{ [5 bytes data]
* Connection state changed (MAX_CONCURRENT_STREAMS == 100)!
} [5 bytes data]
* The requested URL returned error: 404 
* stopped the pause stream!
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Connection #0 to host sye.dk left intact
curl: (22) The requested URL returned error: 404 
```

MichaIng, 2022-04-09:
A new version is available. Nastily the old one has been removed in turn from the server. Please try the following:
```sh
sed -i 's|https://sye.dk/sfpg/Single_File_PHP_Gallery_4.8.1.zip|https://www.sye.dk/sfpg/Single_File_PHP_Gallery_4.9.0.zip|' /boot/dietpi/dietpi-software
```
Then rerun the installation.

MichaIng, 2022-04-14:
PR up: #5432

I asked Kenny (SFPG developer) whether we can provide a "latest" download URL, or an API from where the latest version can be obtained via curl. But otherwise I suggest we revert back to the previous solution, hosting the download at dietpi.com, so that it stays valid while we can update it with a simple download on our server.

closed 2022-04-21 by MichaIng