MichaIng/DietPi #5417: Dietpi software HTPC

born2faq, 2022-04-10:
#### Details:
- Date           | Sun Apr 10 16:27:15 EEST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `pip3 install --no-cache-dir -Ur requirements.txt`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Traceback (most recent call last):
  File "/usr/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
ModuleNotFoundError: No module named 'pip'
```
---------------------------------------------------------------------

closed 2022-04-10 by born2faq

MichaIng, 2022-04-10:
You solved it yourself? Usually when installing HTPC Manager, Python 3, including `pip`, should be automatically installed first, so that "ModuleNotFoundError: No module named 'pip'" shouldn't show up. Did this somehow not happen in your case, or did the Python 3 installation fail?

Joulinar, 2022-04-10:
I did a test on my RPI4B 64bit without issues.