MichaIng/DietPi #5420: Issue while trying to install Postgresql

Bluesign71, 2022-04-11:
#### Details:
- Date           | Mon Apr 11 18:25:20 CEST 2022
- DietPi version | v8.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install postgresql`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

I can't install postgresql by (apt-get install postgresql postgresql-contrib) so I tried from dietpi-software, same result....
#### Expected behaviour:

- ...
#### Actual behaviour:

I seguenti pacchetti hanno dipendenze non soddisfatte:
 postgresql-11 : Dipende: postgresql-common (>= 194~) ma non è installato
 postgresql-client-11 : Dipende: postgresql-client-common (>= 182~) ma non è installato
E: Dipendenze non soddisfatte. Provare "apt --fix-broken install" senza pacchetti (o specificare una soluzione).
root@DietPi:~# apt --fix-broken install postgresql-common
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze
Lettura informazioni sullo stato... Fatto
È utile eseguire "apt --fix-broken install" per correggere ciò.
.... and then again.....
#### Extra details:

- ...
#### Additional logs:
```
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
I tried but I always receive:

apt --fix-broken install
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze
Lettura informazioni sullo stato... Fatto
Correzione delle dipendenze... Fatto
I seguenti pacchetti aggiuntivi saranno inoltre installati:
  postgresql-client-common postgresql-common
Pacchetti suggeriti:
  libjson-perl
Pacchetti raccomandati:
  logrotate
I seguenti pacchetti NUOVI saranno installati:
  postgresql-client-common postgresql-common
0 aggiornati, 2 installati, 0 da rimuovere e 49 non aggiornati.
4 non completamente installati o rimossi.
È necessario scaricare 310 kB di archivi.
Dopo quest'operazione, verranno occupati 847 kB di spazio su disco.
Continuare? [S/n] s
Scaricamento di:1 http://ftp.arnes.si/mirrors/raspbian/raspbian buster/main armhf postgresql-client-common all 200+deb10u4 [85,1 kB]
Scaricamento di:2 http://ftp.arnes.si/mirrors/raspbian/raspbian buster/main armhf postgresql-common all 200+deb10u4 [225 kB]
Recuperati 310 kB in 1s (545 kB/s)
Preconfiguring packages ...
(Lettura del database... 48365 file e directory attualmente installati.)
Preparativi per estrarre .../postgresql-client-common_200+deb10u4_all.deb...
Estrazione di postgresql-client-common (200+deb10u4)...
dpkg: errore nell'elaborare l'archivio /var/cache/apt/archives/postgresql-client-common_200+deb10u4_all.deb (--unpack):
 errore nel creare il collegamento simbolico "./usr/bin/clusterdb": Operazione non permessa
Preparativi per estrarre .../postgresql-common_200+deb10u4_all.deb...
Viene aggiunto "deviazione di /usr/bin/pg_config in /usr/bin/pg_config.libpq-dev da postgresql-common"
Estrazione di postgresql-common (200+deb10u4)...
dpkg: errore nell'elaborare l'archivio /var/cache/apt/archives/postgresql-common_200+deb10u4_all.deb (--unpack):
 impossibile creare "/usr/bin/pg_config.dpkg-new" (durante l'elaborazione di "./usr/bin/pg_config"): Operazione non permessa
dpkg-deb: errore: il sottoprocesso paste è stato terminato dal segnale (Pipe interrotta)
Rimozione di "deviazione di /usr/bin/pg_config in /usr/bin/pg_config.libpq-dev da postgresql-common"
Si sono verificati degli errori nell'elaborazione:
 /var/cache/apt/archives/postgresql-client-common_200+deb10u4_all.deb
 /var/cache/apt/archives/postgresql-common_200+deb10u4_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

please help me if you can !

MichaIng, 2022-04-12:
Many thanks for your report.

I see quite some `Operazione non permessa` (operation not permitted) in the log. Can you try to reset the APT cache:
```sh
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt -f install
```

If it still fails, please the the output of:
```sh
ls -ld /usr/bin /usr/bin/pg_config
```

Bluesign71, 2022-04-14:
Thanks so much for Your attention MichaIng, and excuse me if didn't change language before to post my outputs from dietpi.
I didn't tried to do what You suggested, because on tuesday I had to format my sd card and start everything from zero.
This due to the fact I started having problem even in the boot. I use my raspberry for a nextcloud server, blynk and Apache server and I can't stay without for too long (I can't open my car box without blynk and dietpi !!!, I use it also for my job.
Btw, Your kind answer is the proof that Dietpi it is a correct and responsible choice and I'm right installing it again, with such a level of assistance. Thanks again and all the best

Joulinar, 2022-04-14:
does it mean this issue can be closed?

Bluesign71, 2022-04-14:
Yes please.

closed 2022-04-14 by Joulinar