MichaIng/DietPi #5423: Archivebox

cocoflan, 2022-04-12:
https://archivebox.io/

ArchiveBox is a powerful, self-hosted internet archiving solution to collect, save, and view sites you want to preserve offline.

You can set it up as a [command-line tool](https://archivebox.io/#quickstart), [web app](https://archivebox.io/#quickstart), and [desktop app](https://github.com/ArchiveBox/electron-archivebox) (alpha), on Linux, macOS, and Windows.

You can feed it URLs one at a time, or schedule regular imports from browser bookmarks or history, feeds like RSS, bookmark services like Pocket/Pinboard, and more. See [input formats](https://archivebox.io/#input-formats) for a full list.

It saves snapshots of the URLs you feed it in several formats: HTML, PDF, PNG screenshots, WARC, and more out-of-the-box, with a wide variety of content extracted and preserved automatically (article text, audio/video, git repos, etc.). See [output formats](https://archivebox.io/#output-formats) for a full list.

The goal is to sleep soundly knowing the part of the internet you care about will be automatically preserved in durable, easily accessible formats [for decades](https://archivebox.io/#background--motivation) after it goes down.

cocoflan, 2022-04-12:
I am testing this system, it is really good, and easy to use. A must have for collectors of webpage information to watch offline.

MichaIng, 2022-04-12:
Many thanks for your request. Sounds interesting. Also easy installable via single curl command. Let's hope the script behind it is not so intrusive.

I'll move this to discussion, so others can vote.