MichaIng/DietPi #5427: You don't have enough free space in /var/cache/apt/archives/

G2G2G2G, 2022-04-13:
#### Details:
- Date           | Wed Apr 13 22:13:46 BST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install nginx-light`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ... went to install pi-hole on a brand new dietpi install
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libnginx-mod-http-echo nginx-common
Suggested packages:
  fcgiwrap nginx-doc ssl-cert
The following NEW packages will be installed:
  libnginx-mod-http-echo nginx-common nginx-light
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 701 kB of archives.
After this operation, 1724 kB of additional disk space will be used.
E: You don't have enough free space in /var/cache/apt/archives/.
```

### You can close this, I have already solved the issue by cleaning that folder. Novice users might not be able to so I am only reporting because of this msg:
```
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
 ```

Seems this isn't the first time https://github.com/MichaIng/DietPi/issues/917

Joulinar, 2022-04-13:
you should check the disk usage in general by running following. It will show the usage of all drives and list the 20 biggest files/directories. 

```
df -h
du -a / | sort -n -r | head -n 20
```

In generell apt cache can be cleaned using following command.

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update

```

G2G2G2G, 2022-04-13:
Thanks, I already said I solved it... and to close it. If you need more assistance in your bug reports fill the copy paste thing with more info.

closed 2022-04-13 by G2G2G2G

MichaIng, 2022-04-13:
Just note that cleaning this directory is a temporary solution only. As fast as new package upgrades are applies, it is re-filled, so you can run into the same error again and again. This directory is on the regular root partition, so it seems you require a larger SD card/drive (or remove data/uninstall software). The suggested commands would verify this.