MichaIng/DietPi #5428: Unable to install Bazaar

brogers1812, 2022-04-14:
#### Details:
- Date           | Wed Apr 13 13:52:18 HST 2022
- Bug report     | 1388ea2d-5147-4bf5-850d-dcee50ceb172
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi-Desktop 5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `pip3 install --no-cache-dir -Ur requirements.txt`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. I attempted to install Bazaar through the Dietpi-Software module.

#### Expected behaviour:

- Bazaar should have been installed.
#### Actual behaviour:

- Bazaar fails to install due to the reason listed above
#### Extra details:

- N/A
#### Additional logs:
N/A
Traceback (most recent call last):
  File "/usr/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
ModuleNotFoundError: No module named 'pip'
```

Hopefully this information helps.

Joulinar, 2022-04-14:
Could you do a reboot and try again pls.

MichaIng, 2022-04-14:
> File "/usr/bin/pip3", line 5, in <module>

That is the wrong file path, it should be `/usr/local/bin/pip3` 🤔. Like if `get-pip.py` failed or so. If it still fails after reboot, can you show:
```sh
sudo bash -c '
dpkg -l | grep python
which -a pip3
ls -l /usr/local/bin/pip3
echo $PATH'
```

brogers1812, 2022-04-14:
I've tried rebooting a couple of times. I've uninstalled and reinstalled python3 using the software manager. Other than that, I'm lost on the next steps.

ii  libpython3-dev:amd64    3.9.2-3 amd64    header files and a static library for Python (default)
ii  libpython3-stdlib:amd64  3.9.2-3 amd64  interactive high-level object-oriented language (default python3 version)
ii  libpython3.9:amd64  3.9.2-1 amd64        Shared Python runtime library (version 3.9)
ii  libpython3.9-dev:amd64  3.9.2-1  amd64        Header files and a static library for Python (v3.9)
ii  libpython3.9-minimal:amd64 3.9.2-1 amd64      Minimal subset of the Python language (version 3.9)
ii  libpython3.9-stdlib:amd64  3.9.2-1 amd64      Interactive high-level object-oriented language (standard library, version 3.9)
ii  python-pip-whl 20.3.4-4+deb11u1 all          Python package installer (pip wheels)
ii  python3 3.9.2-3 amd64   interactive high-level object-oriented language (default python3 version)
ii  python3-dev 3.9.2-3  amd64 header files and a static library for Python (default)
ii  python3-distutils   3.9.2-1  all  distutils package for Python 3.x
ii  python3-lib2to3  3.9.2-1  all  Interactive high-level object-oriented language (lib2to3)
ii  python3-libxml2:amd64	2.9.10+dfsg-6.7+    deb11u1         amd64        Python3 bindings for the GNOME XML library
ii  python3-lxml:amd64 4.6.3+dfsg-0.1+d  eb11u1  amd64  pythonic binding for the libxml2 and libxslt libraries
ii  python3-minimal 3.9.2-3 amd64    minimal subset of the Python language (default python3 version)
ii  python3-pip 20.3.4-4+deb11u1   all          Python package installer
ii  python3-pkg-resources  52.0.0-4 all Package Discovery and Resource Access using pkg_resources
ii  python3-setuptools     52.0.0-4 all Python3 Distutils Enhancements
ii  python3-wheel 0.34.2-1  all         built-package format for Python
ii  python3.9 3.9.2-1 amd64  Interactive high-level object-oriented language (version 3.9)
ii  python3.9-dev  3.9.2-1 amd64        Header files and a static library for Python (v3.9)
ii  python3.9-minimal 3.9.2-1 amd64        Minimal subset of the Python language (version 3.9)
/usr/bin/pip3
/bin/pip3
ls: cannot access '/usr/local/bin/pip3': No such file or directory
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

MichaIng, 2022-04-14:
Strange, the APT `pip` package is installed. Only explanation I have is that since recently this prevents `pip` from being installed via `get-pip.py`, without throwing an error message. But even worse is that this `pip` then isn't usable, apparently.

Please re-run the `get-pip.py` installation manually and paste the output here. Then purge the `python3-pip` package and retry:
```sh
cd /tmp
curl -sSfLO 'https://bootstrap.pypa.io/get-pip.py'
python3 get-pip.py
apt autopurge python3-pip
python3 get-pip.py
rm get-pip.py
```

brogers1812, 2022-04-14:
**get-pip.py manual installation output**
```
/usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
Collecting pip
  Using cached pip-22.0.4-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
/usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
Successfully installed pip-22.0.4
```

I ran the code you provided and no issues. I retry installing Bazaar and it throws the same error as before

```
File "/usr/bin/pip3", line 5, in <module>                                   
 from pip._internal.cli.main import main                                   
 ModuleNotFoundError: No module named 'pip'
```

Joulinar, 2022-04-14:
@brogers1812 did you purged the `python3-pip` packages as requested by @MichaIng and tried again?

@MichaIng Just to crosscheck/test, I did an installation without issues on a Bullseye VM.

brogers1812, 2022-04-14:
@Joulinar I ran the code that was provided by @MichaIng . Was there an additional step that I missed? I assumed purging `python3-pip` was included in the code listed below.

```
cd /tmp
curl -sSfLO 'https://bootstrap.pypa.io/get-pip.py'
python3 get-pip.py
apt autopurge python3-pip
python3 get-pip.py
rm get-pip.py
```

MichaIng, 2022-04-14:
Actually it is. Can you check again:
```sh
dpkg -l python3-pip
dpkg -S /usr/bin/pip3
```

brogers1812, 2022-04-14:
I appreciate the quick replies.

`root@DietPi:/tmp# dpkg -l python3-pip`
**dpkg-query: no packages found matching python3-pip**
`root@DietPi:/tmp# dpkg -S /usr/bin/pip3`
**dpkg-query: no path found matching pattern /usr/bin/pip3**

MichaIng, 2022-04-14:
So far so good. Does this file still exist?
```sh
ls -l  /usr/bin/pip3
```

MichaIng, 2022-04-14:
Strange. Please remove it manually, as well as wheel and setuptools packages, which are also installed by `get-pip.py`:
```sh
rm /usr/bin/pip3
apt autopurge python3-setuptools python3-wheel
dpkg -l | grep python
```

Then try again to install via `get-pip.py`
```sh
cd /tmp
curl -sSfLO 'https://bootstrap.pypa.io/get-pip.py'
python3 get-pip.py
rm get-pip.py
ls -l /usr/local/bin/pip3
```

MichaIng, 2022-04-14:
So I guess it is installed to the false path, and isn't usable?
```sh
ls -l /usr/bin/pip3
pip3 -V
```

Do you have any other Python software installed?
```sh
dietpi-software list | grep -i python
apt purge python* # Just to check for dependants: do not confirm, unless there are Python packages purged only and no software that you actually need
```

I hope we are able to solve this strange behaviour by purging all Python APT packages and all modules and installing it freshly.

MichaIng, 2022-05-27:
I'll mark this as closed, feel free to reopen when the issue persists. I'm unable to find a path which leads to the case that `/usr/local/bin/pip3` is not present after installing `pip` via `get-pip.py`, would be still interesting to find out what the reason for this is in your case.

closed 2022-05-27 by MichaIng