MichaIng/DietPi #5429: Dietpi-sync

cocoflan, 2022-04-14:
possibility to sync multi location (directory's) and set sync time manually.

Joulinar, 2022-04-14:
The sync time could be specified using `dietpi-cron`. There you could adjust the daily execution time on your needs.

MichaIng, 2022-04-14:
Would be actually an idea to add `dietpi-cron` to the `dietpi-sync` menu 🤔, like so:
```
Daily Sync Time : Set the time for the daily sync via dietpi-cron
```

MichaIng, 2022-04-14:
That is true. At least this point should me made clear. However, to make users aware that generally the daily cron job execution time, including daily `dietpi-sync`, can be adjusted via `dietpi-cron`, is a good idea, I think. Not everyone has all DietPi tools in mind, and what one can do with it.

MichaIng, 2022-04-14:
Btw, about multiple sync locations, here I collected some ideas about how this could look like, also merging `dietpi-backup` into the same script, to be just another hardcoded sync slot: https://github.com/MichaIng/DietPi/issues/2051#issuecomment-450599769