MichaIng/DietPi #5430: DietPi Update updates PiHole

c-m-w, 2022-04-14:
I'm frustrated having to update PiHole and DietPi separately. It'd be nice to have it consolidated into one command to keep both up to date. Perhaps with an option so that when `dietpi-update` is called, `pihole -up` is as well.

Joulinar, 2022-04-14:
simply run both commands together. Personally I would not do that. Because if thinks are failing, you would not be sure what caused it.

```
pihole -up && dietpi-update
```

MichaIng, 2022-04-14:
We intentionally try to decouple DietPi updates from software updates and in a way even from APT package upgrades (by showing them in banner to be hopefully applied more regularly). As Joulinar stated, updating many thangs in one step means that in case of any issue it is harder for you (and in case for us) to identify what is responsible.

Generally, you should apply updates as fast as they become available (APT and Pi-hole updates are usually security relevant as well!), and you have the time to investigate and fix issues, just in case. Especially software with a web interface, like Pi-hole, usually show available updates right in the web interface, some have an internal web-based updater as well (not true for Pi-hole).

However, where CLI update checkers/updaters are available, we might integrate those into `dietpi-update` and/or the login banner, so that e.g. those updates are shown can can be applied from the `dietpi-update` menu as well, but at first not with all updates applied in one step for mentioned reasons. This would require a clear separation, dedicated confirmation, or a test like: Check whether the Pi-hole web interface and DNS are running after the DietPi update, before applying the Pi-hole update, or the other way round, to not harden debugging unnecessarily.

MichaIng, 2022-05-29:
Marking this as closed. Feel free to reopen when there are still questions.

closed 2022-05-29 by MichaIng