MichaIng/DietPi #5433: USB devices only appears in dmesg and lsusb, not seen or mounted as disks

Bluesign71, 2022-04-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2022-04-15:
Did you used `dietpi-drive_manager` to mount your drive?

Bluesign71, 2022-04-15:
yes, tried also

[image]

Joulinar, 2022-04-15:
Can you share following

`lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid`

Bluesign71, 2022-04-15:
`root@DietPi:~# lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME           FSTYPE  LABEL  SIZE       RO     TYPE   MOUNTPOINT    PARTUUID          UUID
mmcblk0                                29,8G       0      disk
├─mmcblk0p1  vfat               128M       0      part     /boot                455d093c-01     806D-D19A
└─mmcblk0p2  ext4              29,7G       0      part     /                        455d093c-02     9e1cb772-602b-4210-99bd-2eb47ea516a9
`

Bluesign71, 2022-04-15:
root@DietPi:~# lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME        FSTYPE LABEL  SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
mmcblk0                  29,8G  0 disk
├─mmcblk0p1 vfat          128M  0 part /boot      455d093c-01                          806D-D19A
└─mmcblk0p2 ext4         29,7G  0 part /          455d093c-02                          9e1cb772-602b-4210-99bd-2eb47ea516a9

Joulinar, 2022-04-15:
There is no USB drive detected by your system. Did you checked for kernel error message?

`dmesg -l err,crit,alert,emerg`

Bluesign71, 2022-04-15:
root@DietPi:~# dmesg -l err,crit,alert,emerg
[    0.555413] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.

MichaIng, 2022-04-15:
Is it a 2.5" drive, and if so, has it a dedicated PSU or is it powered by USB only? The letter is known to not work reliably with the max shared 1.2A across all USB ports. Especially during powering on, 2.5" USB drives are known to draw more power than available from those USB ports so that they are not detected correctly or the connection gets lost during high I/O. This is true for spinning HDDs as well as SSDs.

Bluesign71, 2022-04-15:
No, it's just a common pen drive, but even if I add the usb SSD.... same result:

[ 9054.121539] usb 1-1.3: new high-speed USB device number 6 using xhci_hcd
[ 9054.222727] usb 1-1.3: New USB device found, idVendor=0781, idProduct=5590, bcdDevice= 1.00
[ 9054.222742] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9054.222749] usb 1-1.3: Product: Ultra
[ 9054.222755] usb 1-1.3: Manufacturer: SanDisk
[ 9054.222760] usb 1-1.3: SerialNumber: 4C530001090311111191
[ 9056.509804] usb 2-2: new SuperSpeed USB device number 6 using xhci_hcd
[ 9056.530631] usb 2-2: New USB device found, idVendor=2537, idProduct=1068, bcdDevice= 1.00
[ 9056.530641] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9056.530648] usb 2-2: Product: NS1068
[ 9056.530653] usb 2-2: Manufacturer: Norelsys
[ 9056.530659] usb 2-2: SerialNumber: 0123456789ABCDE

Please consider that, until formatting and starting from zero my DietPi yesterday, I used the same configuration and disks on DietPi for 2 years and never need an external power source for the SSD

MichaIng, 2022-04-15:
Would not be the first time that a kernel upgrade (like 5.10 => 5.15 recently) made USB powered USB drives unreliable, it however never was even that it may have worked in 90% of cases 😉 . However a pen drive should work, at least when there is no additional 2.5" drive attached.

You did try the other USB ports, and a different USB cable, just to rule out a hardware issue, didn't you?

If so, since it is known to have caused issues by times, also since it caused additional power consumption during boot, please you try to disable initial turbo, via dietpi-config performance options or in /boot/config.txt commenting or setting initial_turbo=0.

I'm not sure whether UAS is actually a topic for pen drives and cannot imagine that a drive which didn't have problems with it suddenly has, but just to rule it out, please try to add a USB quirk for that drive to disable UAS. For this add `usb-storage.quirks=0781:5590:u` to the end of the line in /boot/cmsline.txt and reboot. Vendor ID:product ID taken from lsusb output.

Bluesign71, 2022-04-15:
Well, I tried 4 different pendrive, and the 4 different usb ports, no way. Same for the SSD (that btw should have a power consumption slighty higher than a pen drive, due to the absence of mechanical parts).

I changed cpu settings from "performance" to "powersaving", I added initial_turbo=0 as You said.

Now it see any kind of usb device when I reboot. Now I have 2 pendrive and the SSD 2.5''. All working great, and seen as sba, sbc and sbd. The big problem is that if I unplug one of them..... adieu.... no way to recognize it again if not rebooting !
And this is impossible to accept, especcially if I have some servers working..... (apache, blynk, mqtt, postgresql., nextcloud...)

So, what could I do? it's eventually possible to downgrade to a previous kernel without formatting again everything?

Joulinar, 2022-04-15:
did you tried to change PSU? And how does it behave using a single pen stick only? Are you able to plug/unplug the pen stick?

Of course it is possible to downgrade kernel according official RPI documentation.

https://github.com/Hexxeh/rpi-update#options

The following should downgrade to version 5.10.95.

**!! WARNING !! you do this on your own risk**

```
apt install rpi-update
sudo rpi-update 39821d33e777cde9ba1a3cc8a73cfdd62fbbd2de
reboot
```

Bluesign71, 2022-04-15:
No way Joulinar, even trying plugging and unplugging a single pen stick. I also tried changing PSU. Btw the PSU I'm using is the original 5V 3A --> 15 watt that is much more than needed by raspberry pi4.
So.... even if very disappointed... I'm downgrading the kernel right now and for sure I'll let You know what happens !

Joulinar, 2022-04-15:
we might need to report to RPI guys if it is the kernel.

Bluesign71, 2022-04-15:
We might say to RPI guys to have an Happy Easter... 'cause looks like is not the Kernel, damn !!! 
What can I try? do I update back to last kernel ? or try to go back to an older one ?

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *      8192   270335   262144  128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      270336 62521343 62251008 29,7G 83 Linux
root@DietPi:~# uname -r
5.10.95-v8+
root@DietPi:~#

MichaIng, 2022-04-15:
Have you tried the USB quirks already I suggested above? Add additional ones with vendor and product ID obtained from lsusb.

Or blacklist UAS all together:
```
echo 'blacklist uas' > /etc/modprobe/disable-uas.conf
reboot
```

Bluesign71, 2022-04-15:
Ok, sorry if I didn't answered when You suggested that.

1. I don't have cmsline.txt in /boot.   Should I touch it ?
2. I created disable-uas.conf inside /etc/modprobe.d     It's correct? maybe not....   btw if it was right it didnt' work as well.

MichaIng, 2022-04-15:
Ah sorry, two typos: /boot/cmdline.txt and /etc/modprobe.d, yes. Okay, if the latter does not work (verify via "lsmod" that no "uas" is listed) then the USB quirks won't either.

Maybe it's not a kernel issue then, but we may still report to RPi devs (and wish happy Easter in the same turn): https://github.com/raspberrypi/firmware/issues
Probably someone there has further ideas how to debug. Or RPi forum: https://forums.raspberrypi.com/

Bluesign71, 2022-04-15:
SOLVED !!!!!!!!!!   The problem is the software "VIRTUALHERE" installed by dietpi-software

Please excuse me, so much, I should have guessed before.  But again, it's about 2 years I was using VirtualHere on my DietPi disto, never had a problem.
For sure, at the time, I had installed from the file downloaded from their site and not from dietpi-software.  
I'm so sorry for all the time You spent for me when it was so easy to think about.........
Disinstalled VirtualHere and everythings works.
Now I just have to understand how to make it works, 'cause I really need it for my dongle usb pens (software license)
Thank You very much guys. You're great and DIETPI remain the best one
I think You can close issue
I wish You and Joulinar an Happy Easter.

Paolo

Joulinar, 2022-04-15:
hmm we don't do any magic an VirtualHere. We download the executable only and create a service. Nothing else.

Joulinar, 2022-04-15:
Did you tried `sudo rpi-update`?

Bluesign71, 2022-04-15:
ops... thanks!

Joulinar, 2022-04-15:
I had a look to VirtualHere own install script. Basically we do exactly same.

https://github.com/virtualhere/script/blob/main/install_server

If you like, you could try to install VirtualHere as it is described on their GitHub

To install

```
curl https://raw.githubusercontent.com/virtualhere/script/main/install_server | sudo sh
```

To uninstall

```
curl https://raw.githubusercontent.com/virtualhere/script/main/uninstall_server | sudo sh
```

Would be interesting to know if it is behaving same.

MichaIng, 2022-04-16:
Here are actually some optimised builds for specific CPUs: https://www.virtualhere.com/usb_server_software
E.g. for Raspberry P 4 Cortex-A72 64-bit kernel:
```sh
curl -sSfL 'https://www.virtualhere.com/sites/default/files/usbserver/vhusbdarm64a72' -o /etc/vhusbd/vhusbd
systemctl restart virtualhere
```
When I find some time, I can add those to be installed by DietPi-Software as well. However, not sure if there is any recognisable difference.

Bluesign71, 2022-04-16:
ok, for sure on tuesday I'll try to install vhusbd and I'll let You know. I really need that service, travelling often it's damn usefull to have an usb dongle server, before to know about that I always forgot one of two usb dongles without the possibility to work...
Thks again

Bluesign71, 2022-04-19:
I confirm that I was unable to have a working VirtualHere even using the server downloaded from VirtualHere site. It installs correctly and works but the usb devices works only by the remote client connected to the virtualhere server on DietPi.
No way to mount devices on the raspberry.
Maybe the name of this issue (for me) should change to an VirtualHere issue ? 
I would like to know how to solve, if possible, 'cause can't really stay without virtualhere.
All the best,

Joulinar, 2022-04-19:
Probably the issue might need to be addressed to VirtualHere as it seems to be a behaviour of their software.

Bluesign71, 2022-04-19:
Done, will see what they say

Joulinar, 2022-04-19:
Not sure why but I overlooked a warning we already have on our docs https://dietpi.com/docs/software/remote_desktop/#virtualhere

Basically a know behaviour of VirtualHere 🙄 

As far as I can see, @MichaIng is going to adjust our docs https://github.com/MichaIng/DietPi-Docs/pull/715/files
There seems to be an option `IgnoredDevices` to have USB drives ignored by VirtualHere.

Bluesign71, 2022-04-20:
Yes... now it's clear.... I should have read before.
This is the answer I had form VirtualHere support:

```
_VirtualHere suspends access to new USB devices locally on the pi by default when it is running.

You can disable this behavior so the device(s) will be available locally on the pi if its not being used remotely.

You need to do this:

1. Stop the virtualhere server process
2. Edit the config.ini file on the server
3. Add the line
AutoAttachToKernel=1
4. Save the config.ini file and start the server and now the device.

(https://www.virtualhere.com/configuration_faq)_
```
I think everything's clear now!
Thks again for your time

Joulinar, 2022-04-20:
@MichaIng 
Is that something we should configure from our end to avoid such situation?

MichaIng, 2022-04-20:
But this means that newly attached USB devices are not available to VirtualHere OOTB, isn't it? I haven't tested this case yet, but when USB devices are attached already, they are available to the host until one "uses" them explicitly from a VirtualHere client. Is this different when attaching a new USB device while the VirtualHere server is running? If so, then yes I agree it makes more sense to prevent this by default, so that USB devices are always only reserved by VirtualHere when explicitly used by clients.