MichaIng/DietPi #5435: blue screen on: `systemctl disable --now hciuart`

Mausy5043, 2022-04-16:
### bug report/issue
[image]

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bullseye 0
```
- Kernel version | `uname -a`
```
Linux rbair 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
RPi 3 Model B+ (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)
`N/A`
- SD card used | (EG: SanDisk ultra)
`N/A`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
DietPi
```
[ INFO ] DietPi-PREP | APT install for: --reinstall base-files, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 70.1 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 base-files arm64 11.1+deb11u3 [70.1 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 70.1 kB in 0s (370 kB/s)
(Reading database ... 16183 files and directories currently installed.)
Preparing to unpack .../base-files_11.1+deb11u3_arm64.deb ...
Unpacking base-files (11.1+deb11u3) over (11.1+deb11u3) ...
Setting up base-files (11.1+deb11u3) ...

Configuration file '/etc/issue', does not exist on system.
Installing new config file as you requested.

Configuration file '/etc/issue.net', does not exist on system.
Installing new config file as you requested.

Configuration file '/etc/update-motd.d/10-uname', does not exist on system.
Installing new config file as you requested.
[  OK  ] DietPi-PREP | APT install for: --reinstall base-files
[  OK  ] DietPi-PREP | /var/lib/dpkg/info/base-files.postinst configure
[  OK  ] DietPi-PREP | apt-get clean
[ INFO ] DietPi-PREP | Deleting list of known users and groups, not required by DietPi
userdel: user pi is currently used by process 780
[  OK  ] DietPi-PREP | Creating DietPi user account
[ INFO ] DietPi-PREP | Removing misc files/folders/services, not required by DietPi
[  OK  ] DietPi-PREP | rm -R /var/lib/bluetooth
[  OK  ] DietPi-PREP | rm -Rf /var/lib/dhcp/dhclient.wlan0.leases /var/lib/dhcp/.??* /var/lib/dhcp/.[^.]*
[  OK  ] DietPi-PREP | rm -f /var/lib/misc/*.leases
[  OK  ] DietPi-PREP | rm -Rf /var/backups/alternatives.tar.0 /var/backups/dpkg.arch.0 /var/backups/dpkg.arch.1.gz /var/backups/dpkg.arch.2.gz /var/backups/dpkg.arch.3.gz /var/backups/dpkg.arch.4.gz /var/backups/dpkg.diversions.0 /var/backups/dpkg.diversions.1.gz /var/backups/dpkg.diversions.2.gz /var/backups/dpkg.diversions.3.gz /var/backups/dpkg.diversions.4.gz /var/backups/dpkg.statoverride.0 /var/backups/dpkg.statoverride.1.gz /var/backups/dpkg.statoverride.2.gz /var/backups/dpkg.statoverride.3.gz /var/backups/dpkg.statoverride.4.gz /var/backups/dpkg.status.0 /var/backups/dpkg.status.1.gz /var/backups/dpkg.status.2.gz /var/backups/dpkg.status.3.gz /var/backups/dpkg.status.4.gz /var/backups/.??* /var/backups/.[^.]*
[  OK  ] DietPi-PREP | rm -f /etc/*.org
[  OK  ] DietPi-PREP | rm -Rf /var/www/* /var/www/.??* /var/www/.[^.]*
[  OK  ] DietPi-PREP | rm -Rf /usr/src/* /usr/src/.??* /usr/src/.[^.]*
[... 217 lines cut ...]
  export GITBRANCH='master'
  export IMAGE_CREATOR='Mausy5043'
  export PREIMAGE_INFO='re_install'
  export HW_MODEL=0
  export WIFI_REQUIRED=1
  export DISTRO_TARGET=6
  echo
  date  +"%Y.%m.%d %H:%M:%S"
  echo ""
  echo "Running script..."
  PREP_SCRIPT="/home/dietpi/PREP_SYSTEM_FOR_DIETPI.sh"
  bash "${PREP_SCRIPT}"
```

#### Expected behaviour

- Fully unattended execution of `PREP_SYSTEM_FOR_DIETPI.sh`

#### Actual behaviour

- Blue Screen when the script is trying to execute `systemctl disable --now hciuart`

#### Extra details

Changed the command to `echo systemctl disable --now hciuart` and the re-installation proceeded with no additional problems.

Mausy5043, 2022-04-16:
Probably this piece of code is causing this:

https://github.com/MichaIng/DietPi/blob/1519a5c82373aea95aed400221481562bcf967f3/dietpi/func/dietpi-set_hardware#L1190-L1195

Joulinar, 2022-04-16:
Why are you using the PREP script to setup DietPi on a Raspberry PI? We offer ready to use images for PRI SBC.

@MichaIng maybe we would need to add a check for all services if they're existing before trying to disable/delete.

Mausy5043, 2022-04-16:
@Joulinar I want to remotely (unattended) repurpose my headless Pi with a "fresh" install.   
I understand that old config files may remain. I don't care. I will take care of uninstalling and removing any self-installed stuff.    
But I just don't want to have to go to my Pi; take out the SD-card; go through all the hoops to flash a new image and copy over the `dietpi.txt` and `dietpi-wifi.txt`; then travel back to my Pi install the SD-card and power it up. Especially not when a simple ssh-session to start `screen` and running my reinstallation script could suffice.

Regardless, I think the PREP-script should ignore these kinds of "errors". As you suggested, check for existence of a service before disabling/deleting it and/or ignore any errors that might occur when removing the service.

Joulinar, 2022-04-16:
Well this was never an issue in the past but we noticed similar (not same) situations in the last weeks and we where already thinking on this.

Mausy5043, 2022-04-16:
> Well this was never an issue in the past

This wasn't an issue in the past for me either. Started with v8.3.

Joulinar, 2022-04-16:
I'm not 100% sure, it is just a guess. But maybe there was a change on the latest `systemd` package leading to the behaviour?

MichaIng, 2022-04-16:
It's strange since that command is only called when the service is either enabled `systemctl is-enabled` or active `systemctl is-active`, and I'm not sure how it is possible that one of these is true if the unit file does not exist 🤔.

Mausy5043, 2022-04-17:
Bluetooth is enabled:
```
$ systemctl status hciuart.service
● hciuart.service - Configure Bluetooth Modems connected by UART
     Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-04-17 12:17:40 CEST; 8s ago
    Process: 321 ExecStart=/usr/bin/btuart (code=exited, status=0/SUCCESS)
   Main PID: 538 (hciattach)
      Tasks: 1 (limit: 1070)
        CPU: 149ms
     CGroup: /system.slice/hciuart.service
             └─538 /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow - b8:27:eb:a0:b2:81

Apr 17 12:17:33 rbian systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Apr 17 12:17:40 rbian btuart[348]: bcm43xx_init
Apr 17 12:17:40 rbian btuart[348]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Apr 17 12:17:40 rbian btuart[348]: Set BDADDR UART: b8:27:eb:a0:b2:81
Apr 17 12:17:40 rbian btuart[348]: Set Controller UART speed to 3000000 bit/s
Apr 17 12:17:40 rbian btuart[348]: Device setup complete
Apr 17 12:17:40 rbian systemd[1]: Started Configure Bluetooth Modems connected by UART.
```

And yet it fails:
```
[  OK  ] DietPi-PREP | mv /var/lib/dietpi/dietpi-wifi.db /boot/dietpi-wifi.txt
[ INFO ] DietPi-PREP | Disabling Bluetooth by default
[ SUB1 ] DietPi-Set_hardware > bluetooth (disable)
[FAILED] DietPi-Set_hardware | systemctl disable --now hciuart
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
Warning: The unit file, source configuration file or drop-ins of tmp.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: The unit file, source configuration file or drop-ins of -.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: The unit file, source configuration file or drop-ins of var-log.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: The unit file, source configuration file or drop-ins of boot.mount changed on disk. Run 'systemctl daemon-reload' to reload units.
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 354f7f38-f98f-47b8-805b-469d7727b02c

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/phpbb/viewforum.php?f=11

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

Press any key to continue...
```
---
On a machine that doesn't have Bluetooth enabled:
```
pi@rbups ~ $ systemctl status hciuart.service
Unit hciuart.service could not be found.
```
Then there are also no problems:
```
[  OK  ] DietPi-PREP | mv /var/lib/dietpi/dietpi-wifi.db /boot/dietpi-wifi.txt
[ INFO ] DietPi-PREP | Disabling Bluetooth by default
[ SUB1 ] DietPi-Set_hardware > bluetooth (disable)
[  OK  ] DietPi-Set_hardware | Added setting dtoverlay=disable-bt to end of file /boot/config.txt
[  OK  ] bluetooth disable | Completed
```

MichaIng, 2022-04-17:
> On a machine that doesn't have Bluetooth enabled:
> Then there are also no problems:

Which shows that the service detection generally works. I bet some `systemctl daemon-reload` is required between APT autoremoval and disabling Bluetooth. I'll have a closer look. Since `systemctl disable/enable` implies a reload, we probably need to reorder those so that first one is disabled which is assured to exist, so the reload is done to have systemctl correctly show that hciuart does not exist anymore.

MichaIng, 2022-04-22:
I had a closer look, and actually Dropbear is enabled before, all DietPi-Services controlled services disabled and Cron masked. So there are plenty of `systemctl daemon-reload` implied before this check is done. So it shouldn't be possible that it gives the wrong result, as well as it never did on my RPi image generations.

To debug, when you run into the issue, open a subshell and do these checks:
```sh
systemctl -q is-enabled hciuart 2> /dev/null && echo 'systemd thinks hciuart is enabled'
systemctl -q is-active hciuart 2> /dev/null && echo 'systemd thinks hciuart is active'
systemctl status hciuart
```

I have a hunch that the service might not be stopped when the `pi-bluetooth` package is purged, although debhelper should actually assure that. Will do a test.

MichaIng, 2022-04-22:
Verified, the service is still running, even that the service file does not exist anymore. I don't think this is intended: https://github.com/RPi-Distro/pi-bluetooth/issues/34

Solved with: https://github.com/MichaIng/DietPi/commit/4b73d74
We do not stop the service anymore now, since it cannot be restarted until boot otherwise, so disabling and re-enabling it in the same session fails. This way it keeps running just until next boot session. It is now only disabled when `is-enabled` is true, which is not the case when done in PREP after the package was removed already.

closed 2022-04-22 by MichaIng