MichaIng/DietPi #5439: Segmentation fault with Resilio

msongz, 2022-04-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 1
- Kernel version | `uname -a`
Linux zerow 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi Zero W (armv6l)
- Power supply used | (EG: 5V 1A RAVpower)
normal 
- SD card used | (EG: SanDisk ultra)
SanDisk

#### Additional Information (if applicable)
- Software title | Resilio

#### Steps to reproduce

1. fresh setup zero w with dietpi
2. download https://download-cdn.resilio.com/stable/linux-armhf/resilio-sync_armhf.tar.gz
3. tar -zxvf resilio-sync_armhf.tar.gz
4. ./rslsync

#### Expected behaviour

- resilio run and could be access by zerow:8888

#### Actual behaviour

- Segmentation fault

#### Extra details

- these steps works on years ago
- but now stop working

Joulinar, 2022-04-17:
Looks like the software package you download did not fit tho the architecture of your SBC or the file is corrupted. I guess you would need to ask application vendor for the content download link of your software. 

I found a blog post have a different link. But I'm not sure how old it is.

https://www.resilio.com/blog/how-i-created-my-own-personal-cloud-using-bittorrent-sync-owncloud-and-raspberry-pi

msongz, 2022-04-17:
@Joulinar using `wget https://download-cdn.getsync.com/2.5.12/linux-armhf/resilio-sync_armhf.tar.gz` and get the same result 😢

Joulinar, 2022-04-17:
As stated above, you might need to ask application vendor as it is their program not working.

MichaIng, 2022-04-17:
Likely the armhf binary is no "armv6hf" binary anymore, e.g. due to build stack change ARMv6 RPi models (RPi 1 and Zero) are not supported anymore. Yes, best to ask upstream developers.

MichaIng, 2022-04-17:
Found it: https://help.resilio.com/hc/en-us/articles/206178924

> For Raspberry Pi 1
> ```sh
> sudo dpkg --add-architecture armel && sudo apt-get update && sudo apt-get install resilio-sync:armel
> ```

This is for installation via APT repository. Key point is that the "armel" package is explicitly used for Raspberry Pi 1, instead of the "armhf" package. So the letter is now ARMv7 only, following Debian architecture, instead of Raspbian's. So you need to use the soft-float binary now, either via repository as of the link, or: https://download-cdn.resilio.com/stable/linux-arm/resilio-sync_arm.tar.gz

MichaIng, 2022-05-29:
Does it work with the soft-float binary?

MichaIng, 2022-07-23:
Marking as closed. Feel free to reopen if required.

closed 2022-07-23 by MichaIng