MichaIng/DietPi #5442: WIFI not working on RPI4

louigi600, 2022-04-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | it  won't start properly and has no network connectivity because wifi not working
    In any case I have a freshly imaged https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye.7z

- SBC model | RPi4 Model B revision 3 
- Power supply used | 5V 3A
- SD card used | 16Gb Samsung uSD

#### Additional Information (if applicable)
- Software title | NA
- Was the software title installed freshly or updated/migrated? it's just a fresh image
- Can this issue be replicated on a fresh installation of DietPi? I suppose it can if you have the same harware version

#### Steps to reproduce
1. Download the 64 bit image for RPI4
2. put the image on the uSD card with you favourite image tool (I use dd if=DietPi_RPi-ARMv8-Bullseye.img of=/dev/mmcblk0 bs=4096 or even cat DietPi_RPi-ARMv8-Bullseye.img  > /dev/mmcblk0 from root)
3. put the card i the RPi4
4. boot it and you will observe that the wifi is detected but not working

#### Expected behaviour
Be able to configure wifi connectivity and complete setup.

#### Actual behaviour
Cant configure wifi and on a wifi only network it is useless

Joulinar, 2022-04-18:
Did you configured WiFi inside `dietpi.txt` /`dietpi-wifi.txt` before first boot?

louigi600, 2022-04-19:
No I expected to be able to set it up from the dialog menu thing.
In any case if I fire up a subshell I can't see the integrated wifi card ... the RPi4 integrated BT/WIFI is a pain in the butt for the firmware.

Joulinar, 2022-04-19:
did you activated WiFi before first initial boot inside `dietpi.txt`?

```
AUTO_SETUP_NET_WIFI_ENABLED=1
```

Otherwise WiFi module will be disabled by the installer and you would need to turn it on manually.

louigi600, 2022-04-19:
Having the AUTO_SETUP_NET_WIFI_ENABLED=1 was not enough to unblock the situation
but this may be because I shutdown the installation edit the file  and then reboot installation.

Joulinar, 2022-04-19:
yes, these WiFi settings would need to be done before first initial boot on a fresh image.

louigi600, 2022-04-19:
but please: if wifi if off then the onboard wifi should not be reported as active (on) it is very misleading:
I thought it was telling me that there was no external wifi present while onboard wifi was on and active

MichaIng, 2022-04-20:
On first boot, onboard WiFi is enabled firmware-wise on RPi, so it should be possible to configure it when entering dietpi-config from the network check error prompt.

However, during first boot, onboard WiFi is set to be disabled from next boot on, if the `AUTO_SETUP_NET_WIFI_ENABLED` `dietpi.txt` setting is left `=0`. This should be visible in the dietpi-config menu as `Onboard WiFi : [Off]`. Toggle it to `[On]` to have onboard WiFi available as well after reboot.

So in short:
- On first boot, enter dietpi-config from the error prompt, set `Onboard WiFi : [On]`, then configure the WiFi connection.
- If it is not the first boot anymore, set `Onboard WiFi : [On]`, then reboot, then configure the WiFi connection from error prompt > dietpi-config.

Having network settings shown directly when not configured yet (instead of an error prompt) is on the ToDo.

closed 2022-05-27 by MichaIng