MichaIng/DietPi #5443: dietpi-update Error

francoisVdW, 2022-04-18:
#### Details:
- Date           | Mon Apr 18 10:47:47 CEST 2022
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi-PHP7 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
Extracting templates from packages: 50%W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
Extracting templates from packages: 100%
Preconfiguring packages ...
(Reading database ... 28831 files and directories currently installed.)
Preparing to unpack .../0-python-ldb_2%3a1.1.27-1+deb9u2_armhf.deb ...
  File "/usr/bin/pyclean", line 63
    except (IOError, OSError), e:
                             ^
SyntaxError: invalid syntax
dpkg: warning: subprocess old pre-removal script returned error exit status 1
dpkg: trying script from the new package instead ...
  File "/usr/bin/pyclean", line 63
    except (IOError, OSError), e:
                             ^
SyntaxError: invalid syntax
dpkg: error processing archive /tmp/apt-dpkg-install-97Ggh8/0-python-ldb_2%3a1.1.27-1+deb9u2_armhf.deb (--unpack):
 subprocess new pre-removal script returned error exit status 1
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 35, in <module>
    from debpython.version import SUPPORTED, debsorted, vrepr, \
  File "/usr/share/python/debpython/version.py", line 24, in <module>
    from ConfigParser import SafeConfigParser
ImportError: No module named 'ConfigParser'
dpkg: error while cleaning up:
 subprocess installed post-installation script returned error exit status 1
Preparing to unpack .../1-python2.7_2.7.13-2+deb9u6_armhf.deb ...
Unpacking python2.7 (2.7.13-2+deb9u6) over (2.7.13-2+deb9u5) ...
Preparing to unpack .../2-libpython2.7_2.7.13-2+deb9u6_armhf.deb ...
Unpacking libpython2.7:armhf (2.7.13-2+deb9u6) over (2.7.13-2+deb9u5) ...
Preparing to unpack .../3-libpython2.7-stdlib_2.7.13-2+deb9u6_armhf.deb ...
Unpacking libpython2.7-stdlib:armhf (2.7.13-2+deb9u6) over (2.7.13-2+deb9u5) ...
Preparing to unpack .../4-python2.7-minimal_2.7.13-2+deb9u6_armhf.deb ...
Unpacking python2.7-minimal (2.7.13-2+deb9u6) over (2.7.13-2+deb9u5) ...
Preparing to unpack .../5-libpython2.7-minimal_2.7.13-2+deb9u6_armhf.deb ...
Unpacking libpython2.7-minimal:armhf (2.7.13-2+deb9u6) over (2.7.13-2+deb9u5) ...
Preparing to unpack .../6-zlib1g-dev_1%3a1.2.8.dfsg-5+deb9u1_armhf.deb ...
Unpacking zlib1g-dev:armhf (1:1.2.8.dfsg-5+deb9u1) over (1:1.2.8.dfsg-5) ...
Preparing to unpack .../7-zlib1g_1%3a1.2.8.dfsg-5+deb9u1_armhf.deb ...
Unpacking zlib1g:armhf (1:1.2.8.dfsg-5+deb9u1) over (1:1.2.8.dfsg-5) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-97Ggh8/0-python-ldb_2%3a1.1.27-1+deb9u2_armhf.deb
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2022-04-18:
Basically you are running Debian Stretch still, which we don't support anymore. 

@MichaIng could you have a look pls

As well you have configured 2 times the same PHP repository

```
/etc/apt/sources.list.d/dietpi-php.list
/etc/apt/sources.list.d/php.list
```

Even if unrelated to this issue, something you should fix.

MichaIng, 2022-04-20:
To be sure, let's have a look into both files:
```sh
cat /etc/apt/sources.list.d/dietpi-php.list
cat /etc/apt/sources.list.d/php.list
```
The first is the one intended/installed by `dietpi-software` to enable PHP7.3 installs on Debian Stretch.

`pyclean` and `pycompile` are shipped with the `python-minimal` package. Let's try whether reinstalling it solves the syntax errors:
```sh
apt install --reinstall python-minimal
```

francoisVdW, 2022-04-20:
```
@DietPi-PHP7:~# cat /etc/apt/sources.list.d/dietpi-php.list
deb https://packages.sury.org/php/ stretch main
@DietPi-PHP7:~# cat /etc/apt/sources.list.d/php.list
deb https://packages.sury.org/php/ stretch main
```
I try to re-install python minimal. It fails as you can see below
```
Réception de:1 http://raspbian.42.fr/raspbian stretch/main armhf python-minimal armhf 2.7.13-2 [40,5 kB]
40,5 ko réceptionnés en 0s (69,7 ko/s)
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-php.list:1 and /etc/apt/sources.list.d/php.list:1
(Lecture de la base de données... 28831 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../python-ldb_2%3a1.1.27-1+deb9u2_armhf.deb ...
  File "/usr/bin/pyclean", line 63
    except (IOError, OSError), e:
                             ^
SyntaxError: invalid syntax
dpkg: avertissement: le sous-processus ancien script pre-removal a retourné une erreur de sortie d'état 1
dpkg: tentative d'exécution du script du nouveau paquet à la place...
  File "/usr/bin/pyclean", line 63
    except (IOError, OSError), e:
                             ^
SyntaxError: invalid syntax
dpkg: erreur de traitement de l'archive /var/cache/apt/archives/python-ldb_2%3a1.1.27-1+deb9u2_armhf.deb (--unpack) :
 le sous-processus nouveau script pre-removal a retourné une erreur de sortie d'état 1
Traceback (most recent call last):
  File "/usr/bin/pycompile", line 35, in <module>
    from debpython.version import SUPPORTED, debsorted, vrepr, \
  File "/usr/share/python/debpython/version.py", line 24, in <module>
    from ConfigParser import SafeConfigParser
ImportError: No module named 'ConfigParser'
dpkg: error while cleaning up:
 le sous-processus script post-installation installé a retourné une erreur de sortie d'état 1
Des erreurs ont été rencontrées pendant l'exécution :
 /var/cache/apt/archives/python-ldb_2%3a1.1.27-1+deb9u2_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
The default version of python  is 3.5.3
```
@DietPi-PHP7:~#  python --version
Python 3.5.3
```

MichaIng, 2022-04-20:
Does `pyclean` file as well when you call it from console? Probably it's the `python-ldb` package which doing a faulty `pyclean` call instead. Is something depending on it, i.e. what would be removed together with it?
```sh
apt purge python-ldb
```

francoisVdW, 2022-04-20:
Here are the 4 commands 
```
@DietPi-PHP7:~# which -a python
/usr/bin/python
@DietPi-PHP7:~# dpkg -S "$(command -v python)"
python-minimal: /usr/bin/python
@DietPi-PHP7:~# which -a python2
/usr/bin/python2
@DietPi-PHP7:~# which -a python3
/usr/local/bin/python3

```

MichaIng, 2022-04-20:
Okay, while `/usr/bin/python` is the intended path provided by the intended package, it identifies as Python 3, so I guess it points to the wrong executable. Can you please show:
```sh
readlink -f /usr/bin/python
```
If this is not `/usr/bin/python2.7`, then you can fix it with:
```sh
ln -sf python2.7 /usr/bin/python
```
This should then fix the APT upgrade:
```sh
apt upgrade
```
The question remains what broke this symlink. Since `/usr/local/bin/python3` exists, but no other `python3` executable in path, it seems to have been compiled manually, at least this is not provided by the native Debian `python3` package. Probably the install script for this custom Python 3 build also broke the `python` symlink. On some distros, indeed `python` is pointing to Python 3 nowadays, but on Debian it is expected to point to Python 2. Not sure why you or an installer you use (?) compiled Python 3 manually. Debian ships Python v3.5.3 just like this:
```sh
apt install python3
```
But if this is wanted, it's better to first uninstall the custom build, or doing it manually. This removes as well modules installed via `pip` and in case `pip` itself, so they'd need to be reinstalled.
```sh
rm /usr/local/bin/python3*
rm -R /usr/local/lib/python3*
```

francoisVdW, 2022-04-20:
**It works !**
I follow your instructions : 
- set the link to python27
- rm python3*
- apt upgrade

The "apt ugrade" works gracefully. 

Then I ran "apt install python3". Durring the python3's install one can see
```
python3 is already the newest version (3.5.3-1).
python3 passé en « installé manuellement ».
```
which means *python3 switched to "manually installed"* (as you guessed)

Once again, thank you for your time

Francois.

MichaIng, 2022-04-20:
Ah okay, strange that `which -a python3` showed `/usr/local/bin/python3` only, it should show all executables named `python3` in all PATH entries, and the `python3` package should ship `/usr/bin/python3` 🤔. Can you show this again?
```sh
which -a python3
```

However, great that the upgrade went through now 🙂.

francoisVdW, 2022-04-20:
Now (after remove, then apt install python3) the command  `which -a python3` responds *nothing*.  
Python3  is not in the path anymore.  I'm sorry I can not answer your question.

It's not a problem for me, I do call my python3's scripts with the full path of pyton3 (like ` /usr/bin/python3.5 ./my_script.py`)

MichaIng, 2022-04-20:
This should restore `/usr/bin/python3`:
```sh
apt install --reinstall python3-minimal
```

closed 2022-04-22 by MichaIng