MichaIng/DietPi #5444: qBittorrent fails to login to WebUI on fresh install

DetermineAbsurd, 2022-04-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.3.1`
- Distro version | `bullseye`
- Kernel version | `5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17) x86_64 GNU/Linux`
- SBC model | `Native PC (x86_64)`
- Power supply used | `2x 1000w Platinum Plus`
- SD card used | `N/A`

#### Additional Information (if applicable)
- Software title | `qBittorrent`
- Was the software title installed freshly or updated/migrated? `Yes`
- Can this issue be replicated on a fresh installation of DietPi? `Probably`

#### Steps to reproduce
1. Install qBittorrent via dietpi-software `dietpi-software install 46`
2. Attempt to access the WebUI via LAN address
3. Weblogin page loads, but does not accept the default credentials `qbittorrent dietpi`

#### Expected behaviour
- Login should be successful

#### Actual behaviour
- Page seems to just refresh and not login with any visible error

Joulinar, 2022-04-18:
The password doesn't need to be `dietpi`. It should be the global application password you defined during DietPi setup. Did you changed that password?

DetermineAbsurd, 2022-04-18:
I did, but according to [the docs](https://dietpi.com/docs/software/bittorrent/#qbittorrent) it uses the default password of "dietpi" because of the new hashing system implemented in qBittorrent.

DetermineAbsurd, 2022-04-18:
It's working now seemingly. I don't know what I did. Literally just closed the browser and reopened it to the WebUI and I was logged in. Don't know if it needs further investigation. It might be an upstream issue? I don't know lol.

MichaIng, 2022-04-20:
Indeed, since Bullseye, the hardcoded password "dietpi" is used initially: #5078
<details>
OpenSSL has a function to generate this new PKCS5_PBKDF2_HMAC hash, but it is not made available through the `openssl passwd` command, only via libssl library. There are bindings/functions for perl, PHP, Python and other programming languages, but I'm not keen to install any of these just to generate a hash a single time for the default login. Sadly I also didn't find a slim hasher utility which makes use of libssl directly and supports this algorithm, sadly. Programming one and compiling+distributing it for every architecture and Debian version is overkill as well.

There would be an option to use the qBittorrent web API: https://github.com/qbittorrent/qBittorrent/discussions/15454#discussioncomment-1313256
But since the server needs to run for this already, we'd then need to use a hardcoded password first. As we set a hardcoded password already, it's only starting the server, waiting for the web port to become available, sending the POST request for changing the password, stopping the server. Doesn't sound so hard on a second view 🤔.
</details>

> Literally just closed the browser and reopened it to the WebUI and I was logged in.

Hmm, sounds like only the redirection to the actual web UI didn't happen after login. I just tried to replicate on x86_64 VM, but it works well here. Probably the `/var/log/qbittorrent/qbittorrent.log` or the browser console give a hint when you face this issue again.

MichaIng, 2022-05-27:
I'll mark this as closed. Feel free to reopen if required.

closed 2022-05-27 by MichaIng

alucard87pl, 2022-12-09:
I just redid a fresh install of the newest DietPi and installed qbittorrent.
Doesn't recognize any combination of username (`qbittorrent`, `root`, `dietpi`) and password (tried the global software password and `dietpi` as recommended by the docs)

MichaIng, 2022-12-09:
Can you please open a new issue and fill the bug report template? This makes it easier for us to help debugging.