MichaIng/DietPi #5445: Unable to set persistent wlanX names

wassereimer86, 2022-04-18:
### Creating a bug report/issue

#### Required Information
- DietPi version 8.3.1
- Distro version bullseye
- Kernel version 5.15.32-v8
- SBC model RPi 4 Model B (aarch64)
- Power supply used offical RPi for RPi 4
- SD card used SanDisk ultra

I´ve tried different ways (not at the same time) to set persistent names for two wifi interfaces but it won´t work.

1.
[image]
[image]

2.
[image]
[image]

It just wont work. If the USB device is not connected, the internal wifi is wlan0. If the usb device is added, the internal wifi is wlan1.

Am i doing something wrong?

wassereimer86, 2022-04-20:
Hey @MichaIng and thank you for your time and thoughts on this.

First of all i want to say that i´m no expert in this and that i´ve read a lot of things to come to the solutions in the first post - even that they don´t work.

> The udev rules actually look good, also the priority so that the renaming is done before ifupdown hotplug setup. Did you verify that these are triggered as intended, matching the driver names? Also what I am not sure is whether there can be a collision, e.g. the external adapter is added first, initially with the name `wlan0`, the onboard adapter shortly after, then for some reason the udev rules loop for the onboard adapter is processed faster so that it is tried to be renamed to `wlan0` while that name is in use already.

How can i verify that the rules are triggered as intended? I´ve double checked the driver names with `lsmod`.

> I never really understood why systemd renames interfaces without `systemd-networkd` being used, but it did, hence the `99-default.link` to `/dev/null` to prevent this from happen. It shouldn't play a role when there are other matching configs, but did you try to remove this symlink so that your two configs are the only ones?

Should i test this only with my second solution or with the udev rules applied? I did it without them and it didn´t work. Still wlan1 is used if pluged in at boot.

> Another solution would be to enable native predictable interface naming, just note that `dietpi-config` network setup and some cases where `dietpi-software` tries to find a WiFi device do not work anymore. For this, remove `net.ifnames=0` from `/boot/cmdline.txt`. For older kernels or to apply different naming schemes, there are these udev and systemd based renaming mechanisms. To restore them as well and hence assure Debian default naming, remove as well the `/etc/udev/rules/80-net-setup-link.rules` and `/etc/systemd/network/99-default.link` symlinks, which are masks for the actual udev/systemd(-networkd) default rules/config.

I really want to use dietpi as it is. I love how easy (and minimal ressource usage) everything is even if i´m no absolute linux/server beginner. I hope to find a working solution for it. So this is not a good option for me,

MichaIng, 2022-04-20:
Could you verify the driver name? It looks correct, matching the kernel module name, but let's assure that:
```sh
udevadm info /sys/class/net/wlan0
udevadm info /sys/class/net/wlan1
```

wassereimer86, 2022-04-21:
```
udevadm info /sys/class/net/wlan0
P: /devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0
L: 0
E: DEVPATH=/devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/net/wlan0
E: DEVTYPE=wlan
E: INTERFACE=wlan0
E: IFINDEX=3
E: SUBSYSTEM=net
E: USEC_INITIALIZED=2500878
E: ID_NET_NAMING_SCHEME=v247
E: ID_NET_NAME_MAC=wlxdca632488538
E: ID_OUI_FROM_DATABASE=Raspberry Pi Trading Ltd
E: SYSTEMD_ALIAS=/sys/subsystem/net/devices/wlan0
E: TAGS=:systemd:
E: CURRENT_TAGS=:systemd:
```

Here i can´t find the driver information.

```
udevadm info /sys/class/net/wlan1
P: /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/net/wlan1
L: 0
E: DEVPATH=/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/net/wlan1
E: DEVTYPE=wlan
E: INTERFACE=wlan1
E: IFINDEX=4
E: SUBSYSTEM=net
E: USEC_INITIALIZED=41394366
E: ID_NET_NAMING_SCHEME=v247
E: ID_NET_NAME_MAC=wlx00c0caaaebde
E: ID_OUI_FROM_DATABASE=ALFA, INC.
E: ID_NET_NAME_PATH=wlp1s0u1u1
E: ID_VENDOR=Realtek
E: ID_VENDOR_ENC=Realtek
E: ID_VENDOR_ID=0bda
E: ID_MODEL=802.11n_NIC
E: ID_MODEL_ENC=802.11n\x20NIC
E: ID_MODEL_ID=8812
E: ID_REVISION=0000
E: ID_SERIAL=Realtek_802.11n_NIC_123456
E: ID_SERIAL_SHORT=123456
E: ID_TYPE=generic
E: ID_BUS=usb
E: ID_USB_INTERFACES=:ffffff:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=rtl88XXau
E: ID_VENDOR_FROM_DATABASE=Realtek Semiconductor Corp.
E: ID_MODEL_FROM_DATABASE=RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
E: SYSTEMD_ALIAS=/sys/subsystem/net/devices/wlan1
E: TAGS=:systemd:
E: CURRENT_TAGS=:systemd:
```

It seems to be `rtl88XXau` instead of `88XXau` what `lsmod` said?

wassereimer86, 2022-04-21:
I've re-applied my first solution with the correct driver for the external Wi-Fi, and now it works. But the first driver doesn't seem to be correct, right? If it were, it should have worked and assigned it with the rule. I think it's working now because the second rule is correct. How can I fix the rule for the internal Wi-Fi?

MichaIng, 2022-04-22:
The `DRIVER` variable is not set at all, so it cannot be used as filter in the udev rule. This matches my systems. It seems like network interfaces themselves do not have the adapter drivers assigned like this. Indeed it is found in `ID_USB_DRIVER` (not the name of the kernel module itself, but more a pattern for modules which are searched for when this adapter is attached), but only for the USB WiFi adapter.

To get a full list of all attributes:
```sh
udevadm info -a /sys/class/net/wlan0
udevadm info -a /sys/class/net/wlan1
```
This will also list all info about all parent devices, but you need the final wlan0/wlan1 device's attributes only. You should also see the empty `DRIVER=""` variable there. One of these attributes should work for you, e.g. the MAC address `ATTR{address}`, or `ID_USB_DRIVER` for the USB adapter at least.

wassereimer86, 2022-04-22:
Funny. Both devices have an empty `DRIVER==""`. But why does it work now? Only the parent device of `wlan1` has `DRIVERS=="rtl88XXau"`. And in my first try (see first post) i had the same setting for `wlan0` and it didn´t work. The problem with the MAC address is that it will change so i can´t use that.

Thanks again for your time!

```
udevadm info -a /sys/class/net/wlan1

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.0/net/wlan1':
    KERNEL=="wlan1"
    SUBSYSTEM=="net"
    DRIVER==""
    ATTR{addr_assign_type}=="0"
    ATTR{addr_len}=="6"
    ATTR{address}=="00:c0:ca:aa:eb:de"
    ATTR{broadcast}=="ff:ff:ff:ff:ff:ff"
    ATTR{carrier_changes}=="0"
    ATTR{carrier_down_count}=="0"
    ATTR{carrier_up_count}=="0"
    ATTR{dev_id}=="0x0"
    ATTR{dev_port}=="0"
    ATTR{flags}=="0x1002"
    ATTR{gro_flush_timeout}=="0"
    ATTR{ifalias}==""
    ATTR{ifindex}=="3"
    ATTR{iflink}=="3"
    ATTR{link_mode}=="0"
    ATTR{mtu}=="2312"
    ATTR{name_assign_type}=="4"
    ATTR{napi_defer_hard_irqs}=="0"
    ATTR{netdev_group}=="0"
    ATTR{operstate}=="down"
    ATTR{power/control}=="auto"
    ATTR{power/runtime_active_time}=="0"
    ATTR{power/runtime_status}=="unsupported"
    ATTR{power/runtime_suspended_time}=="0"
    ATTR{proto_down}=="0"
    ATTR{queues/rx-0/rps_cpus}=="0"
    ATTR{queues/rx-0/rps_flow_cnt}=="0"
    ATTR{queues/rx-1/rps_cpus}=="0"
    ATTR{queues/rx-1/rps_flow_cnt}=="0"
    ATTR{queues/rx-2/rps_cpus}=="0"
    ATTR{queues/rx-2/rps_flow_cnt}=="0"
    ATTR{queues/rx-3/rps_cpus}=="0"
    ATTR{queues/rx-3/rps_flow_cnt}=="0"
    ATTR{queues/tx-0/byte_queue_limits/hold_time}=="1000"
    ATTR{queues/tx-0/byte_queue_limits/inflight}=="0"
    ATTR{queues/tx-0/byte_queue_limits/limit}=="0"
    ATTR{queues/tx-0/byte_queue_limits/limit_max}=="1879048192"
    ATTR{queues/tx-0/byte_queue_limits/limit_min}=="0"
    ATTR{queues/tx-0/traffic_class}=="0"
    ATTR{queues/tx-0/tx_maxrate}=="0"
    ATTR{queues/tx-0/tx_timeout}=="0"
    ATTR{queues/tx-0/xps_cpus}=="0"
    ATTR{queues/tx-0/xps_rxqs}=="0"
    ATTR{queues/tx-1/byte_queue_limits/hold_time}=="1000"
    ATTR{queues/tx-1/byte_queue_limits/inflight}=="0"
    ATTR{queues/tx-1/byte_queue_limits/limit}=="0"
    ATTR{queues/tx-1/byte_queue_limits/limit_max}=="1879048192"
    ATTR{queues/tx-1/byte_queue_limits/limit_min}=="0"
    ATTR{queues/tx-1/traffic_class}=="0"
    ATTR{queues/tx-1/tx_maxrate}=="0"
    ATTR{queues/tx-1/tx_timeout}=="0"
    ATTR{queues/tx-1/xps_cpus}=="0"
    ATTR{queues/tx-1/xps_rxqs}=="0"
[... 420 lines cut ...]
    ATTRS{driver_override}=="(null)"
    ATTRS{power/control}=="on"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"

  looking at parent device '/devices/platform/soc':
    KERNELS=="soc"
    SUBSYSTEMS=="platform"
    DRIVERS=="simple-pm-bus"
    ATTRS{driver_override}=="(null)"
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"

  looking at parent device '/devices/platform':
    KERNELS=="platform"
    SUBSYSTEMS==""
    DRIVERS==""
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"
```

MichaIng, 2022-04-22:
If the USB adapter is detected first (obviously it is when it gets `wlan0` by default), it is now as of the filter renamed to `wlan1`, so the afterwards detected onboard adapter gets the free `wlan0` name. At least this sounds logic to me 🙂. But indeed best is to have a matching udev rule for both, just to be safe.

wassereimer86, 2022-04-22:
Thank you for your help!

closed 2022-04-22 by wassereimer86

MichaIng, 2022-04-23:
Btw, udev verbosity can be generally increased by setting `udev_log=debug` in `/etc/udev/udev.conf``. You should then very detailed which rule got parsed on each hardware event in `journalctl`. Interesting once, but much to verbose to keep this in production 😉.