MichaIng/DietPi #5454: dietpi-commands not working

geol, 2022-04-23:
it has happened for weeks. when i ran any command like "sudo dietpi-cofnig" from /boot/dietpi, it says command not found. it used to work fine.
[image]

MichaIng, 2022-04-23:
Many thanks for reporting.

I guess this is not `bash` shell, is it? `dietpi-*` commands are not added via PATH but as `alias`es to bash explicitly. There are some ideas to add zsh, fish and other shells as software options and then add aliases for them as well: #5211

With `/boot/dietpi` added to PATH it should however work in every shell 🤔. How it is called should be visible via:
```sh
command -v dietpi-config
```

MichaIng, 2022-05-27:
I'll mark this as closed. Feel free to reopen if required.

closed 2022-05-27 by MichaIng

massimopavoni, 2025-04-30:
I think this might still be relevant: after installing fish and updating my user's shell, the behavior is the following for most scripts.

:white_check_mark: `dietpi-config` asks for sudo execution (meaning aliases from `/etc/fish/conf.d/dietpi.fish` are loaded)
:white_check_mark: `sudo su` followed by `dietpi-config` correctly shows the requested TUI
:white_check_mark: `sudo /boot/dietpi/dietpi-config` also works
:red_circle: `sudo dietpi-config` fails with `sudo: dietpi-config: command not found`

I'm not quite sure what is causing the issue, as I haven't dived deeper into fish configuration options in general, but still thought I should point the problem out, despite the age of the issue.
If there's any way I can help further investigate, do let me know.

MichaIng, 2025-04-30:
Yeah we recognized already.  bash requires `sudo` to be an alias to call other aliases. fish seems to work differently. Can you try to unset the `sudo` alias:
```sh
unalias sudo
sudo dietpi-launcher
```

massimopavoni, 2025-04-30:
Tried `functions -e sudo` for fish, as fish doesn't have `unalias`, but it seems the problems persists.
I also attempted to fix by manually commenting the sudo alias from `/etc/fish/conf.d/dietpi.fish`, but that also does not help.

[This issue](https://github.com/fish-shell/fish-shell/issues/7044) from the shell's repo might be pertinent; I'm not sure which commenter has the best idea/approach, but the first easy thing being mentioned is using abbreviations instead of aliases.

MichaIng, 2025-04-30:
Wow, that turns out way more complex than I was hoping for. I think we should instead just move DietPi scripts into PATH 😄. They are below `/boot/dietpi` for legacy reasons only.