MichaIng/DietPi #5455: /etc/bash.bashrc file has become corrupted from power failure

abienz, 2022-04-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.3.1
- Distro version | dietpi
- Kernel version | Linux DietPi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
- SBC model | RPi4
- Power supply used | Official RPi adapter
- SD card used | Samsung 500TB

#### Steps to reproduce
1. Power failure
2. SD became slightly corrupted, and I am no longer greeted by dietpi welcome messages when SSH in

#### Expected behaviour
Dietpi welcome screen when SSH in, and access to normal dietpi commands

#### Actual behaviour
Here is my current /etc/bash.bashrc file, you can see the last line is corrupted
`dietpi:~$ cat /etc/bash.bashrc 
# System-wide .bashrc file for interactive bash(1) shells.

# To enable the settings / commands in this file for login shells as well,
# this file has to be sourced in /etc/profile.

# If not running interactively, don't do anything
[ -z "$PS1" ] && return

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, overwrite the one in /etc/profile)
# but only if not SUDOing and have SUDO_PS1 set; then assume smart user.
if ! [ -n "${SUDO_USER}" -a -n "${SUDO_PS1}" ]; then
  PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi

# Commented out, don't overwrite xterm -T "title" -n "icontitle" by default.
# If this is an xterm set the title to user@host:dir
#case "$TERM" in
#xterm*|rxvt*)
#    PROMPT_COMMAND='echo -ne "\033]0;${USER}@${HOSTNAME}: ${PWD}\007"'
#    ;;
#*)
#    ;;
#esac

# enable bash completion in interactive shells
#if ! shopt -oq posix; then
#  if [ -f /usr/share/bash-completion/bash_completion ]; then
#    . /usr/share/bash-completion/bash_completion
#  elif [ -f /etc/bash_completion ]; then
#    . /etc/bash_completion
#  fi
#fi

# if the command-not-found package is installed, use it
if [ -x /usr/lib/command-not-found -o -x /usr/share/command-not-found/command-not-found ]; then
	function command_not_found_handle {
	        # check because c-n-f could've been removed in the meantime
                if [ -x /usr/lib/command-not-found ]; then
		   /usr/lib/command-not-found -- "$1"
                   return $?
                elif [ -x /usr/share/command-not-found/command-not-found ]; then
		   /usr/share/command-not-found/command-not-found -- "$1"
                   return $?
		else
		   printf "%s: command not found\n" "$1" >&2
		   return 127
		fi
	}
fi
r i i in /etc/bashrc.d/*.sh /etc/bashrc.d/*.bash; do [ -b "d�b ]h&& n %xdooe; }~set2
`

#### Extra details
I can't find a good version of this file online, the repo only has https://github.com/MichaIng/DietPi/blob/master/rootfs/etc/bashrc.d/dietpi.bash which isn't the file I'm looking for.

I guess I'm asking for two things, is it possible to reinstall dietpi over SSH, without re-flashing the SD card and losing all my data?

Is it possible for someone to share with me their good /etc/bash.bashrc file?

Joulinar, 2022-04-26:
Hi, there is no need to open the topic twice as you already raised that question on the forum. you could download an image for your SBC, open it with a tool like 7zip and extract the file needed.

abienz, 2022-04-26:
ah, thanks I wasn't sure the two were reaching the same audience.

abienz, 2022-04-26:
The file I need is also not in the image file.

closed 2022-04-26 by abienz

Joulinar, 2022-04-26:
Are you sure? For me it is there

[image]

abienz, 2022-04-26:
I'm not sure what you're showing me there?

I downloaded the RPi installation from the dietpi website, when I open the zip, this is what's inside.

[image]

Inside the img file is the following

[image]

Inside the dietpi folder is the following

[image]

I can't find the bashrc file

Joulinar, 2022-04-26:
Not sure what tool you are using but I'm using 7zip. And if I open the .img file, I see 2 partitions. You need to select the `1.img` folder. There you will see al files located on the ext4 partition. `0.fat` is the boot partition only.

[image]

abienz, 2022-04-26:
Ah thanks, I didn't notice to look there, I have updated my bashrc file and now it works as expected.