MichaIng/DietPi #5457: systemd-timesyncd FAILED

PrzemekSkw, 2022-04-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`

- Distro version | `bullseye`
- Kernel version | `Linux DietPi 5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17) x86_64 GNU/Linux`
- Intel x64 MiniPC - HP
- Power supply used | 64W
- SSD

- Bug report ID | `echo $G_HW_UUID` `5f414704-3a2c-413d-8ada-f79bfd4a0f80`

#### Steps to reproduce
1.  `sudo apt update`

```
#### Details:
- Date           | Tue Apr 26 08:06:58 CEST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Debian mini.iso
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi 5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y update`
- Exit code      | 100
- Software title | DietPi-Config

#### Additional logs:
Hit:1 https://deb.debian.org/debian bullseye InRelease
Get:2 https://download.docker.com/linux/debian bullseye InRelease [43.3 kB]
Hit:3 https://download.mono-project.com/repo/debian buster InRelease
Get:4 https://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Hit:5 https://apt.sonarr.tv/debian buster InRelease
Hit:6 https://apt.hrfee.dev trusty InRelease
Hit:7 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:8 https://www.collaboraoffice.com/repos/CollaboraOnline/CODE-debian10 ./ InRelease
Get:9 https://apt.ombi.app/develop jessie InRelease [10.4 kB]
Hit:10 https://linux-clients.seafile.com/seadrive-deb/buster stable InRelease
Hit:11 https://packagecloud.io/ookla/speedtest-cli/debian bullseye InRelease
Reading package lists...
E: Release file for https://deb.debian.org/debian/dists/bullseye-updates/InRelease is not valid yet (invalid for another 2h 6min 50s). Updates for this repository will not be applied.

---------------------------------------------------------------------
[FAILED] DietPi-Config | Unable to continue, DietPi-Config will now terminate.
```

There's wrong time in system:
```
timedatectl
               Local time: wto 2022-04-26 08:09:22 CEST
           Universal time: wto 2022-04-26 06:09:22 UTC
                 RTC time: wto 2022-04-26 12:20:02
                Time zone: Europe/Warsaw (CEST, +0200)
System clock synchronized: no
              NTP service: n/a
          RTC in local TZ: no
```
Regards.

Joulinar, 2022-04-26:
Hi,

can you check the status

systemctl status systemd-timesyncd
journalctl -u systemd-timesyncd

And to force a time sync

```
/boot/dietpi/func/run_ntpd 1
```

PrzemekSkw, 2022-04-26:
Hi, @Joulinar I install systemd-timesyncd and now show me good time. 
Thanks for help.

closed 2022-04-26 by PrzemekSkw

Joulinar, 2022-04-26:
Did you have done an upgrade from Buster to Bullseye before?

PrzemekSkw, 2022-04-26:
> Did you have done an upgrade from Buster to Bullseye before?

Yes.

Joulinar, 2022-04-26:
Did you followed our blog post? Because there is a hint to install `systemd-timesyncd` after the upgrade again.

PrzemekSkw, 2022-04-26:
Meybe I missed that sorry.

Joulinar, 2022-04-26:
no problem, good it is working again.