MichaIng/DietPi #5461: AdGuardHome blocks DietPi Update

whyisthisbroken, 2022-04-28:
Hi, i´ve installed AGH manually (not via Dietpi-Software) and add it to the Dietpi-Services. (systemd controlled)
The Problem is, that i can´t update Dietpi without remove AGH from the Services. 
DietPi stopp the added service and the Update fails cause no DNS Resolution.

What can I do to prevent the AGH service from stopping when an update is available? Excluding it will completely remove it from dietpi services ? 

#### Required Information
- DietPi version | 8.4.1 Beta
- Distro version | bullseye
- Kernel version | Linux DietPi 5.15.32-v8+
- SBC model | RPi 4 Model B (aarch64)

#### Additional Information (if applicable)
- Software title | AdGuardHome
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Add AdGuardHome manually to dietpi-services
2. If u want to update dietpi, AdGuardHome would be stopped (service) and therefore no DNS Resolution.

#### Expected behaviour

- Exclude AdGuardHome from stopping service. 

#### Actual behaviour

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : AdGuardHome <- This is the Problem
[  OK  ] DietPi-Services | stop : cron

#### Extra details

- U can solve the problem and remove AdGuard from the Dietpi-Services again to perform the Update and add it again later

Joulinar, 2022-04-28:
Basically 3 thinks you could adjust

1) don't set DNS server on DietPi themselves to AGH. There is usually no need to for a DietPi system to run via an adblock like AGH. Usually DietPi will just call Debian package source and GitHub. Therefore use a global public upstream DNS porovider inside DietPi network settings. This will have influence in AGH settings or how your LAN clients resolve DNS request. This would help as well if AGH would be failing an whatever reason. DietPi could continue to resolve it's own DNS request and you would be able to install new software or peform some fixing. At least this is our recommendation, as long as you don't use any desktop.
2) use `dietpi-software` to install AGH. Benefit would be to have the correct settings for AGH service ootb. Because we configure the way, that it will be excluded from stopping.
3) in your case, open `dietpi-services`, select AGH and set it to `Mode = Systemd controlled` + `Include/Exclude = excluded`. This should exclude the service from being stopped and give you full control over it.

whyisthisbroken, 2022-04-28:
Thx - i set the DNS provider from DietPi to Cloudflare. That makes sense.. It's embarrassing that I didn't think of it myself xD
I will test it on the Next Update :)

closed 2022-04-28 by whyisthisbroken

Joulinar, 2022-04-28:
theoretically you could force the current update to be installed again

```
dietpi-update -1
```

whyisthisbroken, 2022-04-28:
Yep - it works.
But tell me if I install AGH with Dietpi software - is there an option to select beta channel and x64 for installation?
I think for now the 32-bit ARMv6 will be installed right?

I didn't find these options, so I installed AGH manually.

Joulinar, 2022-04-28:
We will install AGH stable version. And we will select the correct chip architecture according to your device. However, before you think if doing so, you would need to remove your manual installed version. Otherwise it will be conflicting with ours.