MichaIng/DietPi #5462: Adding a Power button to an Odroid C4

MDAR, 2022-04-28:
Hi

Before I fall down a deep rabbit hole and find myself lost.

Do you think the following would work with DietPi on an Odroid C4?

[GPIO Power Button config](https://wiki.odroid.com/odroid-c4/application_note/gpio/gpio_key_wakeup#sw_set-up_using_bootini)

The instructions suggest that by adding a button across GPIO pin 11 and ground, then adding this text to Boot.ini, I should be able to get a working power button?

```
setenv bootargs "root=UUID=e139ce78-9841-40fe-8823-96a304a09859 rootwait rw ${condev} ${amlogic} no_console_suspend fsck.repair=yes net.ifnames=0 elevator=noop hdmimode=${hdmimode} cvbsmode=576cvbs max_freq_a53=${max_freq_a53} maxcpus=${maxcpus} voutmode=${voutmode} ${cmode} disablehpd=${disablehpd} cvbscable=${cvbscable} overscan=${overscan} ${hid_quirks} monitor_onoff=${monitor_onoff} logo=osd0,loaded ${cec_enable}"
 
### in case of GPIOX.3 (Pin 11) of 2x20 pins connector
setenv gpiopower "479"
setenv bootargs ${bootargs} gpiopower=${gpiopower}
```

What I am actually trying to achieve is a automatic graceful power off when a wider system in powered down and a relay controlled power on (as I will have to put a battery in line with the Odroid to keep it alive while it gracefully shuts down, which means the 12V will remain for days afterwards so the unit won't come to life when the main power is restored (unless the battery dies before hand)

Detecting that the rest of the system has powered down is the easy bit with a few carefully timed Ping commands, it's the powering back on that is the issue.

MichaIng, 2022-04-28:
Does the image you use actually ship with a `/boot/boot.ini`? An older image did, but our current C4/HC4 image uses a modern mainline kernel and U-Boot and a `/boot/dietpiEnv.txt` and `/boot/boot.cmd` instead. Many of the Hardkernel kernel command-line arguments are not valid with mainline Linux, which includes `gpiopower`. If you use the modern image indeed, we'd need to have a look into available device tree overlays or the Armbian forum about how to implement GPIO triggered shutdown in a way where constant polling is not required. A script which constantly reads GPIO states and in case does a shutdown in a loop would be an alternative, though not a very nice one (software/processing overhead).

MDAR, 2022-04-28:
Now you've mentioned all that, I've taken a closer look.

One of the machines here has a boot.ini file, but that's one I haven't updated for a while.

So....

As what I need is a way to switch a machine on, because issuing a poweroff command after it can't see a couple of its friends a few times (Node-Red "failed ping" and count routine) is fairly simple.

So...

How about, I put the timer relay between the battery and the C4.

But power it from before the battery.

Power down - C4 will run off the battery

Node-Red will issue a poweroff command (eventually)

When power is restored

The timer could CUT the power between the battery and the C4 for 5 seconds

C4 will reboot when the relay times out and restores the 12v to the C4.

Thanks.
Sometimes it takes someone to point out the issues to inspire a different solution.

closed 2022-04-28 by MDAR

MichaIng, 2022-04-28:
I'm not sure whether I do understand the whole solution 😅, but involving Node-RED to trigger battery power sounds clever. Great that you found that solution.

The question whether there is a native GPIO triggered shutdown/reboot possible remains, but I'll delay investigation until someone is explicitly asking for it.