MichaIng/DietPi #5463: How to get stretch update?

keeno1982, 2022-04-28:
I got error on my NanoPC-T4 stretch, I know stretch is old, but I like it can flashed into emmc. Please help, thanks.

root@DietPi:~# sed -i 's|https://oph.mdrjr.net|http://fuzon.co.uk|g' /etc/apt/sources.list.d/*.list
root@DietPi:~# G_AGUP
[  OK  ] APT update, please wait...
Get:1 http://fuzon.co.uk/meveric all InRelease
Err:1 http://fuzon.co.uk/meveric all InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Ign:2 http://ftp.debian.org/debian stretch InRelease
Get:3 http://fuzon.co.uk/meveric stretch InRelease
Err:3 http://fuzon.co.uk/meveric stretch InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Hit:4 http://ftp.debian.org/debian stretch-updates InRelease
Get:5 http://ftp.debian.org/debian stretch-backports InRelease [91.8 kB]
Hit:6 http://ftp.debian.org/debian stretch Release
Hit:7 https://deb.debian.org/debian-security stretch/updates InRelease
Ign:5 http://ftp.debian.org/debian stretch-backports InRelease
Fetched 94.4 kB in 1s (47.3 kB/s)
Reading package lists...
W: GPG error: http://ftp.debian.org/debian stretch-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9
W: The repository 'http://ftp.debian.org/debian stretch-backports InRelease' is not signed.
E: Failed to fetch http://fuzon.co.uk/meveric/dists/all/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: Failed to fetch http://fuzon.co.uk/meveric/dists/stretch/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: Some index files failed to download. They have been ignored, or old ones used instead.
E: Encountered a section with no Package: header
E: Problem with MergeList /var/lib/apt/lists/fuzon.co.uk_meveric_dists_all_n1_binary-arm64_Packages
E: The package lists or status file could not be parsed or opened.

Joulinar, 2022-04-28:
Are you sure your source list is correct? `http://fuzon.co.uk/meveric` doesn't seems to be valid one. 

And yes, we don't support Stretch anymore.

keeno1982, 2022-04-28:
I also change http://fuzon.co.uk/meveric with https://oph.mdrjr.net/meveric in /etc/apt/sources.list.d/meveric*.list and it not work too.

MichaIng, 2022-04-28:
Jep, not sure where you got this from, but `http://fuzon.co.uk` is the ancient DietPi server that was used years ago as mirror for Meveric's APT repository. `https://oph.mdrjr.net` is the original repo, which is still valid, so not sure why you changed away from it?

In the meantime we have a mirror on our main server, which should have been applied via DietPi-Update to all Odroids years ago as well. So would be interesting which DietPi version you use. However, you can apply that mirror manually:
```sh
sed -i 's|https://oph.mdrjr.net|https://dietpi.com|g' /etc/apt/sources.list.d/*.list
sed -i 's|http://fuzon.co.uk|https://dietpi.com|g' /etc/apt/sources.list.d/*.list
```

MichaIng, 2022-04-28:
Ah lol wait:
> I got error on my NanoPC-T4 stretch

So it is a NanoPC and not an Odroid? The repository you are trying to use is for Odroids only. Did you add it manually for some reason? Our NanoPC T4 image never shipped with any of these repositories installed 🤔. Can you please show:
```sh
for i in /etc/apt/sources.list{,.d/*.list}
do
[[ -f $i ]] || continue
echo "$i:"
cat "$i"
done
```

keeno1982, 2022-04-28:
/etc/apt/sources.list:
deb http://ftp.debian.org/debian/ stretch main contrib non-free
deb http://ftp.debian.org/debian/ stretch-updates main contrib non-free
deb https://deb.debian.org/debian-security/ stretch/updates main contrib non-free
deb http://ftp.debian.org/debian/ stretch-backports main contrib non-free
/etc/apt/sources.list.d/meveric-all-main.list:
deb http://fuzon.co.uk/meveric all main
/etc/apt/sources.list.d/meveric-all-N1.list:
deb http://fuzon.co.uk/meveric all n1
/etc/apt/sources.list.d/meveric-all-testing.list:
deb http://fuzon.co.uk/meveric all testing
/etc/apt/sources.list.d/meveric-stretch-backports.list:
deb http://fuzon.co.uk/meveric stretch backports
/etc/apt/sources.list.d/meveric-stretch-main.list:
deb http://fuzon.co.uk/meveric stretch main

MichaIng, 2022-04-28:
You do not need, resp. must not use Android tools for the new image anymore. Simply use your favourite flashing tool, Rufus, BalenaEtcher or something like that.

Ah, but the T4 has a fixed eMMC chip, not a module that could be attached to a USB-eMMC adapter and written to externally, right? You could clone the whole SD card over to eMMC from within the running system: https://dietpi.com/blog/?p=1236
_To keep it simple, use an SD card of less than 16 GiB, so you do not need to shrink it before cloning from SD card to internal eMMC._

MichaIng, 2022-04-28:
As said, of you use a small SD card, then the whole step 2 can be skipped, so it remains essentially verifying the device names via `lsblk`, stopping services and re-mounting the root filesystem as read-only, then doing the clone via `dd`.

keeno1982, 2022-04-29:
Maybe Bullseye img had problem, it can't boot on my set by sd card, but Buster works fine, please check. Thanks.

MichaIng, 2022-04-29:
Are there any logs/error messages on screen?

Joulinar, 2022-05-06:
In this case I'm going to close the issue. Feel free to reopen if needed.

closed 2022-05-06 by Joulinar

keeno1982, 2022-05-08:
Very unlucky, I tried flash OS from sd(16G) to emmc(16G), but failed on 99% 

root@DietPi:~# dd if=/dev/mmcblk1 of=/dev/mmcblk2 bs=1M status=progress
15625879552 bytes (16 GB, 15 GiB) copied, 316 s, 49.4 MB/s
dd: error writing '/dev/mmcblk2': No space left on device
14911+0 records in
14910+0 records out
15634268160 bytes (16 GB, 15 GiB) copied, 335.158 s, 46.6 MB/s

keeno1982, 2022-05-08:
Only a suggestion, I find that Armbian provides a easy tool to change OS boot from, like sd to emmc or sd to usb, it's very easy to use. But I like Dietpi more : ), So I hope Dietpi can do the same thing.

MichaIng, 2022-05-09:
We have some guide about how to do this manually: https://dietpi.com/blog/?p=1236

Works also with CLI tools fdisk and resize2fs to reduce the partition and filesystem size, so you can stop dd after the end of the root partition has been reached.

Indeed unlucky that the eMMC is slightly smaller than the SD card.

keeno1982, 2022-05-12:
Thanks, I had reported to https://github.com/MichaIng/DietPi/issues/5487