MichaIng/DietPi #5467: dietpi-update to 8.4.2 fails: Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked

PeterLacknase, 2022-05-01:
#### Details:
- Date           | Sun May  1 10:50:16 CEST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux dietpi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. sudo dietpi-update
#### Expected behaviour:
should update
#### Actual behaviour:
doesn't update; woes above
#### Extra details:
tried it on two local instances (rpi3B; rpi zeroW); always the above error 
#### Additional logs:
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.
```

Joulinar, 2022-05-01:
Hi, did you upgrade both systems from Buster to Bullseye? 

You could try following

```
apt install systemd-timesyncd
```

PeterLacknase, 2022-05-01:
`$ cat /etc/apt/sources.list
deb https://mirror.netcologne.de/raspbian/raspbian/ bullseye main contrib non-free

$ sudo apt install systemd-timesyncd
[...]
Bla ... installed`

Tried again ... works now. Thank You!

closed 2022-05-01 by Joulinar

MichaIng, 2022-05-01:
We should have added a pre-patch with DietPi v8.4:
```sh
dpkg-query -s systemd-timesyncd &> /dev/null && [[ $(apt-mark showauto systemd-timesyncd) ]] && G_EXEC apt-mark manual systemd-timesyncd
```
There have been quite some similar cases, also one on Twitter. We have it covered in our upgrade guide but not everyone is following it and some did upgrade before the guide existed or before the explicit `systemd-timesyncd` install has been added.

Joulinar, 2022-05-01:
Can we add it still? I mean the release ist not that old still 😉

MichaIng, 2022-05-01:
Hmm, yeah actually we can. Not so clean but it doesn't affect any DietPi runtime scripts, just the one that it downloaded, executed and removed during updates. I'll do that, should save some users some hassle, and the patch is pretty clear and does not change anything, just assures that a package which is currently kept installed as dependency, is kept installed as well when not being a dependency anymore.

MichaIng, 2022-05-02:
Added pre-patch for v8.5: https://github.com/MichaIng/DietPi/commit/ed9b4c0
Backported to v8.4: https://github.com/MichaIng/DietPi/commit/5a7be62

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] DietPi-Pre-patches | Marking systemd-timesyncd as manually installed to prevent its autoremoval
[  OK  ] DietPi-Pre-patches | apt-mark manual systemd-timesyncd
[  OK  ] DietPi-Update | Successfully applied pre-patches
```