MichaIng/DietPi #5468: Iteration v8.5 (release plan)

MichaIng, 2022-05-01:
## Day 0: Init new development cycle
To be done until: `2022-05-01`
- [x] Create a new GitHub milestone for DietPi code https://github.com/MichaIng/DietPi/milestones
- [ ] Create a new GitHub milestone for DietPi doc https://github.com/MichaIng/DietPi-Docs/milestones
- [x] Select issues for the new iteration in the new milestone
- [ ] Create a new section in the [DietPi Wiki](https://github.com/MichaIng/DietPi/wiki/Iteration-plans)
- [x] A new development cycle starts with a subversion increment, or major version increment when we drop support for an old Debian version or old DietPi versions. The RC version during the development cycle is "-1", to clearly identify unstable version strings.
- [x] Init [`CHANGELOG.md`](https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt)
- [x] Add support to [DietPi-Survey_report](https://github.com/MichaIng/DietPi/blob/dev/.meta/dietpi-survey_report)

## 3rd week: Beta phase
To be done until: `2022-05-21`
- [ ] Merge live patches from `master` and remove in `dev`.
- [ ] Create [pull requests](https://github.com/MichaIng/DietPi-Docs/pulls) for all [required documentation changes](https://github.com/MichaIng/DietPi-Docs/issues?q=is%3Aopen+is%3Aissue+label%3A"Next+DietPi+release"):

To be done at: `2022-05-21`
- [ ] Launch the initial beta as `.0` RC version
- [ ] Advertise on GitHub
- [ ] Advertise on [DietPi Forum](https://dietpi.com/phpbb/viewforum.php?f=9)
- [ ] Advertise on DietPi-Banner [MOTD](https://dietpi.com/motd)

To be done at: `2022-05-22`
- [ ] Advertise on website slider: https://dietpi.com/
- [ ] Advertise on Twitter ([DietPi account](https://twitter.com/DietPi_))
- [ ] Advertise on Facebook ([DietPi group](https://www.facebook.com/groups/167403007229675))

To be done from: `2022-05-22` - `2021-05-26`
- [ ] Address bug reports with further beta iterations, e.g. `.1` and `.2` RC versions

## 2 days before the release
To be done until: `2022-05-26`
- [ ] Have a release PR open on the main repo which needs to be linked from the [changelog](https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt)/release notes.
- [ ] Assure that [update server version](https://github.com/MichaIng/DietPi/dev/beta/.update/version) and [DietPi-Globals default version](https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-globals) match.
- [ ] Have the docs release PR ready and reviewed, with release notes and required changes merged into dev already.
- [ ] Have a PR on the website ready to advertise the new release via website slide and have the docs PRs all ready and reviewed.

## Release
To be done at: `2022-05-28`
- [ ] Make the release
- [ ] Publish the release notes in [DietPi Docs](https://dietpi.com/docs/releases/)
- [ ] Link the release notes from [website slider](https://dietpi.com/)
- [ ] Mention the release notes in [DietPi Forum](https://dietpi.com/phpbb/viewforum.php?f=10)
- [ ] Update the [MOTD](https://dietpi.com/motd)
- [ ] Update software list in [DietPi Wiki](https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list)

## Release announcements
To be done at: `2022-05-29`
- [ ] Publish on Twitter ([DietPi account](https://twitter.com/DietPi_))
- [ ] Publish on several Facebook groups (e.g. [DietPi group](https://www.facebook.com/groups/167403007229675))
- [ ] Announce on [Hacker News](https://news.ycombinator.com/)
- [ ] Forge and spread a press release
- [ ] Determine if a hotfix release is needed
- [ ] Create a new release plan

Joulinar, 2022-06-19:
I guess something to close 😄

MichaIng, 2022-06-19:
Ah yeah, I wanted to create a new release plan first 😄.

closed 2022-06-19 by MichaIng