MichaIng/DietPi #5469: Update partially failed

omavoss, 2022-05-02:
#### Details:
- Date           | Mon May  2 18:50:09 CEST 2022
- DietPi version | v8.4.0 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | `Linux RasPi1 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 arm                                                                                                                               v6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not fo                                                                                                                               und.
```
Please helb my DietPi, thank you.

MichaIng, 2022-05-02:
Looks like you were unlucky to do the update shortly before I added the pre-patch for this, respectively when somewhere in GitHub cache the old pre-patch file was still stored: https://github.com/MichaIng/DietPi/commit/5a7be62
Or `systemd-timesyncd` has been autoremoved already after a previous `apt autoremove`, however simple to solve.

#### Solution
```sh
apt install systemd-timesyncd
dietpi-update 1
```

#### Background
When upgrading from Buster to Bullseye, `systemd-timesyncd` becomes an own package, previously it was included in `systemd` itself. Until a quite recent `systemd` update, `systemd-timesyncd` was still pulled in and kept as dependency, but got now degraded to a "recommendation" so that any `apt autoremove` removes it, unless it gets marked as "manually" installed or in fact gets manually installed. We cover this in our Buster => Bullseye upgrade guide already, but AFAIK that step wasn't inside right at the beginning, and of course you may have followed other guides or knew the basic steps yourself for the upgrade, so that this part was missing: https://dietpi.com/blog/?p=811#2.2.4-software-migrations

omavoss, 2022-05-02:
Thank you very much, my DietPi executed the solution successfully.

closed 2022-05-02 by omavoss