MichaIng/DietPi #5470: DietPi Update restarts disabled services when finishing

Wondering-Soul, 2022-05-03:
- DietPi version | 8.4.2
- Distro version | Bullseye 1
- Kernel version | 5.15.32+
- SBC model | RPi B+ (armv6l)
- Power supply used | Official RaspberryPi Supply 5.1V / 2.5A
- SD card used | 16GB SanDisk Ultra

Foreword: I am not entirely certain if this is working as intended or not. I understand that setting service priorities etc with `update-rc.d` is not a perfect solution. I know that if I were to run `update-rc.d -f lighttpd remove`, the service would no longer start at boot, however if the package is updated this change will be reversed. I am aware that I can specify specific run levels to ensure that a given service doesn't restart when upgraded, I am wondering whether this is the limitation I am running into or whether DietPi is just ignoring changes made with `update-rc.d` or if the services are registered with DietPi separately somewhere?

#### Steps to reproduce
1. Disable a service using `update-rc.d -f service_name remove`
2. Run `dietpi-update`

#### Expected behaviour
- DietPi Updater should stop all registered running services, complete the update and restart ONLY the enabled services to ensure we return to the same state we were in before updating.

#### Actual behaviour
- Services previously disabled with `update-rc.d` are re-started during the update process.

#### Additional Information:
- As far as I am aware, the given service was not updated during the upgrade.

MichaIng, 2022-05-03:
This is a known limitation:
- All DietPi-controlled services are "disabled" from the init system point of view, so that they are not started by the init system at boot but by DietPi-Services in well defined order. _However, systemd allows simple ordering as well, so that we aim to change this to allow more parallelism at boot and remove this "DietPi-controlled" vs "disabled" confusion._
- Until now however this means that disabling the service (when it is DietPi-controlled already) doesn't change something. You need to "exclude" it either in `dietpi-services` or by adding a line like `- apache2` to `/boot/dietpi/.dietpi-services_include_exclude`. However, it is then neither stopped nor started before/after DietPi updates.

`update-rc.d` btw is the SysV service utility, while DietPi uses `systemd` as init system:
```sh
systemctl disable service_name
```
_This is synced with the SysV runtime levels as well, if the service is actually a SysV service._

> DietPi Updater should stop all registered running services, complete the update and restart ONLY the enabled services to ensure we return to the same state we were in before updating.

That would be inconsistent. If a service is not stopped, it must not be started, and if a service is not started afterwards, it must not be stopped before. Basically what I aim to do is to simply use the `systemctl is-enabled service_name` status to decide whether to start/stop services after/before DietPi maintenance tasks (updates, software installs, backups).

Wondering-Soul, 2022-05-06:
Thanks as always for your detailed explanation of the issue and I think your planned changes are a great solution.

MichaIng, 2022-05-07:
Let me open the issue to track the planned enhancement. I thought there was already one about it, but I couldn't find it.

MichaIng, 2023-03-30:
All done for DietPi v8.16 😃.

closed 2023-03-30 by MichaIng