MichaIng/DietPi #5471: dietpi-drive_manager doesn't  move user_data to NTFS drive

blackdev1l, 2022-05-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
    G_DIETPI_VERSION_CORE=8
    G_DIETPI_VERSION_SUB=4
    G_DIETPI_VERSION_RC=2
    G_GITBRANCH='master'
    G_GITOWNER='MichaIng'
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
- SBC model | RPi 3 Model B (aarch64)

#### Additional Information (if applicable)
I used to have a ntfs hard drive mounted with dietpi and use it for user_data, but now that i clean installed the system i can no longer do it because it says 

````
 Filesystem type not supported:  /mnt/disk1 has a filesystem type of ntfs, which is not supported.   
The filesystem type must be ext2/3/4, F2FS, Btrfs, XFS or ZFS for native symlink and UNIX permissions 
You may format the partition with a supported filesystem type and retry.       
```````

#### Steps to reproduce
1. mount an ntfs drive
2. try to move user_data to it
3. ???
4. profit

#### Expected behaviour
move user_data to the ntfs drive

#### Actual behaviour
error given 

#### Extra details
i think the problem is that the script checks for filesystem type, it finds `fuseblk` instead of ntfs and it goes on error, is there any way to fix this ? Thanks

Joulinar, 2022-05-03:
Somehow as expected as we don't support NTFS drives as target location for DietPi userdata anymore.

MichaIng, 2022-05-03:
Jep, NTFS lacks proper symlinks for reliable file modes support, strictly required for most software to manage data.

MichaIng, 2022-05-05:
No way, this is not a limitation of DietPi but one of Linux software and NTFS drives. If you need access from Windows, use exFAT which is natively supported by Linux with recent kernel.

MichaIng, 2022-05-29:
Marking this as closed. Feel free to reopen when there are still questions.

closed 2022-05-29 by MichaIng