MichaIng/DietPi #5472: Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found

CodeSteph, 2022-05-04:
#### Details:
- Date           | Tue Apr 19 08:56:26 CEST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCK Pi 4 (aarch64) (ID=72)
- Kernel version | `Linux FushichoPi 5.15.25-rockchip64 #22.02.1 SMP PREEMPT Sun Feb 27 09:05:47 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync
#### Steps to reproduce:
the problem occures while I was executing dietpi-update
#### Expected behaviour:
no error is expected
#### Actual behaviour:
- Command: systemctl start systemd-timesyncd   
- Exit code: 5                                                              
- DietPi version: v8.3.1 (MichaIng/master) | HW_MODEL: 72 | HW_ARCH: 3 | DISTRO: 6   
- Image creator: DietPi Core Team                                          
- Pre-image: Armbian                                               
- Error log:                                                                                                        
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.  
#### Extra details:
[image]

#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```

Joulinar, 2022-05-04:
usually this should have been fixed already. 

Can you share output of following

```
dpkg-query -s systemd-timesyncd &> /dev/null && [[ $(apt-mark showauto systemd-timesyncd) ]] && G_EXEC apt-mark manual systemd-timesyncd
```

MichaIng, 2022-05-04:
It is only fixed if `systemd-timesyncd` was not already autoremoved. It it is, we cannot know whether it was intended or not, so I didn't want to touch things then.

Joulinar, 2022-05-04:
Ah okay sorry that was a misunderstanding on my side

CodeSteph, 2022-05-04:
Indeed @MichaIng is right,

The following NEW packages will be installed:
  systemd-timesyncd
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.

It's fixed, 
Thx

closed 2022-05-04 by CodeSteph