MichaIng/DietPi #5473: WiFi Hotspot + Pi-hole not playing nice

privacyguy123, 2022-05-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bullseye 0`
- Kernel version | `Linux DietPi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | ThePiHut 5v 
- SD card used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | WiFi Hotspot + Pi-hole
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Install/setup Pi-hole
2. Install WiFI Hotspot package - get errors.

#### Expected behaviour

WiFi Hotspot package should work out of the box ...

#### Actual behaviour

isc-dhcp-server service is failing because I am running Pi-hole as DHCP server already - I'm too stupid to work out how to fix this, I assumed DietPi would detect this and edit .confs ...

MichaIng, 2022-05-05:
Many thanks for your report, great find 👍.

Indeed this has never been recognised. Out of interest, does Pi-hole setup correctly a way that it listens on the WiFi interface instead of on the internet facing Ethernet interface?

While it is trivial to allow skipping the DHCP server when Pi-hole is installed before or together with the WiFi hotspot, I wonder how we should handle it when the the WiFi hotspot was installed before. Although the DHCP server on Pi-hole cannot be enabled right within the installer, can it?

privacyguy123, 2022-05-05:
I think no because it was off when Pi-hole was installed - I could reconfigure it to pick up the wlan0 interface maybe?

I enable it after installation but maybe there's a conf you could edit? I think having DHCP enabled at install would be a terrible idea though because it should be turned off on your router first. What I can't get my head around is this goofy IP the DietPi installer is assigning the wlan0 interact - 192.168.42.0 l, my router assigns IPs from 192.168.0.10-254 from 192.168.0.1, where is it getting this weird number?

Joulinar, 2022-05-05:
We define the WiFi hotspot as an own separate network. Therefore it will get an own IP range and a DHCP server. On default configuration it is not just an AP point, extending your existing network.

privacyguy123, 2022-05-05:
> We define the WiFi hotspot as an own separate network. Therefore it will get an own IP range and a DHCP server. On default configuration it is not just an AP point, extending your existing network.

All of this is over my head - how do I get it to play nicely with Pi-holes DHCP server?

Joulinar, 2022-05-05:
maybe you could describe what you are planing to do and we could try to help.

privacyguy123, 2022-05-05:
> maybe you could describe what you are planing to do and we could try to help.

Unfortunately I guess what I'm trying to do is not possible confirmed by Michalng!

Joulinar, 2022-05-05:
well that's maybe a misunderstanding. If you have Ethernet as well as WiFi connected, you could create the AP. What is not working, to use the very same WiFi adaptor to broadcast the AP and to connect to internet.

privacyguy123, 2022-05-05:
> well that's maybe a misunderstanding. If you have Ethernet as well as WiFi connected, you could create the AP. What is not working, to use the very same WiFi adaptor to broadcast the AP and to connect to internet.

Yes correct - Pi-holes DHCP server (the issue) is saying there is no available IPs to assign to 192.168.42.0 which is the IP supplied by DietPi in the configs.

MichaIng, 2022-05-07:
I think you need to clarify the exact setup you want to have:
- Is your DietPi connected via Ethernet to a router, so it can reach the Internet through this Ethernet connection?
- Other network members are not connected to the router, but can only connect via WiFi to this DietPi system?
- You want Pi-hole to be the DNS and DHCP server for this WiFi network, and you want to share the DietPi systems Ethernet Internet connection to the WiFi network?

> EDIT: And Pi-hole is serving IPs from 192.168.0.0/24 and complaining 192.168.42.x is out of range

Makes sense. `192.168.42.x` is the subnet range `192.168.42.0/24`, while `192.168.0.0/24` spans only 192.168.0.1 - 192.168.0.255. The CIDR notation `/24` means that the first 24 bits, i.e. the first three octals are fixed and only the 255 addresses defined by the last octal are included.

privacyguy123, 2022-05-07:
Is it possible to change this so the WiFi Hotspot tries to assign IPs within the Pi-holes configured range?

MichaIng, 2022-05-09:
Well, at least we can skip installing the dedicated DHCP server (isc-dhcp-server) so one can configure and run DHCP through Pi-hole.

privacyguy123, 2022-05-09:
> Well, at least we can skip installing the dedicated DHCP server (isc-dhcp-server) so one can configure and run DHCP through Pi-hole.

Ok - so I am confused now. The setup that I am describing is possible or not possible on DietPi?

MichaIng, 2022-05-10:
As of the DHCP server, currently is is always installed with the hotspot, but we can change that for next DietPi release.

privacyguy123, 2022-05-10:
> it is possible but it require manuell configuration as we don't have this scenario available within our standard installation scripts.

Could you talk me through this? I have followed many guides online that suggest making a bridge ```br0``` but this bricks my Pi and I can't connect in to fix it so have to wipe it and start again.

MichaIng, 2023-08-20:
Closing this in favour of a newer issue where we discussed options a little more detail: #6485

closed 2023-08-20 by MichaIng