MichaIng/DietPi #5478: Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.

dr4ugr, 2022-05-06:
#### Details:
- Date           | Fri May  6 10:49:06 EDT 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux DietPi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. In the console I run `sudo dietpi-update` which returns 
[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd

#### Expected behaviour:

- DiePi should update from 8.3.1 to 8.4.2
#### Actual behaviour:

- Cannot start timesync service due to it being masked
#### Extra details:

- In the console, I run: `sudo systemctl start systemd-timesyncd` which returns: `Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.`

- In the console I run `find /etc/systemd -type l -exec test ! -f {} \; -print` which returns:
/etc/systemd/network/99-default.link
/etc/systemd/system/apt-daily-upgrade.timer
/etc/systemd/system/apt-daily.service
/etc/systemd/system/systemd-timesyncd.service
/etc/systemd/system/dbus-org.freedesktop.timesync1.service
/etc/systemd/system/serial-getty@ttyS0.service
/etc/systemd/system/apt-daily-upgrade.service
/etc/systemd/system/sysinit.target.wants/systemd-timesyncd.service
/etc/systemd/system/apt-daily.timer
/etc/systemd/system/sshswitch.service
/etc/systemd/system/serial-getty@ttyAMA0.service
/etc/systemd/system/getty-static.service

- In the console I run `sudo systemctl start systemd-timesyncd.service` which returns `Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.`
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.

Joulinar, 2022-05-06:
Should be similar to this one I guess https://github.com/MichaIng/DietPi/issues/5467

Can you try following

`apt install systemd-timesyncd`

dr4ugr, 2022-05-06:
Was able to update that, thank you! Now seeing:

`$ sudo dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[  OK  ] DietPi-TimeSync | systemctl stop systemd-timesyncd
[  OK  ] DietPi-TimeSync | mkdir -p /run/systemd/timesync
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.4.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.3.1
[ INFO ] DietPi-Update | Latest version  : v8.4.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=11073 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
/boot/dietpi/dietpi-update: line 292: ./pre-patches: Permission denied
[FAILED] DietPi-Update | An error occurred during pre-patching. Please check the above log or /var/tmp/dietpi/logs/dietpi-update.log for errors, and rerun "dietpi-update" after the cause has been solved.`

Joulinar, 2022-05-06:
Are you using the root user?

dr4ugr, 2022-05-06:
I execute as root via `sudo`

MichaIng, 2022-05-07:
If it wasn't root, the script would exit right at the beginning.

```
[  OK  ] DietPi-Update | Applying execute permission
/boot/dietpi/dietpi-update: line 292: ./pre-patches: Permission denied
```
So execute permissions were applied successfully but executing the pre-patches script was still not permitted. Can you try this manually:
```sh
cd /tmp
curl -sSfLO "https://raw.githubusercontent.com/$G_GITOWNER/DietPi/$G_GITBRANCH/.update/pre-patches"
chmod +x pre-patches
ls -l pre-patches
```

dr4ugr, 2022-05-08:
I executed those commands without a problem. Same issue persists.

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercont                                                                ent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.4.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.3.1
[ INFO ] DietPi-Update | Latest version  : v8.4.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=11071 MiB | requir                                                                ed=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
/boot/dietpi/dietpi-update: line 292: ./pre-patches: Permission denied
[FAILED] DietPi-Update | An error occurred during pre-patching. Please check the above log or /var/tmp/dietpi/logs/dietpiupdate.log for errors, and rerun "dietpi-update" after the cause has been solved.

MichaIng, 2022-05-08:
The commands I provided were not for fixing the issue, but for detecting the issue. Please run them again and paste the full output here.

dr4ugr, 2022-05-09:
oof, my bad. See below.
`[username]@DietPi:~$ cd /tmp'
'[username]@DietPi:/tmp$ curl -sSfLO "https://raw.githubusercontent.com/$G_GITOWNER/DietPi/$G_GITBRANCH/.update/pre-patches"'
'[username]@DietPi:/tmp$ chmod +x pre-patches'
'[username]@DietPi:/tmp$ ls -l pre-patches'
'-rwxr-xr-x 1 [username] [groupname] 11103 May  9 10:15 pre-patches'

dr4ugr, 2022-05-09:
I also ran the following:
`sudo systemctl unmask systemd-timesyncd.service`
`sudo systemctl enable systemd-timesyncd.service`
`sudo systemctl start systemd-timesyncd.service`

and the re-ran sudo dietpi-update and received the same FAIL message.

Joulinar, 2022-05-09:
Just out of curiosity, could you try it as native user `roo`

dr4ugr, 2022-05-09:
I can...I disabled root for hardening purposes.

dr4ugr, 2022-05-09:
No change...

> root@DietPi:/home/[username]# dietpi-update
> 
>  DietPi-Update
> ─────────────────────────────────────────────────────
>  Phase: Checking for available DietPi update
> 
> [  OK  ] DietPi-Update | Checking network connectivity
> [  OK  ] DietPi-Update | Checking DNS resolver
> [ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
> [  OK  ] DietPi-Update | Got valid latest version: 8.4.2
> [  OK  ] DietPi-Update | Update available:
> [ INFO ] DietPi-Update | Current version : v8.3.1
> [ INFO ] DietPi-Update | Latest version  : v8.4.2
> 
>  DietPi-Update
> ─────────────────────────────────────────────────────
>  Phase: Checking for update pre-requirements
> 
> [  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
> [  OK  ] DietPi-Update | Free space check: path=/ | available=11068 MiB | required=100 MiB
> [ SUB1 ] DietPi-Services > stop
> [  OK  ] DietPi-Services | stop : cron
> [  OK  ] DietPi-Services | stop : lighttpd
> 
>  DietPi-Update
> ─────────────────────────────────────────────────────
>  Phase: Applying pre-patches
> 
> [  OK  ] DietPi-Update | Downloading pre-patches
> [  OK  ] DietPi-Update | Applying execute permission
> /boot/dietpi/dietpi-update: line 292: ./pre-patches: Permission denied
> [FAILED] DietPi-Update | An error occurred during pre-patching. Please check the above log or /var/tmp/dietpi/logs/dietpi-update.log for errors, and rerun "dietpi-update" after the cause has been solved.
> root@DietPi:/home/[username]#

MichaIng, 2022-05-10:
Confusing. Can you show:
```sh
mount | grep '/tmp'
```
Probably the noexec bit is set for /tmp?

Joulinar, 2022-05-10:
No the username is not confusing but the behauif of the update. You are the only one reporting it atm, so there seems something special on your install.

dr4ugr, 2022-05-10:
Yes, I did! I just recalled after cat'ing. I added the following:
`tmpfs /dev/shm tmpfs defaults,size=489M,noexec,nodev,nosuid 0 0`

Modified the second line to include `strictatime` instead of `noatime`. 

Also modified the size of /var/log to `500M`

```
@DietPi:~$ cat /etc/fstab
# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /dev/shm tmpfs defaults,size=489M,noexec,nodev,nosuid 0 0
tmpfs /tmp tmpfs defaults,size=1023M,strictatime,noexec,nodev,nosuid,mode=1777 0 0
tmpfs /var/log tmpfs defaults,size=500M,noatime,nodev,nosuid,mode=1777 0 0

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf, glusterfs, bind, Btrfs subvolume mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=9282c8f4-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=9282c8f4-01 /boot vfat noatime,lazytime,rw 0 2
/var/swap none swap sw
```

Joulinar, 2022-05-10:
That's pretty much explain your issue know as our update script are stored and execute inside `/tmp`

dr4ugr, 2022-05-10:
can you paste what should be in that file? I've commented the lines out and it's still failing.

Joulinar, 2022-05-10:
thats how it looks like for my system

```
tmpfs /tmp tmpfs size=1944M,noatime,lazytime,nodev,nosuid,mode=1777
```

File system size would need to be adjusted on your needs. Usually we set it to half of the physical memory.

dr4ugr, 2022-05-10:
still doesn't work. I'm going to run this for a while so I can update a new r-pi for my network. I used the post on reddit (below) to harden the r-pi.
https://www.reddit.com/r/pihole/comments/ap69s0/guide_security_hardened_dietpi_os_for_a_great/

Joulinar, 2022-05-10:
is it still the same error message? Did you reboot your system after the change? Can you share following after the reboot

```sh
mount | grep '/tmp'
```

MichaIng, 2022-05-10:
To apply the new flags without a reboot:
```sh
mount -o remount /tmp
```

dr4ugr, 2022-05-10:
> is it still the same error message? Did you reboot your system after the change? Can you share following after the reboot
> 
> ```shell
> mount | grep '/tmp'
> ```
> 
> Btw: you have the following active still? `/etc/systemd/system/local-fs.target.wants/tmp.mount`

Here's what I see:
```
root@DietPi:/home/[name]# mount | grep '/tmp'
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime,lazytime,size=1990656k)
```

dr4ugr, 2022-05-10:
> ```shell
> mount -o remount /tmp
> ```

I had tried this and a reboot, sill didn't work.

MichaIng, 2022-05-10:
With the exact same "Permission denied" error? 🤔

dr4ugr, 2022-05-10:
> m

I just took another look at /etc/fstab...I forgot to comment out one more item. I remounted, as discussed, and it's updating now.

dr4ugr, 2022-05-10:
Yes, will report back and close the issue when/if completes without issue.

dr4ugr, 2022-05-10:
Which, it just did. Thank you @MichaIng and @Joulinar for all of your assistance!

closed 2022-05-10 by dr4ugr