MichaIng/DietPi #5480: Unable to re-install Pi-Hole or Adguard

adityaa2b, 2022-05-07:
Failure description below:

#### Details:
- Date           | Sat May  7 16:17:15 BST 2022
- DietPi version | v8.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -nc 1 -W 10 dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ping: dns9.quad9.net: Temporary failure in name resolution
```

Joulinar, 2022-05-07:
Did you set the DNS server on the DietPi device to localhost?

`cat /etc/resolv.conf`

adityaa2b, 2022-05-08:
Yes this is set. The static IP that is provided is set in the file.

Joulinar, 2022-05-08:
What is set inside the file exactly?

adityaa2b, 2022-05-08:
nameserver and the static IP address.

Joulinar, 2022-05-08:
Which address exactly

adityaa2b, 2022-05-08:
192.168.1.2 is the address provided by the router. The same is added in the static IP address setting in the router and the same is present in dietpi config file

Joulinar, 2022-05-08:
Is this the address of the DietPi device? Do you use STATIC or DHCP on the DietPi device?

adityaa2b, 2022-05-13:
Yes this is the address of the dietpi device and it is static.

Joulinar, 2022-05-13:
I don't recommend to use PiHole and/or AGH as DNS server for the DietPi device who is hosting them. This is leading to exactly the issue you have now. As soon as the local DNS server is not available anymore, you will not be able to resolve any DNS request on this DietPi system. Better to configure DietPi local network settings to use a global upstream DNS like Quad9 or Cloudlfare instead. Usually there is nothing to blocked on DietPi (at least as long as you don't use a desktop/browser). This way DNS resolution on the local DietPi system continue to work even if PiHole and/or AGH are failing. As well, it did not have any influence into the way how LAN clients are going to resolve DNS request.

adityaa2b, 2022-05-13:
ok.. Will this help me blocking the ads?

Joulinar, 2022-05-13:
DNS resolution of the DietPi device themselves has nothing to do with blocking ads by your network clients. These are 2 different thinks.

adityaa2b, 2022-05-13:
I am using dietpi just for blocking ads on the device connected to wifi. So I am making dietpi as my primary dns resolver so that all the request sent and received are passed through dietpi and if anything is related to ads then it will be blocked. This was working fine earlier but when I updated the pi version to new release it stopped working. Like once I setup pi hole or adgaurd it works but the ads are not blocked. So i uninstall  it and re-install it throws me the error that I have attahced.

Joulinar, 2022-05-13:
again, these are 2 different thinks. DNS resolution on your DietPi system directly has nothing to do on how your WiFi clients will do this. You could go into `dietpi-config` and adjust the network settings without any issue. You WiFi clients will continue to use the DNS they get assigned vie DHCP/Router

closed 2022-05-13 by adityaa2b